MarlinFirmware/Marlin #4027: BABYSTEPPING not working?

edwilliams16, 2016-06-13:
I turned on BABYSTEPPING in Configuration_adv.h with

#define BABYSTEPPING
#if ENABLED(BABYSTEPPING)
  #define BABYSTEP_XY  //not only z, but also XY in the menu. more clutter, more functions
                       //not implemented for deltabots!
  #define BABYSTEP_INVERT_Z false  //true for inverse movements in Z
  #define BABYSTEP_MULTIPLICATOR 1 //faster movements
#endif

  The baby-stepping menus appeared on my LCD when printing, but stepping z up 0.4 mm with the arrow keys had no visible effect.  The print completed normally at exactly the same z height with no evidence of missing layers.

edwilliams16, 2016-06-13:
OK.  It is a multiplier issue.  I set the multiplier to 16  and was able to lift z enough to have an effect.
However the LCD appears to display steps/10 - mm would be more useful, so we need a factor of 10/STEPS_PER_UNIT somewhere in the LCD display routine.

edwilliams16, 2016-06-13:
I made a fix for this and submitted a pull request.  Since it's my first try at this git stuff be kind if I got the mechanics wrong.

adamfilip, 2016-06-14:
I would love to be able to set the control dial to by default control the babystep for Z. instead of the speed. when its in the idle menu. i never adjust speed. but i find i adjust babystepping often to get it started right.

can this be an option? 
let you choose the default behaviour of the control dial.

thinkyhead, 2016-06-15:
@adamfilip That sounds like a separate Feature Request. Please post a new issue with that request.

jbrazio, 2016-06-15:
Fixed by #4028

closed 2016-06-15 by jbrazio