MarlinFirmware/Marlin #4030: LCD_I2C_PANELOLU2 not working correctly with latest RC

taiffu, 2016-06-13:
I compiled latest source code for my prusa i3 clone (sanguinololu 1.3 + ic2 lcd + sdcard)
but lcd panel just blinking and beebing, menu rolling itself and changes views automatically.

All funktions are working fine with old stabile release

thinkyhead, 2016-06-14:
Does this problem still exist in the `RCBugFix` branch? We've done a lot of patches since `RC6` and might have fixed this one. If the problem still exists in the latest code, then we have a good starting point to fix the issue.

taiffu, 2016-06-14:
Hmm I took now that RCBugFix branc but now when I trying to compile it I got loooot's of errors.
I will look more this.

Short sample bellow.

 C:\Users\sami\AppData\Local\Temp\build15c6831deedbd5d6201e66142622daa3.tmp\sketch\Configuration.h:1200,

                 from C:\projektit\Marlin-RCBugFix\Marlin-RCBugFix\Marlin\Marlin.ino:37:

fastio.h:47: error: pasting "DIO" and "-" does not give a valid preprocessing token

 #define _READ(IO) ((bool)(DIO ## IO ## _RPORT & MASK(DIO ## IO ## _PIN)))

                           ^

Blue-Marlin, 2016-06-14:
@thinkyhead 
This could be a side effect of the MAX6675 fix. We looked after the sc-cards but not for the displays.

fabtopia, 2016-06-14:
i have the exact same configurations and use Rc6 with no display problems. can't there be a loose connector

taiffu, 2016-06-14:
Same HW works fine with older stabile version, so I think connectors are ok.

fabtopia, 2016-06-14:
Is it right from switching it on or does it starts after doing something?

taiffu, 2016-06-14:
The problem begins when a new firmware (RC6) starts. I will be double check my config and try again to compile RC6 since that RCBugFix branc not compiled.

thinkyhead, 2016-06-15:
I would test-compile this, but don't have `LiquidTWI2.h` installed at the moment. I'll have to continue tomorrow.

taiffu, 2016-06-15:
Ok I now pulled RC6 version via GIT intead download ZIP file. 
I checked my configuration compiled firmware and uploded it in my device.
Display seems works now with fresh source.
Since I can now continue to config my device this case possible close. This was possible my own mistake in code so sorry about this. Thanks to help for every body and keep good job continue :)

jbrazio, 2016-06-15:
@taiffu what is the Liquid Crystal I2C lib you're using ?

taiffu, 2016-06-16:
@jbrazio LiquidTWI2 lastes version from master branch. https://github.com/lincomatic/LiquidTWI2

closed 2016-06-17 by thinkyhead