MarlinFirmware/Marlin #4031: Marlin crashes!  We must have a problem with the SERIAL_PROTOCOL() routines???

Roxy-3DPrintBoard, 2016-06-13:
Please add this code to your Marlin_main.cpp file and give Marlin an M86 command.  It will repeat a few times (16 on my machine) and then Marlin will crash and reboot.    Why did I pick M86 for the sample code?   Well, it 86's my printer!  http://mentalfloss.com/article/51880/where-did-term-86-come

~~~~
      case 77: // Stop print timer
        gcode_M77();
        break;

      #if ENABLED(PRINTCOUNTER)
        case 78: // Show print statistics
          gcode_M78();
          break;
      #endif

      case 86: 	{
		int qwerty=0;
		SERIAL_ECHO_START;
    		SERIAL_PROTOCOLPGM("Checking for crash problem in SERIAL_PROTOCAL() routines:\n");
	      	while( !lcd_clicked() ) {
	      		delay(250);
	      		SERIAL_PROTOCOL( qwerty++ );
      			SERIAL_EOL;
		}
		}
        	break;
~~~~

Blue-Marlin, 2016-06-13:
If you have `#define USE_WATCHDOG` that is what you see.
Put an `idle();` in your loop, or at least a `thermalManager.manage_heater();`

Roxy-3DPrintBoard, 2016-06-13:
Thanks so much Blue-Marlin!!!!!!

closed 2016-06-13 by Roxy-3DPrintBoard

Blue-Marlin, 2016-06-13:
16 * 0.25 = 4 !!!

Roxy-3DPrintBoard, 2016-06-13:
Yes!   I feel like an idiot!   :)