MarlinFirmware/Marlin #4039: M109 Does not work at cooling down the temperature

Amachete, 2016-06-14:
Hello

If I write in my firt lines of my gcodes files:

M109 S195: Extruder at 195ºC

It Works, It waits up to extruder reaches 195ºC.

But If I write in my last gcodes, at the end of the File:

M109 S50;
M81; Shut off the power supply

It does not wait up to the extruder reaches 50ºC, and shuts off the supply, which makes risky for the Extruder support, because the heat can go up through the heat sink as the 12v  power supply is off therefore the extruder fan is off too. I want to cool down the extruder temperatura before shutting off the power supply.

Let me know if you can replicate this issue.

Thank you.

Blue-Marlin, 2016-06-14:
```
 * M109 - Sxxx Wait for extruder current temp to reach target temp. Waits only when heating
 *        Rxxx Wait for extruder current temp to reach target temp. Waits when heating and cooling
 *        IF AUTOTEMP is enabled, S<mintemp> B<maxtemp> F<factor>. Exit autotemp by any M109 without F
```

brainscan, 2016-06-14:
Actually I've noticed this recently too, and I do use M109 R60 in my end code. I wanted to test it a few more times to be sure but if other people are noticing it too then maybe it has stopped working. M190 R still seems to work. If I have more time today I'll check it again to be sure.

thinkyhead, 2016-06-15:
@Amachete You should use `M109 R50` if you want it to wait. `M109 S50` does not wait for cooling.

thinkyhead, 2016-06-15:
> I've noticed this recently too, and I do use `M109 R60`

Please test with `RCBugFix`. We're not aware of any issue with this.

Amachete, 2016-06-15:
"M109 S50" does not work and "M109 R50" does not work either...on RC6

I will try this evening with BUGFIX, and I will let you know.

brainscan, 2016-06-15:
I am using RCBugFix but it's a couple of weeks old, I'm going to download the latest version and check again.

geekgarage, 2016-06-29:
[image]
M109 R45 directly after usingg M109 S245 does not work on latest RCBugFix (Marlin 1.1.0-RCBugFix - 27 April 2016 downloaded 3 days ago)

So it does not work cooling down.

thinkyhead, 2016-06-29:
@geekgarage What are you using for `EXTRUDE_MINTEMP`? Marlin will not wait for temperatures below half of that value, to prevent wait-forever conditions.
```cpp
if (wants_to_cool && theTarget < (EXTRUDE_MINTEMP)/2) break;
```
Perhaps we should allow waiting for temperatures above 45° regardless, but in some hotter parts of the world  that might actually be too low. (I hear that in Australia the western desert lives and breathes at 45°)
```cpp
if (wants_to_cool && theTarget < min((EXTRUDE_MINTEMP)/2, 45)) break;
```

jbrazio, 2016-07-11:
Closed by #4169

closed 2016-07-11 by jbrazio

ProtomakerSprint, 2017-11-21:
Can any one help me. Thanks

**M190 R32; Seems to go at 54C and not 32C** 

Marlin (currently 1.1.6 – October 2017)

End code to wait for heat bed to cool before paying a tune and removing the model

Start Temperature is 65 C on the Heat bed for ABS in my case

Inside End G-Code>

; Part removal Temperature
M400; Wait for other commands to finish
M190 R32; Seems to go at 54C and not 32C : (R instead of S wait for this Temperature
M140 S0 ; make sure the bed is turned off.

> continue G-Gode play a tube which happens at 54C and not at the 32C requested

AnHardt, 2017-11-21:
You may want to play with
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L7668-L7673

ProtomakerSprint, 2017-11-21:
I have defaults of
/**
 * M109: Sxxx Wait for extruder(s) to reach temperature. Waits only when heating.
 *       Rxxx Wait for extruder(s) to reach temperature. Waits when heating and cooling.
 */

#ifndef MIN_COOLING_SLOPE_DEG
  #define MIN_COOLING_SLOPE_DEG 1.50
#endif
#ifndef MIN_COOLING_SLOPE_TIME
  #define MIN_COOLING_SLOPE_TIME 60
#endif

ProtomakerSprint, 2017-11-21:
And defaults:
      // Prevent a wait-forever situation if R is misused i.e. M190 R0
      if (wants_to_cool) {
        // Break after MIN_COOLING_SLOPE_TIME_BED seconds
        // if the temperature did not drop at least MIN_COOLING_SLOPE_DEG_BED
        if (!next_cool_check_ms || ELAPSED(now, next_cool_check_ms)) {
          if (old_temp - temp < MIN_COOLING_SLOPE_DEG_BED) break;
          next_cool_check_ms = now + 1000UL * MIN_COOLING_SLOPE_TIME_BED;
          old_temp = temp;
        }
      }

ProtomakerSprint, 2017-11-21:
Your bed is cooling slowly., Yes my heat bed can take some time to cool :)

ProtomakerSprint, 2017-11-21:
I don't have much of a clue what  1000UL is but I wounder if
MIN_COOLING_SLOPE_TIME 60 to say MIN_COOLING_SLOPE_TIME 300 could help

Do I just change Mairli_main.cpp then recompile upload as any other normal CONFIGURATION.H type changes, Thank You

ProtomakerSprint, 2017-11-22:
THANKING YOU AnHardt 2000%

Sorted now:-

Cooling Head Bed before carrying out next instruction.

M190 R32; Seems to go at 54C and not 32C

Ideal to allow Heat Bed to cool to a required Temperature before removing a model part from the Heat Bed

I found that just using the End Code M190 R32; Seems to go at 54C and not 32C : (R instead of S wait for this Temperature), as requested.

This issue seems to happened because my Heat Bed can take some time to cool down, about one degree a minute, half an hour for about 30 C to cool down

After some investigation, thanks to AnHardt, I discovered that changing Marlin_main.cpp code seem to work for me.

Now the Heat Bed Temperature waits until it is 32C as requested before continuing with the next G-Code command.

NOTE: When the M190 R32; command is reached in the End G-Code the 3D Pint Time on the LCD 12864 is lost because the Display will say:-

Bed Heating.  (Even though it is cooling) and the Time on LCD 12864 the will restart.

Once the Heat Bed is cooled down to the required temperature it will say:-
 
Bed Done.

Then the End G-Gode will continue...

End G-Code
M400 is most likely not required.

; Wait for Heat Bed to cool down to 32 C
; Part removal Temperature
M400; Wait for other commands to finish
M190 R32; Seems to go at 54C and not 32C : (R instead of S wait for this Temperature
; Below fails to set cool down at all
; M190 S32 ;set heatbed temperature to 32
; for cooldown M116 H1 ;wait for heatbed to reach 32
; M116 H1 ;wait for heatbed to reach 32
M140 S0 ; make sure the bed is turned off.

Marlin (currently 1.1.6 – October 2017)
 
In Marlin_main.cpp code, about line 7666 Aprox.

#if HAS_TEMP_BED

  #ifndef MIN_COOLING_SLOPE_DEG_BED
    #define MIN_COOLING_SLOPE_DEG_BED 1.50
  #endif
  #ifndef MIN_COOLING_SLOPE_TIME_BED

/*
Protomaker Black Sprint Original  

Increase twenty times
I am asuming this is seconds.
; Wait for Heat Bed to cool down to 32 C
; Part removal Temperature
M400; Wait for other commands to finish
M190 R32; Seems to go at 54C and not 32C : (R instead of S wait for this Temperature)
; Below fails to set cool down at all
; M190 S32 ;set heatbed temperature to 32
; for cooldown M116 H1 ;wait for heatbed to reach 32
; M116 H1 ;wait for heatbed to reach 32
M140 S0 ; make sure the bed is turned off.

WAS     #define MIN_COOLING_SLOPE_TIME_BED 60
*/
   #define MIN_COOLING_SLOPE_TIME_BED 1200
  #endif

AnHardt, 2017-11-22:
The
```cpp
  #ifndef MIN_COOLING_SLOPE_TIME_BED
    #define MIN_COOLING_SLOPE_TIME_BED 60
  #endif
```
should allow you to define:
```cpp
 #define MIN_COOLING_SLOPE_TIME_BED 1200
```
somewhere in your `Configuration.h`.
It is not there by default because we already have too much options.
I guess you are the first one who asked for this.

ProtomakerSprint, 2017-11-22:
In Configuration.h , I feel this is more Open Source and better to change instead of the change in Marlin_main.cpp code which works

May be, (hot tried yet), something to change here in Configuration.h >

// Bed temperature must be close to target for this long before M190 returns success
#define TEMP_BED_RESIDENCY_TIME 10  // (seconds)
#define TEMP_BED_HYSTERESIS 3       // (degC) range of +/- temperatures considered "close" to the target one
#define TEMP_BED_WINDOW     1       // (degC) Window around target to start the residency timer x degC early.