MarlinFirmware/Marlin #4041: Marlin restarts when controlled by PC

adamfilip, 2016-06-14:
I run a Rambo with Marlin RC6 Bugfix
I also use simplify3d as my controller/slicer
Windows 10 64bit PC

Often I turn on my printer (Makerfarm i3v 12") and tell it to start to preheat for ABS
then after 10 minutes I head into simplify 3d and connect to the rambo.
Once it connects. Marlin immediately restarts/reboots and I have to tell it to heat up again from S3D.

The USb cable stays physically connected all the time. 
I dont recall it doing this with previous versions of Marlin

tiagom62, 2016-06-15:
My (very limited) understanding is that is by design. 

When i connect using repetier-host marlin resets. I would guess is so that the firmware and software are at a known state.

Roxy-3D, 2016-06-15:
So...   I pretty much just use PronterFace.  But I see the exact opposite behavior.  I can reset the board, I can disconnect and reconnect the serial link.   I can do what ever I want and everything stays consistent and behaves as I expect.

But I do not unplug USB cables ever.   Just to help us get more information, can you try PronterFace and tell us if you still see the same problems?

thinkyhead, 2016-06-15:
When I use the "Connect" function in any host it resets Marlin.
I do believe this is by design.

jbrazio, 2016-06-16:
Yes, this is the Arduino's hardware auto-reset feature so you can upload "sketches" to the board using the bootloader/IDE combination.

closed 2016-06-16 by jbrazio

nmaupu, 2020-07-12:
You can disable the "autoreset" pretty easily if it's bothering you but you'll need some hardware modifications...
https://playground.arduino.cc/Main/DisablingAutoResetOnSerialConnection/

As the chip will not be auto reset, no more firmware upgrades will be possible after that except if you find a way to reset the board somehow.