MarlinFirmware/Marlin #4042: RCBugFix-bug?!

QoooQ, 2016-06-14:
hi guy i use RCBugFix firmware 
when i printing........3min later.........tune print stop 
the X-axis  is go and back cycle  ( go ahead    Retreat  cycle)
what is problem ???

and i change firmware 1.1.0 RC5 is  ok!!!

Roxy-3D, 2016-06-15:
Three partial sentences and we are expected to diagnose and solve your problem?

OK.  Here is one full sentence that should solve all your problems:

Set your configuration setting correctly.

thinkyhead, 2016-06-15:
You will have to do more diagnostics to help us figure out what the problem is.

QoooQ, 2016-06-16:
I update mp4   you can watch it know what problem?! Thanks
[https://www.dropbox.com/s/l92j69ttaio2vqh/VIDEO0002_CLIPCHAMP_keep.mp4?dl=0](url)

jbrazio, 2016-06-16:
Are you printing from SD or from a host ?

QoooQ, 2016-06-16:
SD card!!!

thinkyhead, 2016-06-16:
By "diagnostics" I was hoping you could try different "tricks" to see if you can cause a failure under certain conditions. If you can determine a definite action that causes the failure, that helps to point us in some definite direction. Unfortunately, a simple video only tells us that there is a problem, not what causes it, and not where to look. There is no use for us to simply scan all the code hoping the bug jumps out. And there have been so many changes since RC6 we cannot simply compare the old and new code either.

Grogyan, 2016-06-16:
From the video, it appears that the buffer is somehow executing the last few moves, in an endless loop.

maukcc, 2016-06-16:
This is the same behaviour I talked about when we were doing the keypad mashes thingy.

jbrazio, 2016-06-16:
@maukcc even with Scotts patch ?

maukcc, 2016-06-16:
I am using RC6 now, and have not checked RCbugfix for a bit. I will test when I have the time.

adamfilip, 2016-06-16:
Ive run into the exact same issue.. shaking after a stop is issued.

thinkyhead, 2016-06-17:
> I am using RC6 now, and have not checked RCbugfix for a bit.

As such an active contributor you should definitely be using `RCBugFix` as much as possible. We especially need feedback on our patches to those issues that you've been helping us discover.

thinkyhead, 2016-06-17:
Ok, so it sounds like the buffer – either the SD buffer or the regular command queue, is going into a state out of sync with itself. I will look at the older "Stop SD Print" code and compare to the current code. My guess is that either the ring buffer is getting corrupted, or the SD buffer is being submitted continually.

fbarcenas, 2016-06-23:
happen the same...

jbrazio, 2016-06-23:
I can confirm with latest RCBugfix I have the same loop behavior and have to power cycle.

thinkyhead, 2016-06-24:
@jbrazio I've looked at the code and still have no clue as to what is looping. All I know is, before I attempted to clear the command buffer, the machine would execute any commands that were still left in the buffer, but only once. So far I can't see where the buffer-handling logic is screwy, but obviously something is amiss!

Perhaps you can try putting some beeps and moves alternating in a GCode file, then SD print it, and use "Stop print" to see what happens. If it keeps beeping, then we know the command buffer is in an endless loop. If it keeps moving, but not beeping, then we know it's a planner or stepper buffer problem.

fbarcenas, 2016-06-30:
The beeper of my board is broken or happen something rare..., i can't test this.

Checking the problem a lot times i see than always when i stop do a diagonal in X/Y, never the extruder move.

Is like the buffer not is empty???¿?

thinkyhead, 2016-07-01:
> The beeper of my board is broken

It has never worked, or only with the current code?

> when i stop do a diagonal in X/Y,

When you click "Stop print" the printer continues to move diagonally until you turn it off?

> Is like the buffer not is empty

We have determined that much, now we're trying to figure out what the cause is. So far no one has added logging code to the command loop to see what's going on. But perhaps one of us will try that soon.

fbarcenas, 2016-07-03:
1º https://github.com/MarlinFirmware/Marlin/issues/4127
2º yes

ImmersedN3d, 2016-07-03:
I can also confirm that the issue exists on the current RC Bugfix build as of 7/1/16.   Stopping a print from SD results in a "Loop" of a short segment of gcode.

My beeper works.  ( I like the added beep features )

thinkyhead, 2016-07-03:
I'll be looking at this issue next time I get over to the workshop to mess with my printer. I'm sure it will turn out to be something very simple and foolish.

thinkyhead, 2016-07-04:
Here's something that I believe will fix the issue in the meantime, while we try to figure out the mistake in `clear_command_queue`.
```diff
 static void lcd_sdcard_stop() {
   card.stopSDPrint();
-  clear_command_queue();
+  // clear_command_queue();
   stepper.quick_stop();
   print_job_timer.stop();
   thermalManager.autotempShutdown();
   cancel_heatup = true;
   lcd_setstatus(MSG_PRINT_ABORTED, true);
   #if DISABLED(DELTA) && DISABLED(SCARA)
     set_current_position_from_planner();
   #endif // !DELTA && !SCARA
 }
```

maukcc, 2016-07-05:
What I also found in the lcd_sdcard_stop sequence is that it does the lcd_sdcard_stop. And after that it does 3 (not 2, not 4)commands from the gcode file that it has in buffer from where the stop was initiated.

> M190 S60 		;set bed temp and wait
> M109 T0 S195	;set nozzle 1 temp and wait
> M117 Cleaning .....
> G1 X1 F12000		;go to wipe point
> G92 E0			;set E
> G1 Z1.5 F1200

So if I initiate a stop after the M190, it will do the stop sequence and cut of current M190 (bed heating), and then after the sequence, it will do the M109, the M117 and the G1 X1 F12000 (3 commands). 

And then it is finished.

AnHardt, 2016-07-05:
Please try:
```diff
static char* current_command, *current_command_args;
- static int cmd_queue_index_r = 0;
- static int cmd_queue_index_w = 0;
- static int commands_in_queue = 0;
+ static volatile int cmd_queue_index_r = 0;
+ static volatile int cmd_queue_index_w = 0;
+ static volatile int commands_in_queue = 0;
static char command_queue[BUFSIZE][MAX_CMD_SIZE];
```

thinkyhead, 2016-07-05:
I don't see anyplace in the code where `volatile` would make a difference. The compiler _should_ be smart enough not to simply re-use the most recent fetched value in cases like these:
```cpp
  if (commands_in_queue < BUFSIZE) get_available_commands();
  . . .
  if (commands_in_queue) {
    . . .
    commands_in_queue--;
  }
```
```cpp
if (commands_in_queue == 0) stop_buffering = false;
. . .
while (commands_in_queue < BUFSIZE && MYSERIAL.available() > 0) {
  . . .
  _enqueuecommand(serial_line_buffer, true);
  . . .
}
```

thinkyhead, 2016-07-05:
> And after that it does 3 … commands from the gcode file that it has in buffer
> …the same results. 3 commands in "buffer"

Yes, that's the old behavior, and what I had hoped eliminate with the `clear_command_queue()` function. But we find that instead of actually clearing the command queue, it leaves the queue in a state where it continually loops instead.

Logging lines in just the right places should make it obvious what's going on. Unfortunately I am not near my printer today, so I cannot test this right now.

thinkyhead, 2016-07-05:
Ok, so I posted a branch with debugging added at https://github.com/thinkyhead/Marlin/tree/rc_debug_stop_print. It sets a flag to start displaying debug information before calling `clear_command_queue()`. If you have a host connected, then after using "Stop print" we will see the current values of the command-queue variables, used in fetching the next command.
```cpp
if (canceled_sd_print) {
  DEBUG_VAR(card.sdprinting);
  DEBUG_VAR(commands_in_queue);
  DEBUG_VAR(cmd_queue_index_r);
  DEBUG_VAR(cmd_queue_index_w);
}
```

AnHardt, 2016-07-05:
Ok. Got it.
I can reproduce the 'loop' now. Using a 'host' alters the behaviour by sending M105,M27. A simple terminal helps to see what is going on.

SDTEST.G
```
M190 S60 ;set bed temp and wait
G28
G1 X0 Y0 Z5
G1 X100 Y0
G1 X100 Y100
G1 X0 Y100
G1 X0 Y0
```

Log
```
echo:Now fresh file: SDTEST.G
File opened: 1642 Size: 101
File selected
ok P15 B3 R128 "M23 SDTEST.G" 
ok P15 B3 R128 "M24" 
 T:28.8 /0.0 B:56.0 /60.0 B@:0 @:0 W:?
 T:28.9 /0.0 B:56.0 /60.0 B@:127 @:0 W:?
 T:28.7 /0.0 B:55.9 /60.0 B@:127 @:0 W:?
 T:28.7 /0.0 B:55.8 /60.0 B@:127 @:0 W:?
 T:28.7 /0.0 B:55.8 /60.0 B@:127 @:0 W:?
 T:28.8 /0.0 B:56.0 /60.0 B@:127 @:0 W:?
 T:28.7 /0.0 B:56.0 /60.0 B@:127 @:0 W:?
 T:28.7 /0.0 B:56.0 /60.0 B@:127 @:0 W:?
 T:28.7 /0.0 B:56.2 /60.0 B@:127 @:0 W:?
 T:28.8 /0.0 B:56.3 /60.0 B@:127 @:0 W:?
 T:28.8 /0.0 B:56.5 /60.0 B@:127 @:0 W:?
card.sdprinting0
commands_in_queue-1
cmd_queue_index_r0
cmd_queue_index_w3
echo:enqueueing "M84 X Y Z E"
X:100.00 Y:50.00 Z:15.60 E:0.00 Count X: 9412 Y:4706 Z:62400
card.sdprinting0
commands_in_queue-1
cmd_queue_index_r1
cmd_queue_index_w0
card.sdprinting0
commands_in_queue-2
cmd_queue_index_r2
cmd_queue_index_w0
card.sdprinting0
commands_in_queue-3
cmd_queue_index_r3
cmd_queue_index_w0
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
card.sdprinting0
commands_in_queue-4
cmd_queue_index_r0
cmd_queue_index_w0
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
X:100.00 Y:50.00 Z:15.60 E:0.00 Count X: 9412 Y:4706 Z:62400
card.sdprinting0
commands_in_queue-5
cmd_queue_index_r1
cmd_queue_index_w0
card.sdprinting0
commands_in_queue-6
cmd_queue_index_r2
cmd_queue_index_w0
card.sdprinting0
commands_in_queue-7
cmd_queue_index_r3
cmd_queue_index_w0
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
card.sdprinting0
commands_in_queue-8
cmd_queue_index_r0
cmd_queue_index_w0
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
X:100.00 Y:50.00 Z:15.60 E:0.00 Count X: 9412 Y:4706 Z:62400
card.sdprinting0
commands_in_queue-9
```

```
        }
        process_next_command();
      }

    #else

      process_next_command();

    #endif // SDSUPPORT

    commands_in_queue--;
    cmd_queue_index_r = (cmd_queue_index_r + 1) % BUFSIZE;
  }
```
We are in `process_next_command();` executing G190 and there in ideling. In the inner idle we detect the stop and  - execute it - set ` commands_in_queue` to 0, leave M190, leave  `process_next_command();` reduce  `process_next_command();` by one to -1! Our test if we have a new command in the queue is `if (commands_in_queue)`.

thinkyhead, 2016-07-05:
This is what I figured, but I couldn't trace where inside of `loop()` it was setting `commands_in_queue` to zero. Anyway, the fix is simple:
```cpp
if (commands_in_queue) --commands_in_queue;
```

AnHardt, 2016-07-05:
```
    if (commands_in_queue) [
      commands_in_queue--;
      cmd_queue_index_r = (cmd_queue_index_r + 1) % BUFSIZE;
    }
```
!

thinkyhead, 2016-07-05:
Ok, I think we got it! #4214 is merged. Printing should now stop with no more commands processed!

closed 2016-07-05 by thinkyhead

maukcc, 2016-07-06:
Where does this code go? And what is the change? Then I can give it a test.

Blue-Marlin, 2016-07-06:
@maukcc 
The code is already in RCBugFix. More changes in that area are pending. #4217