MarlinFirmware/Marlin #4044: Advance algorithm discussion

edwilliams16, 2016-06-15:
I've seen two possible algorithms discussed
 (1) JKN  -  the system is modeled as a springy filament pushing molten fluid through a narrow pipe (the nozzle). Flow through the pipe is modeled by Poiseuille- Hagen flow, which describes laminar flow of a viscous, incompressible fluid.

 (2) Roberts(?)  - the system is modeled as a springy filament pushing molten fluid through a narrow pipe (the nozzle). Flow through the pipe is modeled by the Bernoulli equation which describes laminar flow of an inviscid, incompressible fluid.

  I've don't know whether this is the right place to post this, but I would like to point out that (2) is totally inappropriate.  What matters is comparing the pressure drop due to viscosity (1) and the pressure required (2) to accelerate the fluid to a higher velocity in the narrower nozzle. (The fluid gains kinetic energy so work has to be done on it by the pressure)  The ratio (2)/(1) is called the Reynolds number. For flow of water in your garden hose or air over an airplane wing, it is large and model (2) would be appropriate.
  But let's put numbers in for a model extruder:

/*********

    ----------------------
                          --------
                          --------
    ----------------------
           L,R,V            l,r,v

***********/

mu = 1000.0; // dynamic viscosity Pa-sec
R = (1.75/2) *1e-3;  // entrance radius m
r = (0.4/2)*1e-3;  // exit radius m
L = 40*1e-3;  // entrance length m
l = 2.0*1e-3;  // exit length m
rho = 1 *1e3;  //fluid density kg/m^3
V = 30*1e-3;  // entrance velocity m/sec

v = V * (R / r)^2;   // exit velocity - incompressible flow

Pb1 = 8 * mu * L * V / R^2; // viscous entrance pressure drop (Poiseulle-Hagen)
Pb2 = 8 * mu * l * v / r^2; // viscous exit pressure drop

Pp = 0.5 * rho * (v^2 - V^2); //kinetic pressure drop (Bernoulli)

Reynolds = Pp/(Pb1 + Pb2);
/*----------------------------

Reynolds = 6.7876e-07
*/

   The Reynolds number is miniscule.  The pressure drop from mechanism (2) (Bernoulli) is a million times smaller than from (1) (Poiseuille).  So model (2) is totally inappropriate - it misses an effect that is a million times larger!
  I used typical numbers for molten ABS with the flow compressing from a 1.75 mm diameter to 0.4 mm.
 I took the nozzle to be 2mm long. The feed length is 4 cm - but that doesn't influence the results.
  The model is geometrically crude - but improvement of the model would at most change things by  factors of a few.

paulusjacobus, 2016-06-15:
>I would like to point out that (2) is totally inappropriate

I think everyone agrees with this and the advance never really worked in
Marlin. SebastianV650 implemented the JNK version which works well. I have
been printing with it for a few weeks without any issues. I believe Scott
has rolled this into the RCbugfix.

thinkyhead, 2016-06-15:
Give the `LIN_ADVANCE` feature a look-over and a test, see if it looks good.

edwilliams16, 2016-06-15:
One thing I'm not clear about in the planner algorithm is whether after sweeping back and forth you are guaranteed to end up with a *speed* profile that is continuous. *velocity* jumps up the maximum jerk are OK.  
    The JKN algorithm is going to have problems if there are any *speed* jumps.

   The other issue that I think might matter is that molten plastic is not a Newtonian fluid - assumed by Poiseuille- Hagen.  In particular a [paper](http://www.cheme.utm.my/mpj/images/090402_3asif.pdf) found says molten ABS is a thixotropic fluid - one whose viscosity decreases with increasing shear (like paint).  I think the change to JKN would be relatively straightforward - but might not be necessary if JKN is *good enough* and would introduce one more parameter (the power of the power-law), which may not be known for all the materials we print with.

jbrazio, 2016-06-15:
> JKN is good enough and would introduce one more parameter (the power of the power-law of viscous stress vs. rate of strain)

@edwilliams16 this is a tunable parameter or a well known constant for a type of plastic ?

edwilliams16, 2016-06-16:
I don't really know.  It's been measured for ABS  - about 0.7 - and probably for other plastics, but would require some research to find the data.    I'm not quite sure how you would tune it just by looking at the print.
  What it controls is how much more viscous (and therefore requiring more advance) the plastic is at low feed rates than at higher rates.  It will also vary somewhat with temperature, but not as much as the overall multiplier does.  It will lie in a pretty narrow range.

  The users will probably need some help tuning the existing multiplier. For instance, it will vary as the filament diameter squared and between the inverse third or fourth power of the nozzle diameter (for a conical or cylindrical opening, respectively).  Any idea what was was used to come up with the current default?  We could then at least provide some clue how it might change with different materials, temperatures, nozzles, filament diameters etc.

jbrazio, 2016-06-16:
@edwilliams16 [this is the work done](https://drive.google.com/open?id=0B5UvosQgK3adaHVtdUI5OFR3VUU) by @Sebastianv650 which did the current implementation of `LIN_ADVANCE`, you may read the [discussion thread here](https://github.com/MarlinFirmware/MarlinDev/issues/402).

The previous exiting method is still broken:
```cpp
// extruder advance constant (s2/mm3)
//
// advance (steps) = STEPS_PER_CUBIC_MM_E * EXTRUDER_ADVANCE_K * cubic mm per second ^ 2
//
// Hooke's law says:    force = k * distance
// Bernoulli's principle says:  v ^ 2 / 2 + g . h + pressure / density = constant
// so: v ^ 2 is proportional to number of steps we advance the extruder
//#define ADVANCE

#if ENABLED(ADVANCE)
  #define EXTRUDER_ADVANCE_K .0
  #define D_FILAMENT 2.85
#endif
```

jbrazio, 2016-06-16:
The biggest problem for this type of features to go mainstream is the tunable factors.. as [most] people are not rocket scientist it's hard to understand what to tweak in order to get it working right.

thinkyhead, 2016-06-16:
@edwilliams16 Do you think JKN can be integrated in just the same manner as we've done `LIN_ADVANCE`? In other words, with hooks in all the same places, even though it may operate differently? Have you checked out where we've added hooks, and compared to how JKN wants to work? I can help to integrate it as (yet another) advance option alongside the others, and this will of course be easiest if it follows the same pattern.

edwilliams16, 2016-06-16:
I'm confused. What algorithms do ADVANCE and LIN_ADVANCE implement?  I thought LIN_ADVANCE was JKN, factoring out the explicit dependence on filament diameter from the K "constant".
    I agree with Sebastien's comment in his presentation that the extra steps during acceleration (and deficit during deceleration) need to be distributed uniformly in time.

 What about my question above:
"The JKN algorithm is going to have problems if there are any speed jumps."  Is speed continuous or not between trapezoids?  I think Kubicek also pointed out this problem somewhere.

thinkyhead, 2016-06-16:
> What algorithms do ADVANCE and LIN_ADVANCE implement?

Beats me. I have to defer to others on this topic. This is not an area where I have delved very deeply.

edwilliams16, 2016-06-16:
/**
 *         __________________________
 *        /|                        |\     _________________         ^
 *       / |                        | \   /|               |\        |
 *      /  |                        |  \ / |               | \       s
 *     /   |                        |   |  |               |  \      p
 *    /    |                        |   |  |               |   \     e
 *   +-----+------------------------+---+--+---------------+----+    e
 *   |               BLOCK 1            |      BLOCK 2          |    d
 *
 *                           time ----->
 *
 *  The trapezoid is the shape the speed curve over time. It starts at block->initial_rate, accelerates
 *  first block->accelerate_until step_events_completed, then keeps going at constant speed until
 *  step_events_completed reaches block->decelerate_after after which it decelerates until the trapezoid generator is reset.
 *  The slope of acceleration is calculated using v = u + at where t is the accumulated timer values of the steps so far.
 */

  This picture implies there is no speed jump at the block boundary - which is good...

edwilliams16, 2016-06-16:
The JKN algorithm increases the extruder step rate by an amount proportional to the step acceleration
- or equivalently during a period of acceleration, we introduce a number of extra steps proportional to the change in step rate over the period.
  Is that what LIN_ADVANCE is implementing?  It isn't easy to read.

Roxy-3D, 2016-06-16:
@edwilliams16   Put 'code' tags around that timing diagram and it will print out ok.  I was in the process of doing that for you when you deleted the post.    (4 ~ all together before, and the same after the code block)

Sebastianv650, 2016-06-16:
Hi @edwilliams16. As @jbrazio already pointed out, I strongly recommend you to read the development thread. It's long, but I don't want to write everything again here ;-) Next steps would be to understand the planner and stepper.cpp, work through the existing code from [JKN](https://github.com/jetty840/Sailfish-MightyBoardFirmware/blob/master/firmware/src/MightyBoard/Motherboard/StepperAccelPlanner.cc#L525) and then my Marlin pressure advance code.

In fact my implementation is a derivate of JKN (advance = somefactor * extrudervelocity), the most important change was the knowledge that the original JKN does something wrong in regards to the distribution of the advance steps, as I pointed out in the presentation linked by jbrazio.

But there are other very important things to keep in mind:
.) Even with a simple equation with one k factor, it's not easy to explain everybody how to tune it.
.) The code has to be very easy to calculate. Due to the fact that advance has to be calculated in the stepper ISR, even my short calculations are pushing the processor to it's limits. At least more than I want it to be..
.) Yes there are speed jumps between the moves. If you read through my old posts, you will find out that I learned that the hard way - my code seems to be underextruding at some points and it created blobs on other locations. Therefore it's important to understand how the planner works. The key word for this is the jerk setting.

In the end, it's much more easy to think about the pressure advance feate of a way to handle the free length of the filament between the hobbed bolt and the nozzle as a spring. You can start with reynolds number and other fancy math, but in the end it's a spring constant. And much more important: This explanation is understandable for most people!

edwilliams16, 2016-06-16:
> advance = somefactor * extrudervelocity

@Sebastianv650 
  Could you expand on what you mean by this?  Suppose the extrudervelocity were a constant - you are drawing a long straight uniform line.  No advance is required because you are in a steady state with constant pressure.  Or is your advance the cumulative advance from when the print started, so that only changes in advance get applied in a given step?

Sebastianv650, 2016-06-16:
One of the three inventors of JKN had the same concerns due to the sudden speed changes than you have - reality is sometimes not what the math says. In Marlin, nothing happens at infinite acceleration. you have always speed steps at time intervalls and you can use the time between two usual stepper ISR calls to distribute the extruder advace steps. I was doing that by seperating the extruder from the normal stepper ISR (the basics were already done by our old advance feature).

I don't have a link, but I believe there are quite good notes in the original grbl planner code? Here is an rough example what the planner does. Say we are starting from 0 to 20mm/s, then in the same direction we go to 50mm/s and than around a corner at 50mm/s:
1) Printer isn't moving at all.
2) Print move starts at jerk speed / 2. So if jerk is set to 10, the first move happens at 5mm/s.
3) The printer accelerates at the given rates until it reaches 20mm/s.
4) When it reaches the point where it should speed up to 50mm/s, it does a speed jump again. If I'm right, it now does a jump at full jerk setting, so the next speed is 20+10=30mm/s
5) It accelerates to 50mm/s
6) At the corner, it slows down to a value between jerk speed / 2 and jerk speed, depending on the corner angle
7) After the corner, it accelerates again to 50mm/s.

Regarding the k calibration, I will start the documentation how to do that on weekend.

advance = k * dv, sorry. Extruder velocity was wrong, of course the change is the important thing. For the code, I'm using somthing "cumulative" so I'm not loosing something due to the speed jumps. I'm always calculating (current speed - last speed), my first (not working) idea was to calculate the speed based on acceleration and time spent accelerting.

edwilliams16, 2016-06-16:
@Sebastianv650 
  Thank you - maybe I'll have another attempt at reading the grbl code.  I quit last time because I feared the Marlin code may be different enough for it to be misleading. 

>2) Print move starts at jerk speed / 2. So if jerk is set to 10, the first move happens at 5mm/s.

What's the rationale there?  Why not jump to full jerk speed?  That would meet the |delta_velocity| <= jerk constraint.  I guess the previous unmentioned move could have been a stop from the opposite direction.

  I must admit, I hadn't thought much about speed changes in a straight line.  Why would a slicer do that?    I know the firmware has to cover the possibility, however unlikely.  But is that a situation that arises in practice?  

  Assuming this all to be the case, the nice ASCII art in the comment is misleading.  The general case should show a speed discontinuity at the block boundary.

  So, it wasn't quite clear to me. Are you shoving the extra steps due to a boundary speed discontinuity as close as possible to the boundary (limited by max accelerations and feedrates) or are you spreading them out over the later acceleration phase (assuming there is one).  The former is closer to the physical model.

   If I can help you with the docs, let me know.

edwilliams16, 2016-06-16:
BTW, to expand on my comment about where to put the steps due to a speed discontinuity.
  If the trapezoid acceleration was determined by the DEFAULT_MAX_ACCELERATION(E) spreading them over the acceleration phase would be entirely appropriate. In practice, however, it is limited by DEFAULT_MAX_ACCELERATION(X,Y,Z) which are lower.

Sebastianv650, 2016-06-17:
The first post [here](https://ultimaker.com/en/community/6438-question-about-acceleration-jerk) explains the jerk calculation including the calculation. In a nutshell, jerk as implemented in Marlin is not exactly what it is according to physics. It's an "easy" way to limit the forces, and jerk speed / 2 is part of this definition. Ask grbl why they do it, I havn't the history in my mind ;)

Speed changes in one direction are not the usual case, but I wanted to have this in my example to show when full jerk jump is applied. In reality, Slic3r may generate something like that during gap fill. If you have a gap between two non-paralel lines, Slic3r splits the necessary variable width infill into small segments. If you use constant flow rate to define your print speed, the print speed will increase as the variable width infill decreases.

That's right, the ASCII chart isn't pointing out jerk things. But it's a small ASCII chart to show the generell look of a path with accel and deceleration, so I think it's OK..

At the moment, the pressure control isn't taking acceleration limits into account. This would need a complete rewrite of the planner, because then you need to precalculate all the steps to find the maximum real acceleration and if it brakes the limit, slow down the acceleration of the segment. At the moment this can't be done, also because the pressure control extruder speed is calculated after the planner work is done inside the stepper ISR.
Beside the complete rework that is needed, I guess the math would be too much on the current boards used at 16MHz..
Until now, nobody had problems due to this fact, most likely because an extruder is much faster then X and Y axis. But it is something that's not 100% clean, I know.

edwilliams16, 2016-06-17:
The first post indeed gives a nice worked example of cornering at constant speed.  The second post anticipated a follow-up question of mine.  Suppose you are printing a gentle curve while accelerating. At each vertex, your speed can jump by (almost) the jerk speed, while also accelerating along each segment. As you make the curve resolution finer and finer, you get more and more speed jumps, one at each vertex, circumventing the firmware's attempts to limit the jerk+acceleration of the hardware.
  It seems the answer is "don't do that!" as Marlin does not limits the rate at which jerks can occur.  Is that right?  

   I appreciate that the extra computational load of computing a new acceleration profile for the extruder to get rid of speed jumps in the original trapezoid may overwhelm the processor.  Just to be sure I understand what you are doing:

> you have always speed steps at time intervals and you can use the time between two usual stepper ISR calls to distribute the extruder advace steps. I was doing that by separating the extruder from the normal stepper ISR (the basics were already done by our old advance feature).

Where are the new steps distributed - all uniformly along the acceleration phase, or do you put those associated with any speed jump at the beginning?

There's not much danger of over-accelerating the E-axis, even if you ignore its limits, as they are so high.

thinkyhead, 2016-06-18:
> the online editor provides back quotes for "code", which didn't work here

Three backticks for multi-line code. Read the [Markdown Guide](https://guides.github.com/features/mastering-markdown/) linked at the bottom of the edit box.

Sebastianv650, 2016-06-18:
Yes many points in a small arc will fool the jerk limitation. It's easy to show when printing a 20x20mm tets cube and another one with a small 0.5mm radius containing enough points. The printer may run through the second one nearly without breaking down at the corners.

I distribute the extra steps between two normal stepper ISR time points. At point one, I know the current speed and the last speed of the extruder, calculating the needed advance steps. It's also known when the stepper ISR will fire again for the next step. I then fire the extruder ISR at time intervalls of availabletime_to_next_stepperISR / esteps_todo.

edwilliams16, 2016-06-18:
@thinkyhead 
> Three backticks for multi-line code.

 In fact  I did refer to the guide link. I only saw a reference to syntax highlighting.  I saw no sign of @Roxy-3D 's solution with four tildes. I was looking for the equivalent of a `<pre>` HTML tag - meaning don't touch this at all!

@Sebastianv650   OK.  I believe you are doing the right thing.  A suggestion by Kubicek, which I saw but can no longer find, suggested spreading the jerk related steps over the previous acceleration phase - easier to implement, but not true to the physical model.
  What prints do you use to test your model?  First thought I had was maybe a single outline hollow triangle built up vertically from the bed, printed in spiral (vase) mode.  Do you have some prints that are especially sensitive to advance settings?

thinkyhead, 2016-06-18:
> syntax highlighting … four tildes

Yeah, I don't know where this 4-tilde thing comes from – a general Markdown option I suppose. The equivalent of a `<pre>` is three backticks, and you can also specify the language that you want to highlight, such as three backticks followed by `diff` or `patch`:
```diff
- // Remove this comment
+ // Add this comment
```

Sebastianv650, 2016-06-19:
> What prints do you use to test your model?

I'm using a cube with the dimensions 25x25x2mm. The best settings for calibration are a low acceleration (minimize overshoots) combined with a high print speed. I'm using 500mm/s² and 70mm/s with 0.2mm layer height. Be sure your cooling settings are not slowing down the speeds and everything should have the high speed (perimeters, infill, top solid infill, ..)
Then, playing with M905 Kxy set various K values. Start with 0 (=off) for a comparison and increase in 20 or step with. The best value is the one where you have no bleeding edges (bleeding edges = K too low) and no inward rounded edges (K too high). You will also see and even more feel the difference in the quality of the top infill, where it changes print direction at the perimeters. But top infill alone is no sufficient calibration criterion as it is also influenced by the infill overlap % that's between 15-20% in most slicers.

For a Gregs wade extruder with PLA, values around 75 are good. With nGen, which is a much more flexible filament, the sweet spot is around 150.

Roxy-3D, 2016-06-19:
> Yeah, I don't know where this 4-tilde thing comes from – a general Markdown option I suppose. 

I went to find the Mark-Up page with that referenced on it.   I did not succeed.   I'm pretty sure that used to be a documented way to denote a <CODE> </CODE> block.  But it sure looks like it is no longer documented.    It does still seem to work???

bobc, 2016-06-19:
There seem to be a few different options in markdown depending on the version in use.

At least on github, tilde is an alternative to back-tick for code blocks, and is an exact equivalent, 3 or more start a code block. Both support the syntax attribute. 
So you can write ~~~cpp or  \`\`\`cpp  to start a code block, or just ~~~ or ``` for text block.

Back tick must be used for `inline quote`.

Oddly, the github gui editor doesn't have a button for code block, which is surely used a lot on github.

edwilliams16, 2016-06-19:
@Sebastianv650   What is your xy_jerk?

Sebastianv650, 2016-06-20:
My jerk value is 8, why?

edwilliams16, 2016-06-20:
@Sebastianv650   We had a lot of discussion about advance in the presence of jerk.

edwilliams16, 2016-06-21:
I did some testing.  I used a 30/60/90 triangle, 3 mm thick,  to provide some different corners.  I'm using Hatchbox yellow ABS at 230 C.  With k=0, the corners were over-extruded.  At k=80, the corners were rounded and the infill was no longer complete.  I got best results at 35-40.  I tried 200 mm/sec^2 and 600 mm/sec^2 accelerations and, as one might hope, the results were almost independent of acceleration.
  I don't think any of this really tests the case where the speed is discontinuous.  Maybe, I have to write/edit some G-code to get that to happen.
  Anyhow, LIN_ADVANCE certainly seems to be able to improve print quality.

  What happens when a printing move is followed by a travel move?  If one reduced the pressure to zero, it might reduce the need for retraction.

Sebastianv650, 2016-06-21:
In fact it does! With advance, i reduced my retraction length down to 0.5mm for PLA.

edwilliams16, 2016-06-21:
I'll think I'll play with PETG.  It's an oozy and elastic filament that may benefit from LIN_ADVANCE.  S3D had various advanced settings - coasting and wiping, that try do what proper pressure control should do.

Wurstnase, 2016-06-23:
Just for completeness a link from Willy and his idea of an advanced algorithm. http://www.dr-henschke.de/advance.html
It is in German, but it could be the same. (Maybe google translator can help) I don't read the hole thread, so sorry, if I missed something here.

Sebastianv650, 2016-06-23:
As far as I understand his point of view (mathematicaly, language isn't the problem), he comes to the same function I'm using. He writes the extra steps as a function depending on time and acceleration, I'm using extruder velocity. Both should be exchangable..

edwilliams16, 2016-06-23:
k = 80 works.  Ooze control may be better...

edwilliams16, 2016-06-23:
@Wurstnase 
   His result came to the same conclusion as mine from the first post in the thread.  It appears we are all on the same page.

Wurstnase, 2016-06-24:
@edwilliams16 That why I post it :+1:
Willy made this calculation 2 years ago.

edwilliams16, 2016-06-24:
I guess that's why they call it (re)search.

Wurstnase, 2016-06-24:
Hehe, agree. You could take a look into RepetierFirmware. It in there already. It is called there `Enable advance algorithm (not stable) (USE_ADVANCE)`.

Nandox7, 2016-06-24:
Does the implementation in Repetier is working?
I had the idea that similar to Marlin it was there but actually not very
functional.

-Nando

Wurstnase, 2016-06-25:
I've never tried it.

thinkyhead, 2016-06-25:
Marlin's `ADVANCE` might be more functional in the `RCBugFix` branch. At least one apparent bug in the algorithm was patched along the way. But it needs more scrutiny to see where it stands, and maybe it will be completely replaced by `LIN_ADVANCE` in future.

dteeter, 2016-11-15:
It seems that both the type of material and the distance between the extruder gear and the extruder output  would be significant factors. Would it be possible to create a simple formula for predicting K, given the distance (as described above) and the material?

Sebastianv650, 2016-11-15:
Have a look at LIN_ADVANCE in Configuration_adv.h. There you will find the formula for K, that should(*) be valid for the newer LIN_ADVANCE function.

(*) Should, as long as #5222 isn't fixed.

boelle, 2017-04-26:
Think we should close this one as we have it

or have i missed something?

even thou we close it it can still be discussed

Sebastianv650, 2017-04-26:
I think it can be closed.

closed 2017-04-26 by Roxy-3D