MarlinFirmware/Marlin #4050: Troubles with Capacitive Sensor due to the pullup resistor

kreexus, 2016-06-15:
Sorry for my English!

I was trying to connect my NPN Normal Opened Capacitive Sensor LJC18A3-H-Z/BX (with voltage divider, BROWN to +12, BLUE to GND, BLACK to D32 pin). Sensor is worked normaly without RAMPS (deactive is 4.5v, active is 0v), but after connecting to the RAMPS my sensor is not working correctly (deactive is 4.5v, active is 3v). This problem is due to the fact that the pullup resistor is constantly switched on all pins. I'm fix this problem after comment out line 1008 in file stepper.cpp:

`
#if ENABLED(ENDSTOPPULLUP_ZMIN_PROBE)
    //WRITE(Z_MIN_PROBE_PIN,HIGH);
#endif
`

After this, my sensor is work corretly and command M119 printed status open in deactive and triggered in active.

Next, the part of the code of my Configuration.h:

`
//===========================================================================
//============================== Endstop Settings ===========================
//===========================================================================

// @section homing

// Specify here all the endstop connectors that are connected to any endstop or probe.
// Almost all printers will be using one per axis. Probes will use one or more of the
// extra connectors. Leave undefined any used for non-endstop and non-probe purposes.
#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
//#define USE_ZMIN_PLUG
//#define USE_XMAX_PLUG
//#define USE_YMAX_PLUG
#define USE_ZMAX_PLUG

// coarse Endstop Settings
//#define ENDSTOPPULLUPS // Comment this out (using // at the start of the line) to disable the endstop pullup resistors

#if DISABLED(ENDSTOPPULLUPS)
  // fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
  #define ENDSTOPPULLUP_XMAX
  #define ENDSTOPPULLUP_YMAX
  #define ENDSTOPPULLUP_ZMAX
  //#define ENDSTOPPULLUP_XMIN
  //#define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  #define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
const bool X_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Y_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Z_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool X_MAX_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Y_MAX_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Z_MAX_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Z_MIN_PROBE_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.

//===========================================================================
//============================= Z Probe Options =============================
//===========================================================================

// Enable Z_MIN_PROBE_ENDSTOP to use _both_ a Z Probe and a Z-min-endstop on the same machine.
// With this option the Z_MIN_PROBE_PIN will only be used for probing, never for homing.
//
// *** PLEASE READ ALL INSTRUCTIONS BELOW FOR SAFETY! ***
//
// To continue using the Z-min-endstop for homing, be sure to disable Z_SAFE_HOMING.
// Example: To park the head outside the bed area when homing with G28.
//
// To use a separate Z probe, your board must define a Z_MIN_PROBE_PIN.
//
// For a servo-based Z probe, you must set up servo support below, including
// NUM_SERVOS, Z_ENDSTOP_SERVO_NR and SERVO_ENDSTOP_ANGLES.
//
// - RAMPS 1.3/1.4 boards may be able to use the 5V, GND, and Aux4->D32 pin.
// - Use 5V for powered (usu. inductive) sensors.
// - Otherwise connect:
//   - normally-closed switches to GND and D32.
//   - normally-open switches to 5V and D32.
//
// Normally-closed switches are advised and are the default.
//
// The Z_MIN_PROBE_PIN sets the Arduino pin to use. (See your board's pins file.)
// Since the RAMPS Aux4->D32 pin maps directly to the Arduino D32 pin, D32 is the
// default pin for all RAMPS-based boards. Some other boards map differently.
// To set or change the pin for your board, edit the appropriate pins_XXXXX.h file.
//
// WARNING:
// Setting the wrong pin may have unexpected and potentially disastrous consequences.
// Use with caution and do your homework.
//
#define Z_MIN_PROBE_ENDSTOP

// Enable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN to use the Z_MIN_PIN for your Z_MIN_PROBE.
// The Z_MIN_PIN will then be used for both Z-homing and probing.
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

// To use a probe you must enable one of the two options above!

// This option disables the use of the Z_MIN_PROBE_PIN
// To enable the Z probe pin but disable its use, uncomment the line below. This only affects a
// Z probe switch if you have a separate Z min endstop also and have activated Z_MIN_PROBE_ENDSTOP above.
// If you're using the Z MIN endstop connector for your Z probe, this has no effect.
//#define DISABLE_Z_MIN_PROBE_ENDSTOP
`

This is Bug?

Blue-Marlin, 2016-06-15:
How about
```diff
//#define ENDSTOPPULLUP_YMIN
//#define ENDSTOPPULLUP_ZMIN
- #define ENDSTOPPULLUP_ZMIN_PROBE
+ //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
const bool X_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
```

chuck787, 2016-06-15:
My implementation was to put a resistor and 5v zener to assure I don't blow the ramps board.  Just to be on the safe side I added a shorting push button in case the sensor allowed the extruder to hit the glass even though my designs should detect at couple millimeters.high.  Saved me a few times.  In pix yellow goes to sensor, red to Ramps and black to ground.
[image]

jbrazio, 2016-07-16:
Closing due to inactivity.
Please re-open this issue if still valid.

closed 2016-07-16 by jbrazio

DeathCookies, 2016-10-01:
I have the same problem (activated 4,74 / deactivated 3,07V).
The solution of the first post does not work...
Currently i am testing it as a normal endstop but it will never trigger. Does anybody have an idea?

hsaturn, 2017-06-19:
Hi

I've bought this thing : LJC18A3-B-Z/BX (NPN)

This is a NPN output, so there is absolutely no voltage that can goes out !

The BLACK output can drives 300mA to the ground whatever it comes from (if grounds are same).

So, divider, diode etc is NOT REQUIRED AT ALL.

And .. this is a lot safer than adding external parts such as diodes and resistor, since when you do that, you ARE wiring the board input to an 12v external source !
So if the diode has a failure, of is mounted in the wrong side, or if any metal part (screw) makes a shorcut etc etc, your board is in trouble.

What I've did is 
-----------------
1 / Connected the BLACK to the old BLACK of my SN04-N
2/ Connect the BLUE to the old BLUE
3/ Connect the BROWN to +12V

Then I've connected this to the printer and checked with M119 if all went ok and that was the case.

After that, just to be sure, I've controlled the voltage, it was perfect, varying from 0 to 4.9v.

*/ Software modification : NONE, this could have worked with the original firmware !
I've let the setting exactly the same as for the SN04-N

(Marlin 1.1.3, Tevo Configuration.h base)

#define Z_MIN_ENDSTOP_INVERTING true
//#define ENDSTOPPULLUPS

//#define ENDSTOPPULLUP_ZMIN
#define ENDSTOPPULLUP_ZMIN_PROBE

I'm pretty sure that your sensor has the same kind of output (NPN) and thus needs also absolutely no external part to work.

You can easilly verify that by

1/ unconnect the black wire of the sensor
2/ Power on the printer
3/ Check that in both cases, you have 0V (sensor near of far from bed

And if still not sure (but only if STILL not sure)
3/ Connect a 10k resistor from black wire to 5v
4/ You should have now either 0 or 5v no more, even if the sensor is connected on a 36v power source.

Remark, the 10K resistor connected in step 3 for testing is .... the pullup resistor of the Mega2560 !!!
So this is exactly why you must not disable pullups.

This is how NPN ouputs are working :-D

This kind of output does not output any volt in fact !
(See open collector ouput for more information).

Regards

Bob-the-Kuhn, 2017-06-19:
Did you take your voltage measurements with the black wire connected to your controller?  If yes I suggest doing the measurement with the black wire disconnected.

I also have a LJC18A3-B-Z/BX.  On mine there is a pullup resistor between the brown & the black wire so when NOT triggered the black wire is at the same voltage as the brown wire.

I've only been able to find a couple of data sheets that show the output stage of a capacitive or inductive proximity sensor.  Both showed the pull up between brown & black.

[image]

It would be nice to steer users to a sensor that doesn't have that pullup resistor.

What is the name of the manufacturer of your sensor?

hsaturn, 2017-06-20:
Hi

I did my voltage measurements with both the black wire connected and
disconnected.

When disconnected, I had 0v in any case.

... 15mins later...

But, now you are asking, I'm confused. I've read some things that says that
there is a pullup resistor (which I found totally useless or even hassle in
a NPN output for a sensor that admit a wide input voltage range...).

I'm gonna re-check this evening. I'm sure that all voltages were perfect.
0v or a litlle less than 4.9v.

Also, before that, I made a 10k / 10k divider (ok, these are not the
correct values). And output was varying from 2.2v to 3.3v  (...) { values
may be wrong, this was 2 days ago, and I saw thousand of other  numbers
since :-D

So I will answer you this evening with more information.

I had my sensor on ebay and do not know the manufacturer.

Chines is written on it...

If I remember well, I made a test with about ~30v just to check for better
detection (which was not the case). And still 0v in output.

As I have absolutely not any external part connected I do not have to
unsolder anything so I will unplug the sensor and verify all that this
evening.

Probably like you I've searched for this sensor datasheet without success.
So we are unable to precisely know what is real.

I will suggest to others not to trust anything in fact.

Maybe the right solution is to

1/ Make some measurement of what voltage we have with output unconnected
1.1/ If 0v in all case, this is my case, no need for any external part.

1.2/ In case we have some volts there :
2/ Make another one with let say a 10k resistor
3/ Input this measurment in a math formula.

Measuring the input pullup resistor that way allow to add just one external
resistor to have exactly 5v.
And mount a ~5v zener also (just in case resistor is hs or bad soldered
etc...).

What do you think about that ?

Maybe my part has a problem in fact. Maybe the 10k pullup resistor has been
omitted or bad soldered or component was bad ???

You have Vcc on your sensor... So the truth is somewhere... but where :-D

What is certain is that my tevo tarentula works just find since two days
with no problem and the sensor connected to 12v. And in two days, I guess
I've switched it off during at max 8 hours :-D

Best regards, see you soon :-D

Francois Biot

hsaturn, 2017-06-20:
Hi again

In short :

**!!!! PLEASE DON'T DO THE SAME AS ME !!!!**

I suggest you to connect the blue to GND, the brown to +12V and measure the output voltage on black output when the sensor is on the bed then far from it.

Should be very cool to post here results....
If you have 0v in both cases, you are in my situation.
If not, you are in bob's situation.

---------------------
In long

1/ Bob says that his sensor have an internal pullup.
2/ I say that this is an open collector output....

Both of us are aware of electronic engeenering and what to do but in that case, I'm confused.

What is sure is
- The sensor works fine the way I've connected.
- Bob's wont work the way I've connected it (and *will* burn the board soon or ... sooner)

We both are not sure, just because the datasheet of the sensor is not precise enough.

Bob has a 10k internal pullup in his sensor. I've do not have any !!! And this is the same part (LJC18A3-B-Z/BX).

Maybe my sensor has a broken resistor or maybe it has been bad soldered or a bad contact.
My tevo works fine like that but who know, if this is a bad contact, my board will burn.

Once at home this evening I'll will take more time to make some more voltage measurement.
In my memory, even at 30v on the brown wire, I had 0v on black one (which is normal in pure NPN open collector output).

Bob, I've found also this 'NPN OUTPUT TYPE' picture you have posted but not related directly to the LJC18A3-B-Z/BX.

My small history :

I've read a lot before mounting my sensor, and tried with resistor. As this was not working at all, I made some measurements and find that solution which was an evidence for me (believing that this was an open collector output).
The solution could came an evidence if we had the datasheet of this sensor (with output stage schematic)....

So I stay confused...

Bob-the-Kuhn, 2017-06-20:
There's multiple manufacturers of that type of sensor.  All it means is that one manufacturer builds it a little different than the other.

When buying on eBay you don't know which manufacturer built the part.  That means we need to measure the output to determine the type of interface circuit to use.

An alternative is to always put a 3.9V 10% zener across the signal & ground and enable the internal pull up.

hsaturn, 2017-06-20:
Hi again, last part of the story

Here are some pictures of my sensor

[image]

[image]

The last one is with the sticker removed so I could take a good photo of all what is written on it.

I'd like to know if you have the same model.

I'm gonna now to make some measurements.

See you

hsaturn, 2017-06-20:
Hehe

Back with some more information, the chinese traduction.

I like the chinese writing, a lot more condensed than others.

I've never saw an electronic part with the function written on it : Proximity switch !!!

Any chinese not aware of electronic can say what is this part made for.
Try to ask to any who speaks only english :-D

[image]

Bob-the-Kuhn, 2017-06-21:
Mine are from a different vendor.

rangua, 2017-12-06:
Hi. 
I have the exact same sensor as hsaturn, but measuring the voltage between the blue and the black wire (brown to +12V, blue to GND) i get 12V when far away from the bed and 0 when i'm close. 

I'm not sure what to do. Should i make a voltage divider to get 5v on the black wire when the bed is far?

Thanks

rangua, 2017-12-06:
or, maybe.. connect it to 5Vs instead (the sensor says it works with 6-36VDC.. maybe it works with 5v?

AnHardt, 2017-12-07:
With a voltage divider it will work.
Using the sensor with too low voltage may work, but range and or noise will suffer.

Solderfinger, 2019-02-13:
Thanks hsaturn for your translation of the sensors label.
What i didn't understand before now makes perfekt sense.
What you translated as "load" was to me an obscure switching mechanism inside the sensor
when in reality it is i.e. a relay outside the sensor.
Now it's perfectly normal for the black wire to carry the same voltage as the brown one
because with both lines on the same potential the load is off!
If the sensor nears a sensable object black goes to ground to activate the load. That way you
have to deal with "high voltage" (exceeding 5v, that is) on the printers controller board which
will fry it sooner or later.
To prevent that a diode in reverse (cathode to the board) should do the trick. It blocks 12v from
damaging the z-switch input but allows pulling the pin to ground when the sensor is active.

RSAGeek, 2019-09-17:
> Thanks hsaturn for your translation of the sensors label.
> What i didn't understand before now makes perfekt sense.
> What you translated as "load" was to me an obscure switching mechanism inside the [...]

HI

So what diode would you suggest Solderfinger?