MarlinFirmware/Marlin #4051: Feature request - Bowden Filament loading & Guided Levelling

AxMod3DPrint, 2016-06-15:
I'm no programmer, else I would have added this in on my fork. Here's a Couple things I'd like to see within Marlin, if this has been asked for before, forgive me, I've not been keeping track. 

1) LCD filament load option for Bowden systems, with Bowden Tube length set in Configuration.h, I don't know if it'd  be possible to piggy back that off the new M600?
2) An interactive guided levelling routine, fired off by the LCD or M Code. Something along the lines of Mesh levelling but allows the user to pick points to go to in configs, manually level the bed then move to the next point. I know this is in different forks of the Marlin firmware, such as the Ultimaker firmware, but I'd like to see this brought into the main branch of Marlin. 

This has been posted for discussion, I fully expect a no, not going to happen and if the idea does get used I fully expect it not be be in for a while. Just a couple of nice-to-haves over being really necessary.

thinkyhead, 2016-06-16:
Unified Bed Leveling is in development, but will not be in 1.1.0. It will have to wait for 1.1.1. We may have better `M600` code in place by 1.1.0, but it needs work.

AxMod3DPrint, 2016-06-16:
@thinkyhead - so the Unified may include a guided system for manual 3 and 4 point bed levelling?
 As for the M600, I've been keeping track, however, the filament loading for Bowden systems is a separate idea and would need to be available on the LCD when not printing and make sure the nozzle is heated before loading. I just suggested it may be able to be piggy-backed off of M600 as a possible way to implement it.

thinkyhead, 2016-06-17:
> the Unified may include a guided system for manual 3 and 4 point bed levelling

The idea is to allow manual or automatic leveling regardless of the type. In other words, you get "manual probing" when you don't have a real probe.

> I just suggested it may be able to be piggy-backed off of M600 as a possible way to implement it.

The updated `M600` that I referred to could include more considerations for Bowden. Please comment and participate at PR #3662.

AxMod3DPrint, 2016-06-17:
@thinkyhead - thanks, for the clarification. I'll go take a look at #3662

closed 2016-07-16 by jbrazio