MarlinFirmware/Marlin #4059: Filament runout sensor

maukcc, 2016-06-16:
Here are a couple of things that just not seem right with the filament runout sensor:
* when no FILRUNOUT_PIN is assigned you get no error
* const bool FIL_RUNOUT_INVERTING = true;  and should be false by default
* when booting the machine you get "printer ready" which is not the case when no filament is loaded

feature request:
when filament runout is detected print goes on so long as extruder has extruded XXmm and then stops.

373996214, 2016-06-20:
These days, I am playing around with it. What I want to do is to use filament runout sensor to detect the filament: when there`s no filament, stop; or, go on.
I changed some codes, but failed. /_\*
Anyway, go on working. *_*

thinkyhead, 2016-06-20:
#4095 added a sanity check to make sure you have the pin defined. #4097 fixes `FIL_RUNOUT_INVERTING`.

Roxy-3D, 2017-02-16:
> These days, I am playing around with it. What I want to do is to use filament runout sensor to detect the filament: when there`s no filament, stop; or, go on.

Fully functional and tested...   And...  It includes a time out to power down the nozzles if you are not around when the filament runs out.  (And of course...  A heat up phase to get things back to normal from a cold state.)

thinkyhead, 2017-02-18:
Merged the latest M600 code — a pretty stable and straightforward menu-driven procedure.

boelle, 2017-04-26:
@thinkyhead so this one is done? should we not close it?

pwave1, 2017-07-16:
Sorry A bit of topic, I have been getting false positives might be a faulty endstop. Any way to add a delay in the software so I could try to avoid them till I get a new one.
 Thanks

Roxy-3D, 2017-07-16:
It might be electrical noise.  You could try running the wires further away from high current wires that are switching power (stepper wires and heater wires).   But if that isn't productive, you might want to just switch the end stop with one of the axis end stops.     You can also try putting a 100 uf capacitor across the switch so any noise gets filtered electrically.

It can be de-bounced in software.  But that is going to be more work (and mess up the code!) than just making the filament sensor work right.

closed 2018-02-06 by thinkyhead

IamGaryGnu, 2020-07-14:
Hi - i know this closed 2.5 years ago.  I"m hoping @Roxy-3D or some other kind soul may help - I have a CR-10 S5 where I believe the endstops were picking up too much noise with extension cables - so i soldered 104 capacitors across them.  Now I'm having filament sensor false positives - @Roxy-3D suggested the 100uf capacitor across - as there are 3 wires versus 2 for the end stops, can or should I use 2 capacitors - on from the center (pin 2) to say, pin 1, and and one from pin 2 to pin 3 - or is there a better layout?  many thanks!!!!  (i'd also love Marlin to allow a longer delay for the filament run out - maybe a few seconds straight which may help out against false positives - many thanks!!!!!