MarlinFirmware/Marlin #4061: Set MANUAL_Z_HOME_POS for deltas with EEPROM

superlou, 2016-06-16:
I've had a lot of luck using the Escher 3D wizard for a delta printer to calibrate endstop corrections and rod lengths.  The output of the wizard looks like the following:

```
M665 R172.11 L355.00
M666 X-0.00 Y0.00 Z-0.34
; Set homed height 299.88mm in config.h
```

The first two commands can be entered easily over a terminal and saved via M500, but the last one requires opening up the IDE, recompiling, and uploading to the controller.  It's not unfeasible, but a little annoying.  Is there already a way to specify this parameter and save/load it from EEPROM?  If so, I'll reach out to Escher about updating the wizard.  If not, could this become a feature down the road?

LVD-AC, 2017-03-31:
This can be easily achieved by changing home_offset[Z_AXIS];

the delta height used in calculations= MANUAL_Z_HOME_POS DELTA_HEIGHT + home_offset[Z_AXIS]

home_offset[X_AXIS and Y_AXIS] have no meaning on delta machines, so it could be a good idea to disable M206 and M428 on deltas.

If you want it a bit more structured:

In configuration.h
... add in delta settings:
#define DELTA_HEIGHT 270 //mm
... and change:
#define MANUAL_Z_HOME_POS DELTA_HEIGHT

In marlin_main.cpp:
... after:
  inline void gcode_M665() {
... add:
    if (code_seen('H')) home_offset[Z_AXIS] = code_value_linear_units()- DELTA_HEIGHT;

And in configuration.store.cpp:
... change:
        SERIAL_ECHOLNPGM("Delta settings: L=diagonal_rod, R=radius, H=height, ...
... and after:
      SERIAL_ECHOPAIR(" R", delta_radius);
... add:
      SERIAL_ECHOPAIR(" H", DELTA_HEIGHT + home_offset[Z_AXIS]);

done (#4022)

boelle, 2017-04-26:
@thinkyhead can we close this one since a PR has been submitted?

LVD-AC, 2017-06-08:
May be closed since this feature was incorporated in marlin 1.1.0

closed 2018-02-06 by thinkyhead