MarlinFirmware/Marlin #4067: [Request] Lock out Advanced Features From Display

Grogyan, 2016-06-17:
Something that would make Makerspace/School printers more useful, would be to have a limited number of features accessible from a smart display, such as the Full Graphic Display.
Leaving the Advanced features locked out unless the user knows of the correct sequence of buttons/knobs/turns on the display.

This would prevent novice/inexperienced persons from meddling with features that can, and do, muck up the printer

jbrazio, 2016-07-11:
In fact I believe the menu system needs to be rewritten for no technical guys..
A proper designed UI needs to have [at least] the most used features on top of the menu.. it makes no sense to move an axis to dig down four levels deep in the menu.. but once again, not for this RC.

Roxy-3D, 2017-02-16:
> In fact I believe the menu system needs to be rewritten for no technical guys..
A proper designed UI needs to have [at least] the most used features on top of the menu.

Agreed!

boelle, 2017-04-26:
@thinkyhead 

maybe add this to 1.1.1 milestone?

landodragon141, 2018-11-08:
@thinkyhead This is probably obsolete.

closed 2018-11-17 by thinkyhead