MarlinFirmware/Marlin #4068: using I2c of RAMPS 1.4

pankajbadatia, 2016-06-17:
Hi all  I want to know how can i connect RAMPS as masster and other arduino as slave using I2c communication.
which code should i edit . 
my slave address is 0x08 
start bit 0x01
command bit 0x02
stop bit 0x17

thinkyhead, 2016-06-17:
With #3258 we added an i2c bus feature. Enable `EXPERIMENTAL_I2CBUS` in `Configuration_adv.h` and you can use it to experiment with sending and receiving bytes using GCode commands.

closed 2016-06-20 by jbrazio

pankajbadatia, 2016-06-20:
Thank you .. I did it but i have confusion between sending hex value and byte or char

jbrazio, 2016-06-20:
What is the confusion ?

pankajbadatia, 2016-06-20:
address is in hex  when 
M155 A8 // slave address
M155 B1 // start bit
M155 B2 // command
M155 B23 // stop bit
M155 S1 // send 

it doesn't work

pankajbadatia, 2016-06-20:
* ; Example #1
 * ; This macro send the string "Marlin" to the slave device with address 0x63
 * ; It uses multiple M155 commands with one B<base 10> arg
 * M155 A63  ; Target slave address
 * M155 B77  ; M
 * M155 B97  ; a
 * M155 B114 ; r
 * M155 B108 ; l
 * M155 B105 ; i
 * M155 B110 ; n
 * M155 S1   ; Send the current buffer

I am reffering this

jbrazio, 2016-06-20:
> It uses multiple M155 commands with one B<base 10> arg

`M155 B77` the 77 is ASCII "M" in decimal (base 10).
In your case you only need to convert `M155 B23` to `M155 B35`.

[image]

thinkyhead, 2016-06-20:
> confusion between sending hex value and byte or char
> you only need to convert M155 B23 (base 16) to M155 B35

Actually, it looks like you did the right conversions. 0x17 is 23 decimal.
What result do you expect when you send

jbrazio, 2016-06-20:
Scott, that's the example from the `Configuration_adv.h`.

pankajbadatia, 2016-06-21:
Does the ramp accept address in hex or decimal ??

pankajbadatia, 2016-06-21:
/* Start and end of packet markers These are 
start of packet SOP 
 and end of packet EOP 
of different microprocessor which i am using*/

#define PACKET_SOP      (0x01u)
#define PACKET_EOP      (0x17u)

/*  any one can be used Commands set */
#define CMD_SET_OFF     (0u)
#define CMD_SET_RED     (1u)
#define CMD_SET_GREEN   (2u)
#define CMD_SET_BLUE    (3u)

Then which commands should i send

thinkyhead, 2016-06-21:
> Scott, that's the example from the Configuration_adv.h.

And it's wrong, alas.

thinkyhead, 2016-06-22:
> Does the ramp accept address in hex or decimal?

The `M155` feature needs the address (`A`) to be specified in decimal — only digits from 0-9 are acceptable in GCode parameters.

thinkyhead, 2016-06-22:
> Then which commands should i send ?

I'm just guessing here, but it looks like this is the idea…
```gcode
M155 A8 ; slave addr
M155 B1 ; PACKET_SOP
M155 B3 ; blue
M155 B23 ; PACKET_EOP
M155 S1 ; <send>
```