MarlinFirmware/Marlin #4075: Printer randomly Hangs print from SD

RicardoGA, 2016-06-18:
Hi the problem still happen with the latest rcbugfix, this is a new report of the problem #3981 

Also i notice when the printer hang the led 1 (ramps 1.4) is blinking and even if i reset the ramps (using reset button) the printer don't respond, i have to do a power cicle (turn off/on) to unfreeze the printer 

i attach a link for a video tath shows the led blinking and that the perinter is still freeze after i press the reset button

@thinkyhead 

https://dl.dropboxusercontent.com/u/55745122/3d%20Print/IMG_3300.MOV

RicardoGA, 2016-06-18:
another freeze just happened, here's the log of repetier while i was printing from sd
[repetier.txt](https://github.com/MarlinFirmware/Marlin/files/321567/repetier.txt)

thinkyhead, 2016-06-18:
Whew! Well, how annoying. Bugs like this can be so hard to track down. One thing we definitely have to be cautious about is instances where a wait-loop calls `idle()` and then some other code calls `idle()` from within an `idle()` call… I can imagine some hard-to-find stack overflow bugs if we're not careful about its usage.

fbarcenas, 2016-06-18:
I think than all this problems is with the buffer

fbarcenas, 2016-06-18:
I do some test with arduino 1.6.0 and seem better than 1.6.8 and 1.6.9.
Maybe this problems is of compiler, not marlin

thinkyhead, 2016-06-18:
> Maybe this problems is of compiler, not marlin

@fbarcenas It's always possible. We've seen issues like this before. Sometimes just moving a few lines of code around is enough to "fix" it. I'd say keep trying `RCBugFix` each day, as we are always fixing things and making adjustments.

Roxy-3D, 2016-06-18:
Here is a similar thread over at RepRap.Org:   http://forums.reprap.org/read.php?415,637016

In theory, SD-Card printing should be more reliable than USB/Serial printing.

Blue-Marlin, 2016-06-18:
Dropped digits?
G0 X23.456 instead of 123.456 could look like homing.
G0 Z23.456 instead of 123.456 could look like unloading filament.

Interrupted SPI transfer?

thinkyhead, 2016-06-18:
> Interrupted SPI transfer?

Interrupts have been pretty well-behaved up to now. Is there a way we can count cycles used by the ISRs and see how much time they are taking up?

RicardoGA, 2016-06-19:
Today i tried the latest rcbugbix to test with arduino ide 1.6.0 and i made a succesfull print and with other the printer hang, it seems that the hangs occurs with longer prints, i don't have any hang with short prints (less than an hour), i'm trying to find a way to make the problem happen in a more predictable way in order that we can replicate the problem

AnHardt, 2016-06-19:
Up to now we do not evaluate the errors the sd-library can find. We take an error as a line end.

Debugging could be a bit easier if we'd know if the error is in reading from sd, or later, when we process the line, or corrupt the buffer, or ... .

As a debug helper i propose:
```cpp
@@ -1129,6 +1129,11 @@ inline void get_serial_commands() {
            || sd_char == '\n' || sd_char == '\r'
            || ((sd_char == '#' || sd_char == ':') && !sd_comment_mode)
        ) {
 +        if (n == -1 & !card_eof) {
 +          SERIAL_ERROR_START;
 +          SERIAL_ECHOPGM("SD read error: ");
 +        }
          if (card_eof) {
            SERIAL_PROTOCOLLNPGM(MSG_FILE_PRINTED);
            print_job_timer.stop();
```

`M111 S7` will help to get a useful log.

thinkyhead, 2016-06-20:
I've added an error message for SD card reading issues, so if that's a culprit, now it will be more obvious.

RicardoGA, 2016-06-20:
thanks @thinkyhead @AnHardt i saw this change yesterday  and i made a couple prints with the new error message but the problem don't show, today i'm going to make more print to check if the pronlem happen.

Thanks for the help everybody

RicardoGA, 2016-06-20:
@thinkyhead @AnHardt  the printer freeze again without showing any error via serial, here's the repetier log 
[repetier Log2.txt](https://github.com/MarlinFirmware/Marlin/files/324145/repetier.Log2.txt)

AnHardt, 2016-06-20:
Sorry for me the log is worthless without __M111 S7__. I need the echos to see what is going on.

fbarcenas, 2016-06-20:
What is the name of the file than you print?

RicardoGA, 2016-06-20:
@AnHardt Sorry i forgot to add m111 s7 to my gcode, i just add it and when it hang again i´ll upload the log. 
@fbarcenas the name of the file is Prusa_pata.gcode

fbarcenas, 2016-06-20:
please you first test with this name "Prusa_pata.gcode" and M111 S7,
after test with name "test.gcode"
Not mayus, not "_". Simply and easy name

RicardoGA, 2016-06-21:
@fbarcenas sorry for late response, i'm naminig my files now to test.gcode (to discard a problem with the name ) the brand o sd card is a kingston sd 8gb and kingston sd 4gb both with less tan 100mb in use and with both cards the printer hang randomly, the software i use to slice is simplify3d

fbarcenas, 2016-06-21:
thanks, we will wait the error with M111 S7.

Could after test with a SDCARD brand Transcend.
You don't answer me a question, what is the name of the partition of the SD card?

RicardoGA, 2016-06-24:
@fbarcenas sorry for the late response i was busy, here´s a capture and log of the printer using m111 s7 with simplify (only let me save the last 999 lines)  because with repetier the printer lost communication, the log don't show any error,   the partition of the sd don´t have name in my pc just have the letter E:
[simplifylog.txt](https://github.com/MarlinFirmware/Marlin/files/332707/simplifylog.txt)
[image]

thinkyhead, 2016-06-25:
>  the partition of the sd don´t have name in my pc just have the letter E:

The card might actually have a name. Examine its Properties in the file explorer.

thinkyhead, 2016-06-25:
> with repetier the printer lost communication

Not a very useful log. It lost communication, and then the print continued for a while? And then the printer hung later?

RicardoGA, 2016-06-25:
@thinkyhead no the card don´t have name, and the lost communication with repetier i dont think its related with the hangs because the printer lost communication and the print continued and sometimes it complete without problem
[image]

Blue-Marlin, 2016-06-25:
Does your computer go into standby?

RicardoGA, 2016-06-25:
@Blue-Marlin No the power options of my computer are set to never go to sleep and never turn off the display

closed 2016-07-23 by jbrazio