MarlinFirmware/Marlin #4079: Safe Home travel speed slow

SimonGolding, 2016-06-18:
I have just used a new copy of marlin 1.1.0 RC6  running on a Prusa i3 and found an issue with safe homing and auto levelling

Basically it would home X and Y at the correct speed then move to the middle of the bed at the speed for the Z axis which was really slow. This also affect G29 auto level (Im using an inductive probe) making it very slow.

I have made two changes to the code that have corrected the issue Marlin_main.cpp

Line 1646: feedrate = xy_travel_speed; 
changed too feedrate = XY_TRAVEL_SPEED;

Not sure if this has any affect infact it looks like the var xy_travel_speed is no longer used once replaced with this constant.

I also modified line 1537 of marlin_main.cpp by adding 
float oldFeedRate = feedrate;
at the beginning of the function then restored the feedrate at the end of the function.

Thanks
Simon

thinkyhead, 2016-06-18:
The saving and restoring of the feed rate makes sense, so I've added that to a few functions in PR #4080.

As I look at the code, I see that `xy_travel_speed` is initialized to `XY_TRAVEL_SPEED` but it is meant to be overridden by `G29 Snn` (for Automatic Bed Leveling). If you leave off the `S` parameter, it is set to `XY_TRAVEL_SPEED`. So it should be fine to leave `feedrate = xy_travel_speed` in place.

jbrazio, 2016-07-18:
@thinkyhead can we consider this one patched so we can close the issue ?

closed 2016-07-18 by SimonGolding