MarlinFirmware/Marlin #4083: Feature Request - Info screen icons & babystepping default

adamfilip, 2016-06-18:
A few ideas
1) It would be great if the info screen had a simple icon or way to indicate that mesh leveling was enabled.

2) also when on the idle info screen, if you could set the rotary encoder by default to adjust babystepping Z instead of adjusting speed by default. (be able to set the default function of the rotary)

Roxy-3D, 2017-02-16:
> also when on the idle info screen, if you could set the rotary encoder by default to adjust babystepping Z instead of adjusting speed by default. (be able to set the default function of the rotary)

Its not merged into RCBugFix...  But it is just a few lines of code and it has been tested fully.   A double click of the Encoder Wheel while at the status screen drops you into Z-BabyStepping.   I use it every print.

boelle, 2017-04-26:
this one should be added to milestone 1.1.1 even if its just for the icon idea

thinkyhead, 2017-11-01:
1. An option for that has been added
2. We now have an option to double-tap (or hold) to go to babystepping

(I note that item 2 currently has a bug where it may go to babystepping instead of the info screen after a print, or after a long time in sub-menus. Should have that fixed soon.)

closed 2017-11-01 by thinkyhead