MarlinFirmware/Marlin #4087: LIN advance randomly produces bad moves to print bed boundaries

chrisqwertz, 2016-06-19:
Using LIN advance randomly leads to the extruder moving in either x or y direction without extrusion  until it hits xy-min/max position, then reverses direction and moves. This happens at random and repeatedly throughout the print.

This is my first post on GitHub. How can I help track this bug down?

I'm using a Mendel90 with RAMPS1.4 and the latest rc_bugfix version.

Sebastianv650, 2016-06-19:
This happens due to problems in the serial communication. Enabling this feature increases the load to the processor which can result in missing received chars. Which print host software are you using, can you see error messages like wrong checksum? For Simplify3D, ensure you have enabled printing with line numbers and checksums. If you are using 250000 baud in Marlin, go back to 115200 baud.

This is a known problem, see https://github.com/MarlinFirmware/Marlin/issues/3680.

chrisqwertz, 2016-06-19:
Thank you for clarifying. Im using S3D @ 250kbaud, until now without ECC.
Will switch on line numbers and checksums.
Thanks again!

closed 2016-06-19 by chrisqwertz