MarlinFirmware/Marlin #4088: mbl with z_max endstops not usable

chrisqwertz, 2016-06-19:
When using mesh bed leveling thorugh the LCD menu on my machine with z-max endstops, the leveling prcess will start at z_max height thus making it very tedious to move all the way down with the encoder.
I tried using MESH_G28_REST_ORIGIN with no success.
Am I missing something?

epatel, 2016-06-19:
@chrisqwertz No you are not missing anything. This is one of the things I have been wanting to add to MBL. If you want to be my tester I can see if I can add this.

chrisqwertz, 2016-06-19:
My spare time is a little rare, but yes sure, if I can help, I'm in.

Roxy-3D, 2016-06-19:
Just as an FYI:    Please consider doing your work and testing using the very latest RCBugFix.   ThinkyHead has started cleaning up a lot of the probe stuff.   It is very possible that it will be much easier to do the work there.   (And even if it isn't easier...   We ultimately need to get your work merged in with what will be the 'current code' anyway.)

epatel, 2016-06-25:
@chrisqwertz Hi, I have this made this commit https://github.com/epatel/Marlin/tree/epatel/max_z_endstop
Please test to see if it works for you.

Note about usage. Set `Z_MAX_POS` so that it is slightly less than the full length from the Z endstop to the bed. The bed should be within the `MESH_HOME_SEARCH_Z` distance, default is 4mm. When starting the probe sequence it should first make a HOME, going up to the Z endstop. Then go down the full `Z_MAX_POS` length, temporarily set position Z=4 (the search distance) and one should be able to use the "encoder" on the display to slowly lower the hotend to the bed (use a paper to feel when it is near). Press the "encoder" when you think its fine. Repeat until you have probed all mesh points.

thinkyhead, 2016-06-26:
@epatel I'm not sure how this affects your commit, but a reminder that after homing to `Z_MIN_POS` the `current_position[Z_AXIS]` will be `Z_MIN_POS + home_offset[Z_AXIS]`, and when homed to `Z_MAX_POS` it will be `Z_MAX_POS + home_offset[Z_AXIS]`.

Of course, this doesn't matter if you're just setting `current_position[Z_AXIS]` for the purposes of mesh leveling, and will set it back to the proper Z position after `G29` is completed.

thinkyhead, 2016-06-26:
n/m!

thinkyhead, 2016-06-26:
Making a PR out of it…

chrisqwertz, 2016-06-26:
roger, will report back with results tonight!

chrisqwertz, 2016-06-26:
Looks like the logic works now, what I have observed although is that the extruder will move to 0 z-height, then scrape across the bed to rest at the first probing point and all subsequent ones. Could it be, that my "MESH_HOME_SEARCH_Z" is not respected? I used the LCD to level the bed and have a M206 Z-0.1 toolhead offset.

epatel, 2016-06-26:
@chrisqwertz Sounds like `Z_MAX_POS` is the distance between the max endstop and the bed. Try making it 2mm shorter, i.e. if your `Z_MAX_POS` is `150` make it `148`. With `MESH_HOME_SEARCH_Z` equals 4 (mm) you should be able to "probe" via the encoder about 2mm down with 2 mm to spare for lower other points (but that would be a crazy bad bed)

thinkyhead, 2016-06-26:
@chrisqwertz Try setting a `MIN_Z_HEIGHT_FOR_HOMING` value also, as this will give you an additional lift in MBL. (Also, I would like to see how it behaves when homing to Z max.)

epatel, 2016-06-26:
By making sure `Z_MAX_POS` is less than the distance to the first probe point one makes sure the software endstops triggers before Z reaches the endstop (and other stuff maybe) over the whole bed. I am not sure how or if `MIN_Z_HEIGHT_FOR_HOMING` helps here. But again, hopefully no-one is actually printing that tall stuff.

thinkyhead, 2016-06-26:
I believe that without `MIN_Z_HEIGHT_FOR_HOMING` there is no lift between probe points in MBL, and in fact it moves back down to the bed, so the nozzle usually ends up dragging on the bed.

epatel, 2016-06-26:
I have never recommended to have the endstops so that the nozzle touches the bed, never! I have always recommended the opposite, set up the printer so it is safe to probe from a safe starting point.

epatel, 2016-06-26:
And my hope is that @Roxy-3D will be able to make something useful of all the changes that has gone in. Please note that I was going to add a PR after @chrisqwertz verified this. But I see you have already has merged this yesterday.

thinkyhead, 2016-06-26:
> I have never recommended to have the endstops so that the nozzle touches the bed, never!

Nevertheless, we are talking about tilted and warped beds here, and people very, very often have their Z min endstops set slightly lower than the bed.

epatel, 2016-06-26:
Nevertheless, I believe it is bad to have the endstop lower than the bed period! as that will push up on the hotend and can cause various effects, moving bed, moving hotend/carriage movement etc and maybe even break it after some time. But just moving it slowly out of position I think is bad enough. Well, thats me. I would not set up my printers like that, ever.

thinkyhead, 2016-06-26:
> I would not set up my printers like that, ever.

The thing is, on a Cartesian machine with a Z_MIN endstop, the endstop switch itself blocks Z from moving any lower. Thus, if you want to have the freedom to move lower (for whatever reason), you must move your endstop a little lower. It's also common on some setups for the Z endstop to slip a little over time. In any case, it isn't entirely uncommon, and we must accommodate a wide variety of "non-ideal" situations by providing some safety margins. A raise between probes in MBL is one such "safety margin."

epatel, 2016-06-27:
The thing is that there are several kinds of switches. Switches with an arm and some without, and optos. Yes, for those without an arm that could be a problem. But, I'd suggest changing the endstop to one with an arm or best get an opto endstop. The price of the hotend is way larger than a good endstop so it should be a no-brainer. But, this issue is not about z min endstops, this is specifically for z max endstop. And I have explained even with a small picture how I recommend configuring for this, making it safe, because with a safer number for `Z_MAX_POS` it is less risk to run pass the z max endstop (with the software endstop). I will not explain this, I am tired.

This seem to be a good deal.
http://www.ebay.com/itm/3pcs-Opto-Endstop-Switch-Kit-for-CNC-3D-Printer-RepRap-Prusa-Mendel-RAMPS1-4-/161836552716

epatel, 2016-06-27:
> Not good. It should be the distance from the nozzle to the bed when Z is homed at the top.

Yeah, you are right. But I thought it was obvious that I meant nozzle-to-bed, because that is basically what `Z_MAX_POS` means. And really the distance between nozzle and z max endstop when endstop is triggered is not really interesting (can't recall seeing anyone ever mentioning it).

So sorry for not being precis enough.

thinkyhead, 2016-06-27:
> I thought it was obvious that I meant nozzle-to-bed

I'm just thinking ahead, to the person who finds this thread months from now and gets out their tape measure, and…

chrisqwertz, 2016-06-29:
Is there anything more I can do to assist?

epatel99, 2016-06-29:
@chrisqwertz Uh, I think the fix is in RCBugFix now so test that maybe? Or that was what you did? Otherwise I think it is all fine and it's ok to close this issue if you think it's ok now.

chrisqwertz, 2016-06-29:
Allright, will try to check tonight

jbrazio, 2016-07-11:
@chrisqwertz did you had the opportunity to check ?

Offtopic:
@epatel something wrong with your Github account ? I now see a @epatel99 using your avatar.

epatel, 2016-07-12:
@jbrazio No problem, I just sometime happen to use my "job" github account (when I am too lazy to logout).

boelle, 2016-08-11:
has this one been reverified with RC7?

closed 2016-08-11 by thinkyhead