MarlinFirmware/Marlin #4089: very minor issue: latest rcbugfix does not have the m851 definition in configuration.h

DjDemonD, 2016-06-19:
Previous RC version have included:

_// Custom M code points
#define CUSTOM_M_CODES
#if ENABLED(CUSTOM_M_CODES)
  #if ENABLED(AUTO_BED_LEVELING_FEATURE)
    #define CUSTOM_M_CODE_SET_Z_PROBE_OFFSET 851
    #define Z_PROBE_OFFSET_RANGE_MIN -20
    #define Z_PROBE_OFFSET_RANGE_MAX 20
  #endif
#endif_

In the additional features section of configuration.h.
Has it been omitted intentionally or just accidentally?
Without this m code setting z offset after measuring distance from probing point to nozzle is more difficult. I tried with m206 but whichever way I set it i.e. m206 z-2 or m206 z2 it did not move the nozzle to the bed on start of printing. Using m851 z-2 this functions perfectly.

esenapaj, 2016-06-19:
I haven't knowledge of those, but it looks like that `CUSTOM_M_CODES` and `CUSTOM_M_CODE_SET_Z_PROBE_OFFSET ` were abolished
in the merged PR #4066 (Make M851 standard with a bed probe).
But it looks like that `Z_PROBE_OFFSET_RANGE_MIN` and `Z_PROBE_OFFSET_RANGE_MAX` are still valid.

DjDemonD, 2016-06-19:
So with the version of bugfix I am testing right now (18-06-16) I cannot send m851 or receive any info using m851 but it is still a variable in the eeprom, even if I reset eeprom to defaults. My console shows 
echo: M851 Z0 at the end of the eeprom m501 readout. But it does not return echo: Z Offset: z-0.x.

However if I restore the configuration I quoted above (custom m codes) to the configuration.h in the additional features section of this version of bugfix, then this functionality returns.
Two things spring to mind (and they may be wrong if I am not understanding this well enough)
1) its an omission and needs to be put back in
2) its deliberate - in which case I presume probe offset entered in configuration.h under

_#define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below +above  [the nozzle]

is automatically utilised to position/lower the nozzle when printing begins, after homing/autolevelling.
If this is the case then its a backwards step, when adjusting probes/bed level etc.. the ability to change the z offset from the console/LCD and then save to eeprom is extremely useful. Changing the firmware and re-uploading it is time consuming.

RicardoGA, 2016-06-19:
@DjDemonD download the latest rcbugfix that problem was fixed yesterdad, the change is deliberate to standarize the gcode, with the latest rcbugfix to ajust rhe z ofset you should send (ex: m851 z-2.0)

jbrazio, 2016-06-19:
It was intentional with PR #4066.

closed 2016-06-19 by jbrazio