MarlinFirmware/Marlin #4090: FILAMENT_RUNOUT_SENSOR not working?

lavato, 2016-06-19:
I am trying to use the FILAMENT_RUNOUT_SENSOR option for printer to pause when there is no filament however, it does not seam to work.

I use RAMPS EEB config and filament sensor which is HIGH (4.9v)  on the signal side when the filament is not available.
Regardless if I supply HIGH  to Signal or not (which I do manually when testing) nothing happens, printer does not pause.

I have tried two pins D4  (servo) & D2 (X_MAX) both don't work.

When trying D4, I did below updates to config:
```
#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  const bool FIL_RUNOUT_INVERTING = true;
  #define ENDSTOPPULLUP_FIL_RUNOUT
  #define FILAMENT_RUNOUT_SCRIPT "M600"
#endif
```

[image]

When trying D2, I did the same as for D4 put also did updated the "pins_RAMPS_14.h" with below:
```
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define FILRUNOUT_PIN     2
#endif

```
[image]

Is there something else which needs to be set?

thinkyhead, 2016-06-20:
I presume you are testing with SD printing? (It only interrupts the print when SD printing.)

lavato, 2016-06-20:
Yes

thinkyhead, 2016-06-20:
I've made some tweaks to this feature, but probably not anything that will affect your results. Still, start with the latest code and try also changing `FIL_RUNOUT_INVERTING` to `false`, both with and without `ENDSTOPPULLUP_FIL_RUNOUT`, to see if the results change.

lavato, 2016-06-20:
Because this never worked for me, can you tell me when does Marlin start checking for the status of the pin, e.g. Is it from the moment the print start or during the warm-up?

lavato, 2016-06-20:
I tried but it did not work, since the TRIGGER is on HIGH. If something, it should be a pull down.

In any case to prove the point I un-commented: `#define USE_XMAX_PLUG` and I get a following output.
For first M119, the signal is LOW and for the second the Signal is HIGH.

```
SENDING:M119
Reporting endstop status
x_min: open
x_max: TRIGGERED
y_min: open
z_min: TRIGGERED
>>>m119
SENDING:M119
Reporting endstop status
x_min: open
x_max: open
y_min: open
z_min: TRIGGERED
```

In any case this proves that the wiring is correct at least.

Note that it does not mater what I do, ether setting the signal pin to HIGH or LOW,  the printer will not pause.

thinkyhead, 2016-06-21:
It's pretty straightforward…
```cpp
  #if ENABLED(FILAMENT_RUNOUT_SENSOR)
    if (IS_SD_PRINTING && !(READ(FIL_RUNOUT_PIN) ^ FIL_RUNOUT_INVERTING))
      handle_filament_runout(); // Send "M600" command
  #endif
```
If you have `FILAMENTCHANGEENABLE` enabled then when SD printing it should run the `M600` command immediately upon receiving the signal from the runout sensor.

lavato, 2016-06-21:
Not sure I understand what you say. Do I need to do any changes to Marlin code for this to work or this is some config issue?

thinkyhead, 2016-06-22:
> M600 command via Pronterface, again nothing happens

So you definitely have `FILAMENTCHANGEENABLE` enabled too…? Hmm, strange behavior.

> `filament_ran_out` in `Marlin_main.cpp` not being reset

Right. Actually the `M600` handler resets it, so it's the only supported command for the runout script.

lavato, 2016-06-22:
> If you have `FILAMENTCHANGEENABLE `enabled then when SD printing it should run the M600 command soon after receiving the signal from the runout sensor.

and 

> So you definitely have `FILAMENTCHANGEENABLE `enabled too…? Hmm, strange behavior

Oh, I missed this earlier. I don't actually. I will check it out tonight and come back.

However may I suggest, if it is that, comments be updated in the Confirguration.h file next to `FILAMENT_RUNOUT_SENSOR` section as it is **not apparent at all** that  this has to be done. (as you can see ;) )

**Question:**
 Also, the `FILAMENT_RUNOUT_SENSOR` poller starts during the warm-up and auto level, should't it start polling when the printing actually starts, as if you have a sensor which detects blockages it will trigger during the warm-up since the filament is not moving?

thinkyhead, 2016-06-22:
Simply enable the option. `ULTIPANEL` is set in `Conditionals.h` when you select `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`…
```cpp
  #if ENABLED(REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER)
    #define DOGLCD
    #define U8GLIB_ST7920
    #define REPRAP_DISCOUNT_SMART_CONTROLLER
  #endif

  #if ENABLED(ULTIMAKERCONTROLLER) || ENABLED(REPRAP_DISCOUNT_SMART_CONTROLLER) || ENABLED(G3D_PANEL) || ENABLED(RIGIDBOT_PANEL)
    #define ULTIPANEL
    #define NEWPANEL
  #endif
```

lavato, 2016-06-22:
Sounds great. I will try tonight and come back with results.

May I suggest comments be updated in the  'Configuration.h' next to  section for `FILAMENT_RUNOUT_SENSOR`,  as it is **not apparent at all** that  this has to be done.

Also, the  `FILAMENT_RUNOUT_SENSOR` process starts polling during the warm-up and auto level, should't it start polling when the printing actually starts printing, as if you have a sensor which detects blockages, it will trigger during the warm-up since the filament is not moving?

closed 2016-06-22 by lavato

thinkyhead, 2016-06-23:
> shouldn't it start polling when the printing actually starts printing

I suppose so. There's no concept in Marlin of "currently printing" but it can do things like monitor the sensor whenever there has been a recent movement command.

thinkyhead, 2016-06-23:
> The "Change Filament" message does not disappear from the LCD screen once print is resumed.

The current `M600` feature is going to be replaced pretty soon.
But this sounds like something we could fix up in advance of that.

lavato, 2016-06-23:
> I suppose so. There's no concept in Marlin of "currently printing" but it can do things like monitor the sensor whenever there has been a recent movement command.

`But this sounds like something we could fix up in advance of that.`

Is it best to raise the above two as separate issues?

jbrazio, 2016-06-23:
The big question is what do we should consider "printing", is it moving the tower ? Is it extruding plastic ? We don't have a "start" and a "stop" command, we do some eyeballing and guessing for the print timers but that's not accurate enough to detect a printing state.

lavato, 2016-06-23:
I know what you mean... but isn't it better to start polling when extruding plastic then during the heat-up sequence,  would't you say?