MarlinFirmware/Marlin #4091: Da Vinci + Ramps Thermistor Temp Profile Option

opensourcefan, 2016-06-19:
I apologize if this has already come up, I searched and couldn't find it. 

Would it be possible to add a temp profile for the non standard Da Vinci thermistors (extr + bed) so thermistors don't need to be swapped out when doing a Ramps retrofit? It seems as though it's becoming more common to Ramp a Da Vinci and swapping out the thermistors is preventing a simple plug and print  kit. I'm not sure if it's possible to use profiles that are already in Rep92.

thinkyhead, 2016-06-20:
Do both the bed and the hot-end use the same thermistor? There's a [Da Vinci version of Repetier](https://github.com/luc-github/Repetier-Firmware-0.92) that might give some clues….

I see some data is [posted in a spreadsheet](https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/bed-extruder-tables.xlsx), and then there are some thermistor tables in [this Da Vinci Repetier code](https://github.com/luc-github/Repetier-Firmware-0.92/blob/44a1430032513559d3f3ae7e16c567e6ce9fa144/src/ArduinoAVR/Repetier/Extruder.cpp#L755).

opensourcefan, 2016-06-20:
I believe both the bed and extruder use the same thermistor. In all my studying I think I heard that they may be 500k versions or something very obscure. I did look through Repetier 92 as it has proven to work but my eyes aren't experienced enough to know what I'm looking at ;)

opensourcefan, 2016-06-20:
Looking at your links, yep that all seems to make sense. I'm pretty sure that Repetier is ready for Da Vinci as written so whatever is default 'should' be the correct settings. I have no clue what to do with that information though in respect to adding it to Marlin.

thinkyhead, 2016-06-20:
https://github.com/luc-github/Repetier-Firmware-0.92/issues/104

opensourcefan, 2016-06-20:
Ya, that confuses me a bit. Makes me think that using a custom table could make the Da Vinci thermistors work with Ramps?

opensourcefan, 2016-06-20:
In respect to the forum link it appears that they are trying different thermistors with the Da Vinci boards no? This situation is when using a Ramps setup, effectively negating everything except the stock thermistor which just needs to be compensated for. I could be thinking about this all wrong though or maybe not explaining it right.

thinkyhead, 2016-06-20:
A custom table might work, we just need to find the right values to place into it. We also have to see whether a basic RAMPS can read a 500K thermistor without any alterations. The highest resistance I see in the current thermistor tables is 200K.

opensourcefan, 2016-06-20:
I think even if we can get close we can compensate on our own after some trial and error. I'm not sure if stock is 500k or 300k, I've seen both written. I'd like to find a datasheet and see if this tool can create a table for us... https://nathan7.eu/stuff/RepRapCalculator/RepRapCalculator.html#TempLookup

thinkyhead, 2016-06-20:
300K probably. As I search around, all [the conversions so far](http://www.soliforum.com/post/115278/#p115278) just replace the thermistor with something Marlin knows about.

opensourcefan, 2016-06-20:
Yep and that was/is the only current option. My goal was to create a kit that is simply plug and print so anyone could easily do the conversion without having to worry about swapping them out.

jbrazio, 2016-07-18:
Closing due to inactivity.
Please re-open this issue if still valid.

closed 2016-07-18 by jbrazio

opensourcefan, 2016-07-18:
Making this work is beyond my expertise, but considering the XYZ Da Vinci thermistor setup is so non standard best just change them out when retrofitting, as usual.