MarlinFirmware/Marlin #4104: manual bed leveling not being respected

uilson, 2016-06-21:
Hi, I don't know if anyone else notice this also but I did,

Before starting a print I level the bed manually (using the bed level option in menu) and them send a M500 to store the values to the EEPROM (or am I using this M500 wrongly?) using the Pronterface,

But for some reason it does not respect the values that were measured when I send the print by USB, and it sends the nozzle to the Z min (in my case 0).

But using the SD card I notice that it respected the values, at least the 2 times I printed via SD card 

Can anyone else confirm this or is this a isolate case?  

I'm using the **Marlin 1.1.0-RC6 - 24 April 2016**, I did not added any RC fix and only configured the firmware with the stock values from the Prusa i3 that I bought, seller sent me the firmware the he used and I just altered the values from RC6 to match it and activated the mesh/manual bed leveling.

PS: I'm new to 3D printing

epatel, 2016-06-21:
In RCBugFix there is an addition for MBL to handle homing single (or less than all axises). Before MBL would become disabled if one homed (G28) less than all axises. But this should be fixed now in RCBugFix.

GreatGrizzly, 2016-06-21:
@uilson You'll need to pull the whole branch and make your modifications. Way too many things change in the configuration.h for you to reuse your old one, but you can try first.

jbrazio, 2016-07-11:
Closing due to inactivity.

closed 2016-07-11 by jbrazio