MarlinFirmware/Marlin #4112: Support For Dual Zone Heatbeds

AlexDoran, 2016-06-21:
Hi there,

I'm working with a new Dual Zone Heatbed, it can be controlled via the RAMPS, however i want to be able to read each Zone's Temp independently, is it a huge task to add Support for another Heatbed   ##Therm input? I was going to just register it as another Hotend, but ideally want it to be displayed as a second heatbed on the Full Graphics LCD.

Thanks

Alex

crncrussell, 2016-06-21:
I second this request.  Would be great to be able to control the second zone with appropriate G/M codes and have seperate independent temperature control via the LCD as well.

thinkyhead, 2016-06-22:
##### Notes:
Will need to add to `Configuration.h`…
```cpp
#define TEMP_SENSOR_BED2
```
For `pins_THEBOARD.h` we will need…
```cpp
#define TEMP_BED2_PIN nn
#define HEATER_BED2_PIN nn
```
Other bed-specific parameters will apply to both beds.

AlexDoran, 2016-06-22:
Sounds Promising! Is this something a noob could implement.

crncrussell, 2016-06-22:
Thinkyhead, is this something you think can make it into the RC7, or are we looking down the road a bit on the roadmap?

Roxy-3D, 2016-06-22:
My vote is I think we should focus on just bug fixes for existing features right now.  If we do that, we should be able to get the current code very solid and promote it to the 'Stable Release' in 3 or 4 weeks.   At that point, we can clone the Stable Release and create as many Development branches as makes sense.   This can be one of the Development branches.    

The important thing is we need some rock solid branch for people that don't want new features and mostly need stability.

I would urge you to do this work off of the existing RC-6 RCBugFix branch and try to stay in sync with it.  That will make it easier to create a Development Branch as soon as we go 'Golden'.

AlexDoran, 2016-06-22:
I can assure you that I 100% have no idea how to implement this myself LOL. 

Thanks Again

Alex

Roxy-3D, 2016-06-22:
OK...   Well, it really isn't that hard to impliment.  Most of the logic is already in place to support this.   It isn't much different than supporting an extra hotend.    But a lot of stuff has to be tweaked to make this fully supported.   These kind of changes really should be done in a Development Branch and not one that we are trying to Stabilize.

jbrazio, 2016-07-11:
The questions are..
Why a dual bed hot zone is needed ?
How many Marlin users require this feature ?

jbrazio, 2016-07-16:
Closing due to inactivity.
Please re-open this issue if still valid.

dragondgold, 2017-11-20:
Is there any update on this issue? It would be great to have this on big printers!

thinkyhead, 2017-11-27:
@dragondgold This feature is definitely on our TODO list. It just needs a good specification, and then it will be a minor pile of work. It would need to be integrated into various temperature-related G-codes, the PWM / PID bed heater functions, and the appropriate LCD menus. Marlin is pretty well hardcoded right now for a single bed heater with a single bed temperature sensor.

jlipavsky, 2017-12-28:
Still looking for a way to control 2 heatbed zones for large printbeds without the use of a separate controler

fvergniaud, 2018-01-17:
I'm also interested on this feature, but to control the heater for the chamber,or any other slow temperature/heater device in the printer. I take a look to implement these by myself ,Marlin reference to extruders with posive numbers and all negative numbers  ( <0 ) reference to the  bed, only one bed. But it ever use -1 as parameter when call the temperature functions, So we need to adjust these first.

zebrabunny, 2018-09-03:
Yep, me too. energy costs for the 12 x 24 heatbed are pretty substantial, and a method for fixing this would be nice!

Rovexx, 2019-01-05:
I am planning a build for a 500x700mm corexy printer. Dual heatzones would save a lot of money on heating. 

So I would love this feature

rusnob, 2019-07-21:
Does the two zone heating work now?

kurtnelle, 2020-08-09:
How about how? Has it been implemented?

ellensp, 2020-08-09:
This feature (from marlin 1.1.x,  4 years ago) does not look to have been continued in Marlin 2.

sjasonsmith, 2020-11-29:
This is a duplicate of #4112. One or the other should be closed.

merald, 2022-03-25:
What is status on this today? I really need att least 2 zones on my 500x500 printer. 1/4 or 4/4 heating pads.

thisiskeithb, 2022-03-25:
> What is status on this today? 

No updates. This is still an open feature request among [nearly 500 others](https://github.com/MarlinFirmware/Marlin/issues?q=is%3Aopen+is%3Aissue+label%3A%22T%3A+Feature+Request%22).

kurtnelle, 2022-03-25:
> What is status on this today? I really need att least 2 zones on my 500x500 printer. 1/4 or 4/4 heating pads.

I predict that they will never implement this; see if another eco system can afford you this feature.

thisiskeithb, 2022-03-25:
> I predict that they will never implement this

We’ll happily accept a PR!

kurtnelle, 2022-03-25:
> We’ll happily accept a PR!

What is a PR?

thisiskeithb, 2022-03-25:
> What is a PR?

[Pull Request](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/about-pull-requests#:~:text=Pull%20requests%20let%20you%20tell,merged%20into%20the%20base%20branch.)

We’ll accept code for this feature request if someone wants to work on it.

InsanityAutomation, 2022-03-25:
One word of warning, and the reason I wont touch it, is that this may lead to warping of the plate. The smaller heater will also be working at a notably higher duty cycle as the cooler porting saps off all of the heat. Quite simply, I havnt seen a situation where its worth the risks.

kurtnelle, 2022-03-25:
> I havnt seen a situation where its worth the risks.

I want to use it on multiple heat beds with a tempered glass plate on top; it's not going to warp enough to matter.