MarlinFirmware/Marlin #4115: Turning On PID_BED_DEBUG serves no purpose

Roxy-3D, 2016-06-21:
This code serves no purpose.  None of the floating point values print out.   They all print as 0.
~~~~
    #if ENABLED(PID_BED_DEBUG)
      SERIAL_ECHO_START;
      SERIAL_ECHO(" PID_BED_DEBUG ");
      SERIAL_ECHO(": Input ");
      SERIAL_ECHO(current_temperature_bed);
      SERIAL_ECHO(" Output ");
      SERIAL_ECHO(pid_output);
      SERIAL_ECHO(" pTerm ");
      SERIAL_ECHO(pTerm_bed);
      SERIAL_ECHO(" iTerm ");
      SERIAL_ECHO(iTerm_bed);
      SERIAL_ECHO(" dTerm ");
      SERIAL_ECHOLN(dTerm_bed);
    #endif //PID_BED_DEBUG
~~~~

Blue-Marlin, 2016-06-21:
Bed perfectly holding 55°. Todays RCBugFix. ???
```
21:51:53.148 : N96 M105*24
21:51:53.150 : echo:N96 M105*24
21:51:53.167 : ok T:25.0 /0.0 B:55.1 /55.0 B@:0 @:0
21:51:53.343 : echo: PID_BED_DEBUG : Input 55.07 Output 9.43 pTerm -19.99 iTerm 36.13 dTerm 6.72
21:51:53.540 : echo: PID_BED_DEBUG : Input 55.08 Output 1.57 pTerm -23.98 iTerm 35.82 dTerm 10.27
21:51:53.736 : echo: PID_BED_DEBUG : Input 55.07 Output 9.71 pTerm -19.99 iTerm 35.57 dTerm 5.87
21:51:53.933 : echo: PID_BED_DEBUG : Input 55.09 Output 0.00 pTerm -27.98 iTerm 35.21 dTerm 13.35
21:51:54.129 : echo: PID_BED_DEBUG : Input 55.10 Output 0.00 pTerm -31.98 iTerm 35.16 dTerm 16.57
21:51:54.342 : echo: PID_BED_DEBUG : Input 55.07 Output 11.25 pTerm -19.99 iTerm 35.31 dTerm 4.08
21:51:54.522 : echo: PID_BED_DEBUG : Input 55.08 Output 3.26 pTerm -23.98 iTerm 35.01 dTerm 7.76
21:51:54.719 : echo: PID_BED_DEBUG : Input 55.03 Output 35.08 pTerm -8.00 iTerm 34.90 dTerm -8.18
21:51:54.916 : echo: PID_BED_DEBUG : Input 55.05 Output 18.70 pTerm -15.99 iTerm 34.70 dTerm 0.01
21:51:55.112 : echo: PID_BED_DEBUG : Input 55.08 Output 2.62 pTerm -23.98 iTerm 34.39 dTerm 7.78
21:51:55.309 : echo: PID_BED_DEBUG : Input 55.04 Output 26.51 pTerm -11.99 iTerm 34.24 dTerm -4.27
21:51:55.505 : echo: PID_BED_DEBUG : Input 55.03 Output 34.08 pTerm -8.00 iTerm 34.13 dTerm -7.94
21:51:55.702 : echo: PID_BED_DEBUG : Input 55.01 Output 41.52 pTerm -4.00 iTerm 34.08 dTerm -11.43
21:51:55.898 : echo: PID_BED_DEBUG : Input 55.01 Output 40.90 pTerm -4.00 iTerm 34.03 dTerm -10.86
21:51:56.091 : echo: PID_BED_DEBUG : Input 55.00 Output 48.24 pTerm 0.00 iTerm 34.03 dTerm -14.21
21:51:56.205 : N97 M105*25
21:51:56.210 : echo:N97 M105*25
21:51:56.223 : ok T:25.0 /0.0 B:55.0 /55.0 B@:24 @:0
21:51:56.288 : echo: PID_BED_DEBUG : Input 55.00 Output 47.53 pTerm 0.00 iTerm 34.03 dTerm -13.50
21:51:56.484 : echo: PID_BED_DEBUG : Input 55.00 Output 46.85 pTerm 0.00 iTerm 34.03 dTerm -12.82
21:51:56.681 : echo: PID_BED_DEBUG : Input 55.00 Output 46.21 pTerm 0.00 iTerm 34.03 dTerm -12.18
21:51:56.878 : echo: PID_BED_DEBUG : Input 55.00 Output 45.60 pTerm 0.00 iTerm 34.03 dTerm -11.57
21:51:57.074 : echo: PID_BED_DEBUG : Input 55.00 Output 45.03 pTerm 0.00 iTerm 34.03 dTerm -10.99
21:51:57.271 : echo: PID_BED_DEBUG : Input 55.00 Output 44.48 pTerm 0.00 iTerm 34.03 dTerm -10.44
21:51:57.467 : echo: PID_BED_DEBUG : Input 55.00 Output 43.95 pTerm 0.00 iTerm 34.03 dTerm -9.92
21:51:57.664 : echo: PID_BED_DEBUG : Input 55.00 Output 43.46 pTerm 0.00 iTerm 34.03 dTerm -9.42
21:51:57.861 : echo: PID_BED_DEBUG : Input 55.00 Output 42.99 pTerm 0.00 iTerm 34.03 dTerm -8.95
21:51:58.057 : echo: PID_BED_DEBUG : Input 55.00 Output 42.54 pTerm 0.00 iTerm 34.03 dTerm -8.51
21:51:58.254 : echo: PID_BED_DEBUG : Input 55.00 Output 42.11 pTerm 0.00 iTerm 34.03 dTerm -8.08
21:51:58.450 : echo: PID_BED_DEBUG : Input 55.00 Output 41.71 pTerm 0.00 iTerm 34.03 dTerm -7.68
21:51:58.647 : echo: PID_BED_DEBUG : Input 55.00 Output 41.33 pTerm 0.00 iTerm 34.03 dTerm -7.29
21:51:58.843 : echo: PID_BED_DEBUG : Input 55.00 Output 40.96 pTerm 0.00 iTerm 34.03 dTerm -6.93
21:51:59.040 : echo: PID_BED_DEBUG : Input 55.00 Output 40.61 pTerm 0.00 iTerm 34.03 dTerm -6.58
21:51:59.237 : echo: PID_BED_DEBUG : Input 55.00 Output 40.29 pTerm 0.00 iTerm 34.03 dTerm -6.25
21:51:59.265 : N98 M105*22
21:51:59.269 : echo:N98 M105*22
21:51:59.282 : ok T:25.0 /0.0 B:55.0 /55.0 B@:20 @:0
21:51:59.433 : echo: PID_BED_DEBUG : Input 55.00 Output 39.97 pTerm 0.00 iTerm 34.03 dTerm -5.94
21:51:59.630 : echo: PID_BED_DEBUG : Input 55.00 Output 39.68 pTerm 0.00 iTerm 34.03 dTerm -5.64
21:51:59.826 : echo: PID_BED_DEBUG : Input 55.00 Output 39.39 pTerm 0.00 iTerm 34.03 dTerm -5.36
21:52:00.023 : echo: PID_BED_DEBUG : Input 55.00 Output 39.13 pTerm 0.00 iTerm 34.03 dTerm -5.09
21:52:00.219 : echo: PID_BED_DEBUG : Input 55.00 Output 38.87 pTerm 0.00 iTerm 34.03 dTerm -4.84
21:52:00.416 : echo: PID_BED_DEBUG : Input 55.00 Output 38.63 pTerm 0.00 iTerm 34.03 dTerm -4.60
21:52:00.613 : echo: PID_BED_DEBUG : Input 55.00 Output 38.40 pTerm 0.00 iTerm 34.03 dTerm -4.37
21:52:00.809 : echo: PID_BED_DEBUG : Input 55.00 Output 38.18 pTerm 0.00 iTerm 34.03 dTerm -4.15
21:52:01.006 : echo: PID_BED_DEBUG : Input 55.00 Output 37.97 pTerm 0.00 iTerm 34.03 dTerm -3.94
21:52:01.202 : echo: PID_BED_DEBUG : Input 55.00 Output 37.78 pTerm 0.00 iTerm 34.03 dTerm -3.74
21:52:01.399 : echo: PID_BED_DEBUG : Input 55.00 Output 37.59 pTerm 0.00 iTerm 34.03 dTerm -3.56
21:52:01.596 : echo: PID_BED_DEBUG : Input 55.00 Output 37.41 pTerm 0.00 iTerm 34.03 dTerm -3.38
21:52:01.792 : echo: PID_BED_DEBUG : Input 55.00 Output 37.24 pTerm 0.00 iTerm 34.03 dTerm -3.21
21:52:01.988 : echo: PID_BED_DEBUG : Input 55.00 Output 37.08 pTerm 0.00 iTerm 34.03 dTerm -3.05
21:52:02.185 : echo: PID_BED_DEBUG : Input 55.00 Output 36.93 pTerm 0.00 iTerm 34.03 dTerm -2.90
21:52:02.321 : N99 M105*23
21:52:02.341 : echo:N99 M105*23
21:52:02.357 : ok T:25.2 /0.0 B:55.0 /55.0 B@:18 @:0
21:52:02.382 : echo: PID_BED_DEBUG : Input 55.00 Output 36.78 pTerm 0.00 iTerm 34.03 dTerm -2.75
21:52:02.578 : echo: PID_BED_DEBUG : Input 55.00 Output 36.65 pTerm 0.00 iTerm 34.03 dTerm -2.61
21:52:02.775 : echo: PID_BED_DEBUG : Input 55.00 Output 36.52 pTerm 0.00 iTerm 34.03 dTerm -2.48
21:52:02.971 : echo: PID_BED_DEBUG : Input 55.00 Output 36.39 pTerm 0.00 iTerm 34.03 dTerm -2.36
21:52:03.168 : echo: PID_BED_DEBUG : Input 55.00 Output 36.27 pTerm 0.00 iTerm 34.03 dTerm -2.24
[... cut ...]
21:52:16.729 : echo: PID_BED_DEBUG : Input 55.00 Output 34.10 pTerm 0.00 iTerm 34.03 dTerm -0.07
21:52:16.926 : echo: PID_BED_DEBUG : Input 55.00 Output 34.09 pTerm 0.00 iTerm 34.03 dTerm -0.06
21:52:17.122 : echo: PID_BED_DEBUG : Input 55.00 Output 34.09 pTerm 0.00 iTerm 34.03 dTerm -0.06
21:52:17.319 : echo: PID_BED_DEBUG : Input 55.00 Output 34.09 pTerm 0.00 iTerm 34.03 dTerm -0.06
21:52:17.515 : echo: PID_BED_DEBUG : Input 55.00 Output 34.09 pTerm 0.00 iTerm 34.03 dTerm -0.05
21:52:17.611 : N104 M105*34
```

Roxy-3D, 2016-06-21:
This log is trying to heat the bed to 210 C.  Once the temperature gets within 10 C (at 200 C) it switches out of BANG_MAX mode and tries to do PID.   Look at the P, I, and D terms.  Everything goes to 0.
~~~~
echo: PID_DEBUG 0: Input 190.86 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 191.28 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 191.61 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 192.04 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 192.53 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 192.76 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 193.16 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 193.65 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 194.01 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 194.41 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 194.77 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 195.16 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 195.53 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 195.79 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 196.22 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 196.45 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 196.88 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 197.17 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 197.47 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 197.86 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 198.26 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 198.59 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 198.91 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 199.31 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 199.57 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 199.97 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 200.18 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 200.85 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 201.25 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 201.52 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 202.10 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 202.50 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 202.90 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 203.35 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 203.62 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 204.11 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 204.38 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 204.87 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 205.04 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 205.54 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 205.76 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 205.98 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 206.38 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 206.52 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 207.01 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 207.14 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 207.41 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 207.81 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 207.86 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 208.17 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 208.53 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 208.57 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 208.71 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 209.02 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
echo: PID_DEBUG 0: Input 209.24 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
Disconnected.
~~~~

closed 2016-06-21 by Roxy-3D