MarlinFirmware/Marlin #4116: hard to navigate code

Sgaggino, 2016-06-21:
I am not a skilled programmer, but whenever i want to look up for something it`s really hard to keep track of all the redirections related to #defines, #ifdef, etc. I think it could be better to modularize the code. The setup has to be something separated from the working code itself.

paulusjacobus, 2016-06-21:
That's because the Arduino environment is meant for hobbyist/school
education. It's not a full professional development environment. Having
said that, the Open Source community has pushed this far to the bleeding
edges of what it can do. I suggest to read the books from Simon Monk
("*Programming
Arduino Getting Started with Sketches"). *He is brilliant in explaining how
it works and how to squeeze out the few more bytes and processing power
through coding tricks. These are all applied in marlin but you won't
understand it unless you get the deep explanation. Enjoy reading! It's very
well written for novices.

Roxy-3D, 2016-06-21:
> I think it could be better to modularize the code. 

It is planned.   But the top priority is to get a very stable branch of the code published so we can take our time cleaning up the code.

closed 2016-06-22 by jbrazio

Sgaggino, 2016-06-23:
@paulusjacobus I will read about that.
@Roxy-3D I understand your point, but i am seeing that there are lot of electronic platforms and it's getting difficult to port because of the complexity of the code (i guess). I don't mean to modularize everything, but if only we can select a planner regardless of the firmware i could be beneficial. See... right now there's GRBL and TINYG (getting traction), maybe on the future we'll have more planners to choose. Even tho, which parts do you think ar most unstable? Do you think i could help on some topics?