MarlinFirmware/Marlin #4124: The latest RCBugFix compiles with error when enabling M48

lrpirlet, 2016-06-22:
@thinkyhead  I just tried the latest RCBugFix using the following value in the Configuration.h...
BTW, thanks a lot for that work.. I appreciate...

``` diff
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index a55c878..a313b8a 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -95,10 +95,10 @@
 // User-specified version info of this build to display in [Pronterface, etc] terminal window during
 // startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
 // build by the user have been successfully uploaded into firmware.
-#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
+#define STRING_CONFIG_H_AUTHOR "LRP Motorized MBL + LIN_Advance" // Who made the changes.
 #define SHOW_BOOTSCREEN
 #define STRING_SPLASH_LINE1 SHORT_BUILD_VERSION // will be shown during bootup in line 1
-//#define STRING_SPLASH_LINE2 STRING_DISTRIBUTION_DATE // will be shown during bootup in line 2
+#define STRING_SPLASH_LINE2 STRING_CONFIG_H_AUTHOR // STRING_DISTRIBUTION_DATE // will be shown during bootup in line 2

 // @section machine

@@ -123,7 +123,7 @@

 // Optional custom name for your RepStrap or other custom machine
 // Displayed in the LCD "Ready" message
-//#define CUSTOM_MACHINE_NAME "3D Printer"
+#define CUSTOM_MACHINE_NAME "Pi3 3D LabteK"

 // Define this to set a unique identifier for this printer, (Used by some programs to differentiate between machines)
 // You can use an online service to generate a random UUID. (eg http://www.uuidgenerator.net/version4)
@@ -202,7 +202,7 @@
 #define TEMP_SENSOR_1 0
 #define TEMP_SENSOR_2 0
 #define TEMP_SENSOR_3 0
-#define TEMP_SENSOR_BED 0
+#define TEMP_SENSOR_BED 1

 // This makes temp sensor 1 a redundant sensor for sensor 0. If the temperatures difference between these sensors is to high the print will be aborted.
 //#define TEMP_SENSOR_1_AS_REDUNDANT
@@ -250,7 +250,7 @@
 #define BANG_MAX 255 // limits current to nozzle while in bang-bang mode; 255=full current
 #define PID_MAX BANG_MAX // limits current to nozzle while PID is active (see PID_FUNCTIONAL_RANGE below); 255=full current
 #if ENABLED(PIDTEMP)
-  //#define PID_AUTOTUNE_MENU // Add PID Autotune to the LCD "Temperature" menu to run M303 and apply the result.
+  #define PID_AUTOTUNE_MENU // Add PID Autotune to the LCD "Temperature" menu to run M303 and apply the result.
   //#define PID_DEBUG // Sends debug data to the serial port.
   //#define PID_OPENLOOP 1 // Puts PID in open loop. M104/M140 sets the output power from 0 to PID_MAX
   //#define SLOW_PWM_HEATERS // PWM with very low frequency (roughly 0.125Hz=8s) and minimum state time of approximately 1s useful for heaters driven by a relay
@@ -262,10 +262,17 @@
   #define K1 0.95 //smoothing factor within the PID

   // If you are using a pre-configured hotend then you can use one of the value sets by uncommenting it
+  // lrp found using pidautotune M303 E0 S200 C8 This will heat the first nozzle (E0), and cycle around
+  // the target temperature 8 times (C8) at the given temperature (S200) and return values for P I and D
+  // see http://reprap.org/wiki/PID_Tuning
+  #define  DEFAULT_Kp 22.41
+  #define  DEFAULT_Ki 1.56
+  #define  DEFAULT_Kd 80.41
+
   // Ultimaker
-  #define  DEFAULT_Kp 22.2
-  #define  DEFAULT_Ki 1.08
-  #define  DEFAULT_Kd 114
+  //#define  DEFAULT_Kp 22.2
+  //#define  DEFAULT_Ki 1.08
+  //#define  DEFAULT_Kd 114

   // MakerGear
   //#define  DEFAULT_Kp 7.0
@@ -397,13 +404,13 @@
[... 318 lines cut ...]

                                                            ^

Marlin_main.cpp:4274: error: 'probe_pt' was not declared in this scope

       verbose_level);

                    ^

Marlin_main.cpp:4361: error: 'ProbeStow' was not declared in this scope

         deploy_probe_for_each_reading ? ProbeDeployAndStow : last_probe ? ProbeStow : ProbeStay,

                                                                           ^

Marlin_main.cpp:4361: error: 'ProbeStay' was not declared in this scope

         deploy_probe_for_each_reading ? ProbeDeployAndStow : last_probe ? ProbeStow : ProbeStay,

                                                                                       ^

exit status 1
'axis_unhomed_error' was not declared in this scope

```

thinkyhead, 2016-06-23:
I just merged #4021 which makes several changes to help the situation. See if the code now compiles better with your configuration.

lrpirlet, 2016-06-23:
@thinkyhead. Thanks a lot, I could compile and send it to the printer... Seem to react correctly...

Now, I would suggest to also be "physical" in M48 when Mesh Bed Leveling is active....
I suggest

```diff
From 418728f7c91c94867ba1f44e983a71b9c766fc60 Mon Sep 17 00:00:00 2001
From: lrpirlet <lrpirlet@gmail.com>
Date: Thu, 23 Jun 2016 20:57:18 +0200
Subject: [PATCH] makes M48 physical for Mesh Leveling too

---
 Marlin/Marlin_main.cpp | 3 +++
 1 file changed, 3 insertions(+)

diff --git a/Marlin/Marlin_main.cpp b/Marlin/Marlin_main.cpp
index 0e42a4b..aecfaa1 100644
--- a/Marlin/Marlin_main.cpp
+++ b/Marlin/Marlin_main.cpp
@@ -4261,6 +4261,9 @@ inline void gcode_M42() {
     #elif ENABLED(AUTO_BED_LEVELING_FEATURE)
       // we don't do bed level correction in M48 because we want the raw data when we probe
       planner.bed_level_matrix.set_to_identity();
+    #elif ENABLED(MESH_BED_LEVELING)
+      // we don't do bed level correction in M48 because we want the raw data when we probe
+      mbl.set_active(false);
     #endif

     setup_for_endstop_or_probe_move();
--
2.8.0.windows.1
```

Roxy-3D, 2016-06-24:
> Now, I would suggest to also be "physical" in M48 when Mesh Bed Leveling is active....
> I suggest:

Agreed!   However, I don't think this is hurting your measurements.   What ever offset is getting added on, it will be the same for the entire sample set.

thinkyhead, 2016-06-24:
It would be nice (perhaps) to be able to turn all forms of leveling compensation on and off, without necessarily clearing the matrix, so we can turn it off at the start of `M48` and then re-enable compensation after it's done.

closed 2016-06-24 by thinkyhead