MarlinFirmware/Marlin #4127: LCD BTN not work well. VIKI 2.

fbarcenas, 2016-06-23:
Some weeks i don't update with the last rcbugfix. Today i do and what is my surprise than the button of the LCD when i press the beeper always sound BUT not go inside the menus... some times work, another times not... is really desperate.

Some idea?

fbarcenas, 2016-06-23:
If i change:
#define BEEPER_PIN 61
to
//#define BEEPER_PIN 61

all works!!!

problem with the BEEPER?

jbrazio, 2016-06-23:
Maybe, what is your setting for `#define SPEAKER` ?

fbarcenas, 2016-06-23:
is:
//#define SPEAKER

thinkyhead, 2016-06-24:
@fbarcenas What kind of LCD? Does enabling `SPEAKER` allow you to play musical tones?

fbarcenas, 2016-06-24:
i use VIKI 2

Yes i can play musical tones.
___________________________
Francisco Barcenas Gertrudix
CNC-BARCENAS-BELLON, S.L.
Pol. Ind. Entrecaminos.
Avda. Holanda,42
13300   VALDEPEÑAS   
C. REAL     ESPAÑA                                                 
TL. 926648985                                            
info@cncbarcenas.com <mailto:info@cncbarcenas.com>                          
www.cncbarcenas.com <http://www.cncbarcenas.com/>

AVISO DE CONFIDENCIALIDAD  Y PROTECCIÓN DE DATOS
Sus datos figuran en un fichero propiedad de CNC-Bárcenas-Bellón SL. Si desea acceder a sus datos, rectificarlos, cancelarlos u oponerse a su tratamiento, diríjase a CNC-Bárcenas-Bellón SL, C/ Norte, nº 66, de Valdepeñas, provincia de Ciudad Real, con Código Postal 13300.
Tanto este mensaje como todos los posibles documentos adjuntos al mismo,son confidenciales y están dirigidos exclusivamente a los destinatarios de los mismos. Por favor, si Usted no es uno de dichos destinatarios, notifíquenos este hecho y elimine el mensaje de su sistema. Queda prohibida la copia, difusión o revelación de su contenido a terceros sin el previo consentimiento por escrito de CNC-Bárcenas-Bellón SL. En caso contrario, vulnerará la legislación vigente.

thinkyhead, 2016-06-24:
> Yes i can play musical tones.

Uncomment `#define SPEAKER` for best results. I believe that if you insert the suggested delay in `lcd_quick_feedback` the button will be more reliable.

jbrazio, 2016-08-01:
@fbarcenas can we close this ticket, is your problem solved ?

boelle, 2016-08-11:
i think this one can be closed... it should not take 10 days to respond

closed 2016-08-11 by thinkyhead