MarlinFirmware/Marlin #4130: [Request] Individual Auto Bed Leveling Offset Point

sarf2k4, 2016-06-23:
I would like to ask whether or not this could be a good idea or not, typically with abl alone, especially on my machine, the left side were higher compared to the right side. But physically the nozzle were already far from the bed.

If I were to use the z-offset alone, on the right side, the nozzle would be scraping the bed and no plastic would be extruded.

The abl offset point is to have an array of offset depending on the number of auto bed leveling points. It is similar to the manual mesh bed leveling but instead this one just by using an additional individual z-offset for that particular point.

I'm pretty sure that there are a number of people tried to resolve this issue even with myself here tried to fix it didn't solve either by adjusting the hardware calibration of the z-axis and the bed height. I'm sure that most solution probably is to reduce the probe distance between the nozzle but on some machine there are a lot of works and cannot be achieved so easily.

I want to know how this thing goes by incorporating ABL offset point by the dev team

Roxy-3D, 2016-06-23:
Is your machine a Prusa i3 derivitive?  If so, it kind of sounds like you don't have your Z Height set the same on both threaded rods.

jbrazio, 2016-06-23:
Like Roxy said, you should at least level the X axis not to the bed but to the rods where the bed slides on (assuming a Prusa design). You'll have to go back and forth between the two Z towers to get it right, don't trust the first alignment. Another advice I can give you is to lock with your finger the coupler between the stepper and the thread rod on the axis you're not aligning currently, otherwise they both can turn at the same time.

sarf2k4, 2016-06-23:
Yes i am using the prusa i3 variant and i did the calibration on the z
tower. Apparently it doesnt fix this kind of issues though in which i have
been experiencing couple of month ago and i do believe that i posted in
another abl issues on other thread but that one regarding the delta
kinematics.

I also gone all the trouvles of using the digital calipers to calibrate the
z towers left and right on the x rod holders with the motor/top bracket
mount as precise as i could.

sarf2k4, 2016-06-23:
I did the z tower calibration again and gonna do a test print again

jbrazio, 2016-06-23:
What you might have is Z twist between the two Z axis and this is impossible to compensate currently and it cause the nozzle to swing.

Roxy-3D, 2016-06-23:
> I also gone all the trouvles of using the digital calipers to calibrate the z towers left and right on the x rod holders with the motor/top bracket mount as precise as i could.

What I do it bring the nozzle down very close to the bed on one side.   And then move the nozzle over to the far side.  And adjust just one of the Z motors to get things closer to correct.  And then I repeate the process again.

sarf2k4, 2016-06-23:
I printed a bed test from tom sanladerer, only a layer. Initial layer
height is 0.2mm and here is the results

X-min, y-min = 0.15mm
X-min, y-max = 0.20mm
X-max, y-min = 0.15mm
X-max, y-max = 0.08mm

The print parameter are as followed for slic3r
In advanced tab, all are set to 0.4mm while infill are 0.6, overlaps are
15%, speeds are 60mm/s, while external are 50%, first layer speed are 35%.
As for the model, i used ths 190mm size type and scaled it down to 90% the
size because it exceeds my x y area.

The measurement were taken on the perimeter only because if the first solid
infill were taken into account, it will give a results somewhere around
0.4mm due to overflow on bottom solids.

If i increase the z offset i would get a good results for x-max, y-max but
not for x-min, y-max area.

I also wanted to avoid any extruder motors skipping steps due to nozzle are
being too close to the bed during the first layers. This is a very bad case
for pla cause pla would stick too well on the glass even without gluestick
layers.

closed 2016-07-18 by jbrazio