MarlinFirmware/Marlin #4131: [Request] Pause/Stop Post G-Code Script

sarf2k4, 2016-06-23:
Would like to request this feature especially for stopping prints from the lcd menu for example

`define post_stop "g28 x; g0 y180; m92 e0; ....."`

While for pause

`define post_pause "g0 z+10"`

But for the pause script, a resume script must be defined as well such as

`define before_resume "g0 z-10"`

What's your opinion on this one?

jbrazio, 2016-06-23:
@sarf2k4 I really thing pre/post scripts are a thing to be delegated into the slicer.
Concerning the pause script it may make sense to be within Marlin and maybe be part of the filament loading rewrite. @clexpert @petrzjunior

sarf2k4, 2016-06-23:
It is assuming that i wanted to stop the printer midway due to failed
prints, if can do it automatically from the post gcode script it would be a
good one since on stopping from the lcd menu, marlin didnt turn off both
heaters. Is it just me or am i missing something?

As for the pause print i would highly recommend to implement since pausing
it the print job it would be just touching the printed parts there.

Once again, this is a custom script for pause/stop accessible from lcd, not
start/end gcode from slicer

What of the m600 custom script command in the adv_config.h? Is it still
buggy?

thinkyhead, 2016-06-24:
We're looking to improve `M600`. See #3662.

closed 2016-07-11 by jbrazio