MarlinFirmware/Marlin #4132: would you please help me i need to remove temp display on lcd reprapdiscount smart controller

Iaurentcnc, 2016-06-23:
hello ,
where in the firmware may i set the display of the temperature off ?
i am using marlin on a cnc machine so no need of this display
thanks a lot 
laurent

Grogyan, 2016-06-23:
I reckon Marlin needs another configuration, for CNC cutting.
Like use the dummy thermistor, and have an icon on the display for the CNC cutting mode

thinkyhead, 2016-06-24:
@laurentcnc For the time being you can just edit the `lcd_implementation_status_screen` function directly and remove the lines where temperature is being printed.

jbrazio, 2016-07-11:
We really need to internally fork the CNC features from the 3D features.. this is becoming a must have feature. Closing this PR and opening a generic one.

closed 2016-07-11 by jbrazio