MarlinFirmware/Marlin #4136: How can I apply a new serial

youngyhw, 2016-06-24:


Blue-Marlin, 2016-06-24:
MarlinSerial is not made for more than one device.
The port number is #defined - not a variable.

Blue-Marlin, 2016-06-24:
"Remove MarlinSerial #2945" describes how to remove MarlinSerial and replace it with the stock Arduino one. (Would need some updates)

thinkyhead, 2016-06-24:
And in that thread someone mentioned…
> …hack[ing] into the newer Arduino HardwareSerial…

Would that be a straightforward process at this point?

youngyhw, 2016-06-30:
@Blue-Marlin From "Remove MarlinSerial #2945" , I didn't find how to add a new serial port, it just show us that apply Marlin serial is better than Hardware serial in storing. 
Is there any other issues can show how to do this or guidence?
Thanks.

youngyhw, 2016-06-30:
@thinkyhead how can I open this thread?

Blue-Marlin, 2016-06-30:
@youngyhw 
This tread is open.

As said before - MarlinSerial does not allow a second device.  #2945 shows how to remove MarlinSerial use the original Arduino HardwareSerial instead. Arduinos HardwareSerial can be handled as usual - just open a second device.
For the reasons described in #2945 we will not merge that.
Currently i don't see the option to import Arduinos HardwareSerial into Marlin. It would add another four files. Putting in "the hack" is not very difficult, but how to throw out the dependency to the stream-library is currently not obvious to me.

youngyhw, 2016-07-01:
@Blue-Marlin Thank you very much. I'll have a try.
Best regards.

jbrazio, 2016-07-11:
@youngyhw can we close this thread ?

closed 2016-07-18 by jbrazio