MarlinFirmware/Marlin #4137: Feature Request - Saving Auto Bed Leveling Matrix

enlight3d, 2016-06-24:
Hello, I thought it would be nice to have a function to allow to save to EEPROM the matrix saved by the auto bed leveling procedure. I know there was a mod allowing it 2 years ago, but I tried on latest RC version and it didn't work... so that's why :)
To make it work properly, there should be an option to allow or not to reprobe the bed before a print if a custom g-code is entered (just like when you enter G29) or maybe a prompt on the printer's screen to ask the user if it should reprobe the bed or not and thus erase the saved matrix with the new one.
The purpose of saving the matrix is because for now, if you want to print in multicolor, it tends to be difficult if you have auto bed leveling activated. The only way I found to this was by commenting out a section of the firmware saying that when a G28 is launched, it would erase the matrix saved in the current memory (not the eeprom as it doesn't seem implemented at all), and thus, by launching the second part of the print in a different colour but without a G28, it would kinda work for me. But it's not a good way to do it.

That's it folks :)

Roxy-3D, 2016-06-24:
There is a new version of Auto Bed Leveling in the works.   It allows (encourages) you to save the matrix (Mesh) that you are using.    Sit tight for 2 or 3 weeks and it should appear as a Development branch based off of RC-7.

adamfilip, 2016-06-24:
Marking my Calendar! I cant wait!!

enlight3d, 2016-06-24:
damn this is great ! Hope it will work good :p

waran14, 2016-06-24:
quick question, does doing a G1 X0 Y0 reset all the values created by G29 ? I am basically homing ->> BEd leveling >> Going to start position to do Wipe.. Here is my starting script

G28
G29
M109 S210 T0
G1 X0 Y0

Will the G1 X0 Y0 act as G28 X0 Y0 ?

jstefanop, 2016-06-24:
any ETA on RC-7?

Roxy-3D, 2016-06-24:
I'm guessing 2 or 3 weeks until it is running well enough that I feel comfortable releasing it.  I want it to be pretty solid when it gets released.   Right now, I'm delayed a little bit because I'm trying to stay synchronized to the RC-6 RCBugFix and all the probe changes have blown me out of the water.

thinkyhead, 2016-06-25:
> Will the G1 X0 Y0 act as G28 X0 Y0?

No. It will simply move to your current 0, 0 position.

thinkyhead, 2016-06-25:
> probe changes have blown me out of the water

Should be pretty much cleaned up now. I incorporated several suggested changes from Andreas. I think we're done shuffling code all over the place now. Let me know if you need help integrating your UBL changes with the updated code.

Roxy-3D, 2016-06-25:
I'll be OK.   It just isn't fun.  It rates up there with taking out the garbage and doing dishes!   I'll merge with the current RCBugFix in the morning.

jbrazio, 2016-07-11:
Closing due to marked as duplicate.

closed 2016-07-11 by jbrazio

davidandorange, 2018-05-26:
If I also put G1/G0 X0, Y0, Z0, I want to know if this will keep the auto bed matrix data or not?

Thanks

thinkyhead, 2018-05-27:
Those `G0` and `G1` commands will move the nozzle to the origin, that's all.