MarlinFirmware/Marlin #4138: Please help!

FoXxx87, 2016-06-24:
I have an older version of the [firmware 1.0.3](https://github.com/necdnk/Marlin) it is written function ( 'Electronic_scale_Z_probe'). I want to upgrade to a newer version. Help rewrite the function 'Electronic_scale_Z_probe' from the old (my) firmware to a new version of the firmware (or teach me how to do it yourself). I tried to copy itself, I have neither of which did not work, the printer is working but the function 'Electronic_scale_Z_probe not work. Thank you.

Roxy-3D, 2016-06-24:
The probe code has just gone through a major clean up to simplify it.    But your changes are fairly small.  It shouldn't be too difficult to cross that functionality over to the latest RC-BugFix.

FoXxx87, 2016-06-24:
I did, I did not work (((

thinkyhead, 2016-06-24:
It looks like you can just use the `HX711.cpp` and `HX711.h` files as they are.

Most of the changes from https://github.com/necdnk/Marlin/commit/6a23bcc800d5d5571a6bc297d2adafc9fbe0bdd8 should be straightforward to apply to the new code. Instead of `st_synchronize` we now have `stepper.synchronize`. It should be pretty obvious how to make the custom `synchronize` function for the scale probe.

The sanity check additions in https://github.com/necdnk/Marlin/commit/31f31bd4878a02864ce8f12ed53b26ff60604f17 will be similar. We now have several probe options, so regarding `Configuration.h`, `Conditionals.h`, and `SanityCheck.h`, simply integrate this probe by following the same style as the other probes in those files. (Search "PROBE" to find those instances.)

We now have a `Z_PROBE` pin and endstop bit. So for the code in `stepper.cpp` that is being adjusted by https://github.com/necdnk/Marlin/commit/9280719ad0df8a37f4b19682ef6fd1732333eb50, you should rewrite the endstop bit-setting part to work the same way as the other probes / endstops now do.

FoXxx87, 2016-06-25:
@thinkyhead Thank you ! I will try to do so.

jbrazio, 2016-07-11:
Closing due to inactivity.
Please re-open this issue if still valid.

closed 2016-07-11 by jbrazio

Dennis02121978, 2022-03-15:
Here is the HX711 for Marlin and Klipper.

https://www.3dartmaker.de/forums/topic/platine-end-version/