MarlinFirmware/Marlin #4139: SINGLENOZZLE: not extruding

Edie47, 2016-06-24:
Hi, when I uncomment SINGLENOZZLE and set EXTRUDERS to 2 my printer is not extruding. Returning EXTRUDERS to 1 and commenting SINGLENOZZLE extrusion works. This is my fork of RCBugFix branch https://github.com/Edie47/Marlin/tree/RCBugFix. I just copied the recent RCBugFix branch and merged it with DyzeDesign High-Temperature-Thermistor-Support branch. I needed the 500C thermistor support.

Is there any simple way to solve it?

thinkyhead, 2016-06-24:
I'll take a look at that shortly. Thanks for testing `SINGLENOZZLE` as it is a very new addition.

Edie47, 2016-06-24:
Sorry, for the false bug report. I already solved the problem. I had to set DISABLE_INACTIVE_EXTRUDER to false.  My extruder has a single nozzle and a single stepper motor, with a servo mechanism that switch between filaments (extruders). So the stepper was always deactivated.

closed 2016-06-24 by Edie47

thinkyhead, 2016-06-25:
> I already solved the problem

Thank you for the update!

> single nozzle and a single stepper motor, with a servo mechanism that switches between filaments

I see! I have something better coming along for that type of extruder. See #3994

Edie47, 2016-06-25:
Thank you for your recommendation. Is it not intended just for extruders with two nozzles? I found that commit but I also found in SanityCheck.h that SWITCHING_EXTRUDER and SINGLENOZZLE are incompatibile.

I think it is like:
SINGLENOZZLE = 1 NOZZLE and 2+ STEPPER MOTORS
SWITCHING_EXTRUDER = 2 NOZZLES and 1 STEPPER MOTOR

My option is 1 NOZZLE and 1 STEPPER MOTOR, therefore I chosed SINGLENOZZLE as it is easier to implement stepper switching than one nozzle in SWITCHING_EXTRUDER.

thinkyhead, 2016-06-25:
> SWITCHING_EXTRUDER = 2 NOZZLES and 1 STEPPER MOTOR

Nope.

`SWITCHING_EXTRUDER` is specifically for an extruder that has:
- A single nozzle
- A single stepper that reverses its direction
- A servo that switches the side of the stepper used to drive the filament.

Edie47, 2016-06-26:
Then I missunderstood it from that thread.
Great, thank you. I try it.

Now I am using SINGLENOZZLE with DISABLE_INACTIVE_EXTRUDER false and INVERT_E1_DIR true. I also putted few rows of code to Marlin_main.cpp and it is working fine. :-) Pins of E0 and E1 I defined the same.

thinkyhead, 2016-06-26:
@Edie47 From my knowledge of Marlin stepper handling, I can see that those changes will work, but of course they are unsupported, and there will be some harmless side-effects, such as the E stepper motor pins being set twice in operations that deal with all the steppers at once, such as initializing, enabling, disabling, etc.

If you peek at the code changes for the switching extruder, you will see that surprisingly little needed to be changed to make it work. Mainly it separates the number of E steppers from the number of extruders, (which we needed to do anyways) so Marlin knows there is really only one E stepper. Among other things, this allows Marlin to assign the next unused E plug to some other function, such as a Dual Z axis or Dual X axis.

Edie47, 2016-07-08:
Hi, I tested the SWITCHING_EXTRUDER feature from #4163 for my extruder (single stepper and single nozzle) and it looks like it is not intended for this type of an extruder with a single nozzle.

1. The display is showing two nozzles (two temperatures).
2. For the T1 the MINTEMP error is triggered.
3. At each tool change, some Z-hop occurs what is actually not required.
4. I am missing a feature that allows to define a delay after the switch (for example 700ms) and then detaches the servo. ...the servo might wobble during print when it is active. This might lead to problems during print.

In the board definition, I could assign same pins for both T0 and T1 heater and thermistor to solve issue 2, but then issues 1 and 3 remain. Issue 4 is simple to implement in code.

For this reason, I prefer SINGLENOZZLE with hard coded servo switching mechanism. It works and it is much simpler.

thinkyhead, 2016-07-09:
> it looks like it is not intended for this type of an extruder with a single nozzl

Correct. It is for a dual-nozzle extruder with one stepper motor.

schimmi70, 2017-02-22:
why SINGLENOZZLE and SWITCHING_EXTRUDER can't be combinared ??
It makes sence ... or not ?

thinkyhead, 2017-02-26:
> why `SINGLENOZZLE` and `SWITCHING_EXTRUDER` can't be combin~~ar~~ed ??

`SINGLENOZZLE` is specifically for the case where multiple steppers feed into a single nozzle. The `SWITCHING_EXTRUDER` is for the specific case where the extruder has only a single stepper plus a servo for performing the switch.

schimmi70, 2017-02-27:
OK, I know this, but we need SINGLENOZZLE_SWITCHING_EXTRUDER for up to 2 Servors / 2 Stepmotors (4 Extruder) and single nozzle  So boards with only 2 extruder motor driver can support 4 filaments at same time... look here for 2 Filaments: http://www.thingiverse.com/thing:1643906 or http://www.thingiverse.com/thing:1888865

Both are SINGLENOZZLE SWITCHING_EXTRUDER and combinared both extruder-typs.

thinkyhead, 2017-02-28:
I see. That will be a `SWITCHING_EXTRUDER` (which is another type of "single nozzle") but allow `E_STEPPERS` to be set to 2 (or more) and allow specifying second (third, etc.) servo. Support for this kind of setup will require some rewriting of the core stepper-indirection code, and some re-write of tool-changing code to allow a tool-setting of "2", for example, to specify the 1st position of the 2nd extruder motor, rather than the 3rd extruder motor. It's not a trivial project, but I'm sure it's doable.

MrJulz, 2017-02-28:
Just seeking clarification: 
Is `SWITCHING_EXTRUDER` for a dual nozzle (servo-alternated), or a single nozzle?

Above is stated (15/June): "`SWITCHING_EXTRUDER` is specifically for an extruder that has: A single nozzle".
but then (8/July):. It is for a dual-nozzle extruder with one stepper motor.
but then (28/Feb/2017): That will be a `SWITCHING_EXTRUDER` (which is another type of "single nozzle")

Is there some subtlety in distinction I am missing?

----

I am seeking how to enable dual nozzles (servo actuated) #5921 , dual steppers - i.e exactly as with `SWITCHING_EXTRUDER` but without the "alternating-single-stepper" code introduced in #3994.  Any advice appreciated.

thinkyhead, 2017-03-01:
> Is there some subtlety in distinction I am missing?

Sorry for my conflicting statements. It's been a while since I worked on this code. Apparently "switching extruders" do require two nozzles. That said, if one exists which only has a single nozzle it can be made to work by setting the hotend offsets to zero and tweaking the code to only heat one nozzle.

thinkyhead, 2017-03-01:
> I am seeking how to enable dual nozzles (servo actuated) #5921 , dual steppers

Is the servo simply to raise and lower the nozzles when swapped?

Dual nozzles with dual steppers is essentially just `EXTRUDERS 2`. The only addition is you need to set the servo on tool-switch. You can just do that in GCode currently by using `M280`.

MrJulz, 2017-03-01:
Yes, but just the second nozzle.  The first nozzle is fixed in space, and the second nozzle simply slides vertically, to either above or below the first nozzle. Prevents collisions, minimal moving parts, fast, low weight.

schimmi70, 2017-03-16:
If you want, i open a brunch SINGLENOZZLE_SWITCHING_EXTRUDER i have a working version for 1 Hotend and 2 swichting extruder build from RC8 ... Please say me where i can open a new brunch.

thinkyhead, 2017-04-02:
> Where … can i open a new branch?

In your own fork.