MarlinFirmware/Marlin #4141: G29 Issue

Skyalchemist0, 2016-06-25:


Blue-Marlin, 2016-06-25:
What kind of bed leveling?
What branch?
From when?

Skyalchemist0, 2016-06-25:
Ok lets see if I can answer most of these questions.

Using an inductive sensor with 4point grid ABL.
Im not sure what version of marlin this is. 

Start G code;

G21 ; set units to millimeters
M104 S0 ; set temperature
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion

G28 
G29 ; Tramming

G28 X ; Heat over park
G28 Y

M190 S110 ; wait for bed temp
M109 S245 ; wait for temperature to be reached

G92 E0

Get read outs of
22:48:47.305 : Bed x: 15.00 y: 20.00 z: -0.25
22:48:51.459 : Bed x: 170.00 y: 20.00 z: 0.18
22:48:55.780 : Bed x: 170.00 y: 180.00 z: 0.11
22:49:00.139 : Bed x: 15.00 y: 180.00 z: -0.26
22:49:00.143 : Eqn coefficients: a: 0.00 b: -0.00 d: -0.27
22:49:00.147 : planeNormal x: -0.00 y: 0.00 z: 1.00
22:49:00.150 : echo:endstops hit:  Y:-0.07 Z:-0.25

But once print starts no compensation actually happens.

Blue-Marlin, 2016-06-25:
> G28
G29 ; Tramming
G28 X ; Heat over park
G28 Y

Don't use G28 after the G29. That does reset the levelling.

Skyalchemist0, 2016-06-25:
Thank you, Ill give this a try.

Roxy-3D, 2016-06-25:
The Bed Level Correction matrix doesn't tell you about the topology of the bed.  It only tells you about the tilt.    If you do a G29 V4 P5 T   You will get much more information about how flat (or not flat) your bed is.

Skyalchemist0, 2016-06-25:
I was using my eyes, after printed a giant X shape on my printer. The center does seem raised.

lavato, 2016-06-25:
I think I am experiencing the same issue, I also probe on 9 points. My bed is 200x200 and the center is a bit is a bit raised.  See example below:
N-normal/R-raised/@-probe. 

N@NN@NN@N
NNNRRRRNNN
N@RR@RR@N
NNNRRRRNNN
N@NN@NN@N

 I see compensation (e.g. Z moving up and down slightly) but it is widely off. E.g. the filament height is 0.3mm on the normal section and 0.01, I would say, on the middle "raised" part, as the head is pressing against the bed.

Roxy-3D, 2016-06-25:
If you have more than +/- 1/2 a layer thickness from the mean, you are going to have problems in those parts of the bed.   Mesh leveling is going to work better for you if your bed is warped or not flat.

Skyalchemist0, 2016-06-26:
Enabled mesh leveling in marlin, all I get is a "mesh leveling disabled" reading once I G29

lavato, 2016-06-26:
> more than +/- 1/2 a layer thickness from the mean

Can you elaborate on the above?

Blue-Marlin, 2016-06-26:
Lay a book on a ball. Till the book. Is there any possibility to minimize the average distance between book and ball? 
If your ball is big the average distance is smaller then if the ball is small. If the ball is big enough you can print everywhere because the maximum distance is small. With tilling the book you can't improve the average distance.
1/2 a layer is a rule of thumb.

Blue-Marlin, 2016-06-26:
If you press the plastic through the nozzle in free air the cross section is a circle. If you are at nozzle diameter above the bed the cross section is still a circle - the contact area to the bed is zero. If you are at level zero the the nozzle is clogged - no material can leave the nozzle. For that the layer height is supposed to be about 1/2 the nozzle diameter in the normal case. If the error is bigger than about 1/2 layer height = 1/4 nozzle diameter you likely see the one or the other effect in a disturbing magnitude.

Roxy-3D, 2016-06-27:
This is the same thing as Blue-Marlin is saying, but in different words.   The first layer needs the plastic to get squeezed a little bit as it comes out of the nozzle.  If the nozzle is more than an extra 1/2 layer above the glass, it probably won't get squeezed very much and it may not adhear to the glass.    Conversely, if the nozzle is too close to the glass you may damage the bed or the nozzle.

So...  a 1/2 layer error is about the most you want to have.  You will do much better if you can keep it down to a 1/4 layer.

psavva, 2016-06-28:
@Skyalchemist0 
Given the warp in the middle of your bed, Auto Bed Levelling isn't going to help you much.
Auto bed levelling is used for a surface with a single plane and tilt.

In your case, you have multiple planes that need to be compensated.

You need Mesh Bed Levelling. Check the config file on how to enable it.
Mesh leveling is somewhat manual now, but will be automated in a future release...
@Roxy-3D can confirm..

closed 2016-06-28 by thinkyhead

lavato, 2016-06-28:
Sorry for the late reply. 
Yes it does makes sense to use mesh.

In that case I would still ask what is the point of allowing the probe to take a 9 samples.

I would have thought if I have scenario like below , 0 being nominal + being raised, I would expect the linear compensation to work 

(0.0)(0.1)+(0.2)(0.2)+(0.1)(0.0)
(0.0)(0.1)+(0.2)(0.2)+(0.1)(0.0)
+++++++++++++++++++++++++
(0.0)(0.1)+(0.2)(0.2)+(0.1)(0.0)
(0.0)(0.1)+(0.2)(0.2)+(0.1)(0.0)
(0.0)(0.1)+(0.2)(0.2)+(0.1)(0.0)
(0.0)(0.1)+(0.2)(0.2)+(0.1)(0.0)
 x axis ->>

Blue-Marlin, 2016-06-28:
The original concept of levelling with a turn-matrix was measuring 3 points. That is theoretically sufficient to calculate the matrix. But some probes do not measure very exact. So in grid bed levelling more points are taken and averaged. Even with grid levelling the correction mechanism is still a turn-matrix so it relies on a theoretically flat (not warped) bed and a (theoretically) perfect printer geometry (except the tilt bed).   
Delta and mesh levelling are different. They interpolate between the measured points.

lavato, 2016-06-28:
Thank you - that totally makes sense.