MarlinFirmware/Marlin #4142: Feature Request - Auto turn power on and off

fabtopia, 2016-06-25:
if defined the PS_DEFAULT_OFF and want to move of heat i must think of switching the power on first. If i forget the system goes into error when trying to heat. Not odd but not user friendly.
Is it possible to detect de high power actions and automatically turn on the power and perhaps put it off (standby) after timeout?

thinkyhead, 2016-06-26:
If we can read the state of the `PS_ON_PIN` that would be one way. Otherwise we need to add a flag for the current state of the power supply. Then, whenever you initiated anything requiring power, whether it was stepper motion, heating, or turning on a fan, you could check and make sure to turn on the PSU first. It shouldn't be limited only to heating operations.

fabtopia, 2016-06-26:
the m80 is my first gcode too. but when using the display you have to turn it on first. that option is in  Sub menu and needs a lot of scrolling. thats not user friendly place.  when you forget and heatup to switch filament, Marlin goes into error and you have to reset. there is no warning. moveving is a litle less painfull. there is no movement and you can switch on without a reset.
so i thought in maybe auto poweron and timout standby is a nice feature. (will help  novice users with slicers as cure where it is more difficult to make M80 realy the first command, but that would be a nice site affect)

thinkyhead, 2016-06-26:
Of course if I'm playing a GCode file full of only `M300` commands (play a song) I don't necessarily want the PSU to be turned on automatically. There may be other instances of GCode that wants to do things that don't require power, though I admit they are rare.

ntoff, 2016-06-27:
Look at teacup firmware, pretty sure it handles power on and off automatically.

thinkyhead, 2016-06-27:
Yep. Teacup basically calls `power_on` every time it needs to enable the steppers. And it will call `power_off` after a configured inactivity timeout.

macst34, 2016-11-27:
I'd like this feature also. +1 if we're voting,

closed 2018-03-16 by thinkyhead