MarlinFirmware/Marlin #4143: No obvious development branch or communication method

birkett, 2016-06-25:
As a developer who has recently started working with the Marlin community, I have some concerns regarding the project structure.

Marlin development seems very fractured, with no clear way of accessing or submitting patches for the current development branch.

- The MarlinFirmware/MarlinDev repository is marked as deprecated, to be deleted soon.
- The MarlinFirmware/Marlin/1.0.x branch is no longer under active development.
- The RC branch only receives fixes from RCBugFix.
- RCBugFix is only for fixes between RC's.
 

To add to this, there has been a huge number of commits between RC's - leading to new code being added to branches that should be used for bug fixes. There is no clear general development branch.

I suggest changing the branches, and pull process.
- master rolling branch for day to day development, including new features / code.
- RC branch to be renamed to 1.1.x
- RC's are tagged and bug fixes applied directly to each major versions branch (1.0.x, 1.1.x, 1.2.x)
- Strict policy for locking down RC's, only bug fixes once tagged.

In addition to this, I would suggest an official communication method, for general questions and to follow the development. IRC, forums, or Discord?

That being said, how far along with 1.1.x are you? When can we expect RC7, or an actual release? Where is the development happening?

jbrazio, 2016-06-25:
A typical case of not searching before posting.If you search a bit, all your questions and points are discussed as open issues, nevertheless thank you for the advice.

The only question which is not answered is this one:
> In addition to this, I would suggest an official communication method, for general questions and to follow the development. IRC, forums, or Discord?

We're working a the official website marlinfw.org, that should be the official "non-technical" communication channel in the future. But as always, bleeding edge is our PR/Issue list.

Roxy-3D, 2016-06-25:
@birkett   Marlin has never had a 'Stable Branch' and 'Development Branch'.  We are trying very hard to get there.   Normally, there are very few changes to a 'Stable Branch'.   That is accepted as a 'Best Practice'.   We are deviating from that accepted norm because some of the highly valued and prolific contributors have said "We will have to live with and support the Stable Branch for a long while.  Let's clean up the spagettii code before we promote it to 'Stable'."    

I am normally in the camp that a Stable Branch (or Release Candidate) should have very minimal changes made to it.   They should be very minor and clearly correct changes.   As soon as we promote an Release Candidate to 'Stable' we will shift gears and behave in a more traditional manner.   But right now, there are going to be more changes than is typical.

With that said, we do have some very interesting Development branches planned, and as soon as we go 'Golden' you will get to play with them.

thinkyhead, 2016-06-25:
@Roxy-3D Please stop saying "Marlin has never had a stable branch" and mis-using the term "stable." I've already explained this before.

Version 1.0.2-1 is stable. It's not changing in any fundamental way, only bugs fixed.
Version 1.1.0, once released, will be stable. We will only be fixing bugs.

The term "stable" means we have settled the features and configuration options for the release.

Prior to version 1.0.1 Marlin had no concept of versioning or releases. So there wasn't anything "stable" there, for sure. But once we started tagging releases and following more methodical practices, we were able to put out what is called "stable" code.

If we're looking for a better term for "not crashing" then perhaps "bug-free" will do? Have we had any "bug-free" releases? Nope! And I doubt the next one will be. However, we have a huge list of bugs that were in 1.0.2-1 that we know will definitely not be in 1.1.0.

thinkyhead, 2016-06-25:
> I suggest changing the branches, and pull process.

@birkett

Yes, we plan to rename the RC branch to 1.1.x after the release. The "pull process" isn't very flexible on Github, as it automatically targets the "default branch." For this reason I created several shell scripts to help with the process. You can find them in the `buildroot/share/git` folder.

> how far along with 1.1.x are you?

Version 1.1.x is coming along very well, but please visit the issue queue to see if there are any outstanding bugs that you can help with.

> Strict policy for locking down RC's

We prefer to keep things a little looser. Participation and testing ebbs and flows here, and we are all volunteers. We get bug reports, feature requests, and plain old complaints, and we try to address them as best we can. Version 1.1.x is not "specified" so we are staying open to configuration changes, new features (disabled by default), code cleanup, and even comprehensive rewrites of existing features. All this is fine (with me, at least) as long as it produces a release that we can be proud of.

> Where is the development happening?

The old Development fork is going away, as it caused too much divergence from the next release work, at a time when the code needed many, many fixes. It was simply too soon to do this divergence, and a lot of that work was obsoleted by updates to the Arduino toolset.

So for current development, additions, bug fixes, etc. we're accepting PRs against `RCBugFix` now. This is so that we can keep these PRs up to date with the current code-base. I rebase my own on-hold PRs pretty frequently to make sure they stay in sync. The code that lives in `RCBugFix` today, after the 1.1.0 release, is going to form the basis for version 1.2.x development — but we will also be reorganizing the code into sub-folders at that time. We don't want to do that until after the release.

> an official communication method

The Issue Queue works very well, and allows us to track progress. Most of our communication relates to issues, after all. But Roxy, João, and I have been having semi-regular meetings on Google Hangouts to discuss the issues of the day as well.

birkett, 2016-06-26:
@jbrazio There was a significant amount of reading done before opening this. The open issues you refer to, there is only one. and it's closed (https://github.com/MarlinFirmware/Marlin/issues/3997). I opened this because I feel that #3997 had no tangible outcome. While there was clear discussion on the plans for future development branches, there was no resolution to the current issue. Developers may be turned away from contributing, because they cannot immediately see where the rolling development is happening, or follow it. Most follow a model where any branch marked as RC is strictly bug fix only.

Please don't misconstrued my repetition of points as ignorance. I certainly wouldn't waste my own, or indeed your time, by opening something that had I felt had no scope for discussion.

@Roxy-3D @thinkyhead Thank you for the concise explanations. It's good to see there is a solid plan moving forward.

jbrazio, 2016-06-26:
> Please don't misconstrued my repetition of points as ignorance. I certainly wouldn't waste my own, or indeed your time, by opening something that had I felt had no scope for discussion.

This has been so over discussed.. seriously.

This RC cycle it's not the "common" RC cycle people may expect, please keep this on the back of your minds. On this RC cycle we see a lot of changes which should not be part of an RC, we are no idiots.

We know we have a lot of issues from a Project Governance point of view.
We know Marlin has no significant major release for some time.
We had a lot of pending bug reports *-- changes, lots of them*
Marlin was lagging behind other firmware *-- new features*

Things will get more close to what you're used to after this release.

thinkyhead, 2016-06-26:
> Developers may be turned away from contributing, because they cannot immediately see where the rolling development is happening, or follow it. Most follow a model where any branch marked as RC is strictly bug fix only.

Actually, that was in some sense the "ideal" we had in mind when we started this whole RC process. But the sheer number of issues and half-done features was really a bummer. I don't think any of us expected to need to tag 7 releases to reach a point of satisfaction.

But, we are very close to that point of satisfaction! Actually, I'm very pleased with the way the new (user-contributed!) `M600` is looking. So there's another "broken feature" that will be much better in 1.1.0, because we took the extra time to consider it, vet it, test it, refine it, patch it, and work closely with the developers. I truly wish all the new features in the PR Queue could get merged and out to the community, in this release, because they fill some huge holes and I know we'd get some great feedback.

Then there's Documentation. Ideally, wouldn't it be great to have nearly complete documentation at the time of release? I'm not holding out for that, but I would like to see at least a page for each major feature. And I have a lot of work to do to finish the GCode section, especially with all the fancy new GCodes.

We love to get encouragement, suggestions, and useful criticism. But mostly we love material contributions, @birkett, so I hope that you feel passionate enough to lend your blood, sweat, and tears to those areas where we really need help.

closed 2016-07-16 by jbrazio