MarlinFirmware/Marlin #4144: Trying to update old FolgerTech 3d cloner Marlin firmware to latest but it uses a now deprecated function..

darknezz19, 2016-06-25:
The old firmware relied on' DISABLE_MAX_ENDSTOP' and without it my XYZ only go in one direction. With that function enabled the steppers work fine. Using this function in the latest firmware will not compile. Error log says 'Use individual USE_*_PLUG options instead.'

Found this and thought I had something.
https://github.com/MarlinFirmware/Marlin/issues/3573

#define USE_XMAX_PLUG
#define USE_YMAX_PLUG
#define USE_ZMAX_PLUG

Tried to add these to the coofiguration.h but now the steppers wont go either way. Says it's hitting the endstops when it's not. 

Any insight would be appreciated.

Roxy-3D, 2016-06-25:
Please move the extruder to the middle of the print envelope.   Verify that no endstop switches are being pressed.   Then give it a M119 and tell us what it reports.  It should not say any endstops are triggered.  If it does, we will need to invert those endstops in Configuration.h.

You should also be able to verify that when you press (and hold) an endstop switch and give it a M119 command you can see that particular endstop toggle its reported value.

Its too bad you aren't running a FolgerTech i3-2020.   You could just copy my setup.

darknezz19, 2016-06-25:
Hi Roxy. I tried what you said and indeed all 3 endstops are 'TRIGGERED' and when M119 is sent. Manually held one close and sent the command again, that single one was now 'open'

Do I change these to false?

const bool X_ENDSTOPS_INVERTING = true; // set to true to invert the logic of the endstops. 
const bool Y_ENDSTOPS_INVERTING = true; // set to true to invert the logic of the endstops. 
const bool Z_ENDSTOPS_INVERTING = true; // set to true to invert the logic of the endstops. 

I am worried that voltage will flip with ground and fry my endstops.

Here is my configuration.h in case it's needed.
http://pastebin.com/EQEPr0RD

darknezz19, 2016-06-25:
Okay Ive been trrying different coombination of of inveteing these three settings in hopes of fininding the right setting.

const bool X_ENDSTOPS_INVERTING = true; // set to true to invert the logic of the endstops.
const bool Y_ENDSTOPS_INVERTING = true; // set to true to invert the logic of the endstops.
const bool Z_ENDSTOPS_INVERTING = true; // set to true to invert the logic of the endstops. 

const bool X_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Y_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Z_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.

const bool X_MAX_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Y_MAX_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Z_MAX_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.

Have not found the correct combination yet.

What is the difference between these two commands as well?

//  #define ENDSTOPPULLUP_XMAX
//  #define ENDSTOPPULLUP_YMAX
//  #define ENDSTOPPULLUP_ZMAX
  #define ENDSTOPPULLUP_XMIN
  #define ENDSTOPPULLUP_YMIN
  #define ENDSTOPPULLUP_ZMIN

Roxy-3D, 2016-06-25:
> Do I change these to false?
> I am worried that voltage will flip with ground and fry my endstops.

No need to worry!   The 'Invert' only flips the logic, not the voltage levels.   Go ahead and invert any endstop that is saying 'Triggered' when it isn't pressed.   And it wouldn't hurt to verify that all 3 of them can be toggled manually by pressing them with your finger before you tell the machine to home.

darknezz19, 2016-06-25:
I was overthinking it.

Thought the stepper being aloud to move one way and not the other was not how it should work. Of course that's how it should work><

Thanks for the help Roxy:)

darknezz19, 2016-06-25:
Alright I got the X axis end stop working, goes only away when it's hit and goes home in the right direction.

The other two I just can't seem to get them to move only away from the end stop when it's closed. I'm still trying different combinations of these settings posted previously but maybe it's something else I'm missing?

thinkyhead, 2016-06-26:
If you have no max endstops then you need to comment out these lines:
```cpp
//#define USE_XMAX_PLUG
//#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG
```

darknezz19, 2016-06-26:
Thinkyhead finally just figured it out. There was duplicate entries for the flip direction function and I was editing the first one. All is well, time to test print:)

closed 2016-06-26 by thinkyhead