MarlinFirmware/Marlin #4145: M48 may crash in the bed under some circumstances....

lrpirlet, 2016-06-25:
I was "playing" with M48, in debug mode, when I noticed a different  predicted blocking move from the executed... see last 2 lines of the Repetier log...

```
21:17:12.232 : N94 M48 V4 X10 Y10*97
21:17:12.294 : echo:N94 M48 V4 X10 Y10*97
21:17:12.294 : M48 Z-Probe Repeatability test
21:17:12.294 : Positioning the probe...
21:17:12.294 : setup_for_endstop_or_probe_move > current_position: (0.00, 0.00, 0.07)
21:17:12.294 : probe_pt >>>
21:17:12.294 : > stow:0
21:17:12.294 : > current_position: (0.00, 0.00, 0.07)
21:17:12.310 : > do_blocking_move_to 10.00, 10.00, 5.00
21:17:12.310 : do_blocking_move_to: (10.00, 10.00, 0.07)
```
The probe_pt function contains the following with a value set for Z in the debug message but not in the actual instructions...

```C++
  static float probe_pt(float x, float y, bool stow = true, int verbose_level = 1) {
    #if ENABLED(DEBUG_LEVELING_FEATURE)
      if (DEBUGGING(LEVELING)) {
        SERIAL_ECHOLNPGM("probe_pt >>>");
        SERIAL_ECHOPAIR("> stow:", stow);
        SERIAL_EOL;
        DEBUG_POS("", current_position);
      }
    #endif

    float old_feedrate = feedrate;

    // Raise by z_raise, then move the Z probe to the given XY
    #if ENABLED(DEBUG_LEVELING_FEATURE)
      if (DEBUGGING(LEVELING)) {
        SERIAL_ECHOPAIR("> do_blocking_move_to ", x - (X_PROBE_OFFSET_FROM_EXTRUDER));
        SERIAL_ECHOPAIR(", ", y - (Y_PROBE_OFFSET_FROM_EXTRUDER));
        SERIAL_ECHOPAIR(", ", max(current_position[Z_AXIS], Z_RAISE_BETWEEN_PROBINGS));
        SERIAL_EOL;
      }
    #endif

    feedrate = XY_PROBE_FEEDRATE;
    do_blocking_move_to_xy(x - (X_PROBE_OFFSET_FROM_EXTRUDER), y - (Y_PROBE_OFFSET_FROM_EXTRUDER));

    #if ENABLED(DEBUG_LEVELING_FEATURE)
      if (DEBUGGING(LEVELING)) SERIAL_ECHOLNPGM("> deploy_z_probe");
    #endif
    deploy_z_probe();
```
I suggest adding the following...

```diff
diff --git a/Marlin/Marlin_main.cpp b/Marlin/Marlin_main.cpp
index b71d4f6..178fe31 100644
--- a/Marlin/Marlin_main.cpp
+++ b/Marlin/Marlin_main.cpp
@@ -2115,6 +2115,7 @@ static void clean_up_after_endstop_or_probe_move() {
     #endif

     feedrate = XY_PROBE_FEEDRATE;
+    current_position[Z_AXIS] = max(current_position[Z_AXIS], Z_RAISE_BETWEEN_PROBINGS);
     do_blocking_move_to_xy(x - (X_PROBE_OFFSET_FROM_EXTRUDER), y - (Y_PROBE_OFFSET_FROM_EXTRUDER));

     #if ENABLED(DEBUG_LEVELING_FEATURE)
@

```
I tested that modification.. It works for me...
Sorry, I will need to drop away from my computer and have probably no access to my PC before next week...I have no time to push a clean branch to submit a patch request...

Roxy-3D, 2016-06-25:
Actually...  The way the code used to be was this:
~~~~
    do_probe_raise(Z_RAISE_BEFORE_PROBING);
    feedrate = XY_PROBE_FEEDRATE;
    do_blocking_move_to_xy(X_probe_location - (X_PROBE_OFFSET_FROM_EXTRUDER), Y_probe_location - (Y_PROBE_OFFSET_FROM_EXTRUDER));
~~~~
In general, it is better to do the Z movement first without any X & Y movement.   This is because if the probe (or nozzle) is too low, it can drag across the bed if the X & Y are changing.   So it may be more 'correct' to do_probe_raise() first, and then the X/Y move.

thinkyhead, 2016-06-26:
That `Z_RAISE_BEFORE_PROBING` in `probe_pt` can't be there. We don't always want that to happen.

I modified `probe_pt` yesterday based on code from @AnHardt and clearly we missed something. The way `probe_pt` now works, it will always raise Z and deploy the probe if needed (after the XY move). But it will only stow the probe if flagged. If not flagged to stow, it will raise by `Z_RAISE_BETWEEN_PROBINGS` instead. So on exiting `probe_pt`, the probe should always be ready to move to the next probe point.

So, apart from the log being incorrect, does the probe fail to raise, deploy, or stow at the right times?

lrpirlet, 2016-06-26:
@Roxy-3D <https://github.com/Roxy-3D> , welcome… I had just enough time to hit the reset button while executing M48 just after a G29 S5 ( yes my motorized Mesh Bed level sort of works, willl push my local branch later when I got access to my PC)… The hot tip was pressing hard on the bed, I just could not let it go…

 

@AnHardt <https://github.com/AnHardt> , @thinkyhead <https://github.com/thinkyhead>  are we sure that we will NEVER enter probe_pt()with a distance higther than Z_RAISE_BETWEEN_PROBINGS so that it will not raize but lower to Z_RAISE_BETWEEN_PROBINGS ???

 

This is why I still suggest using the bigger value of EITHER current Z we enter the function with OR Z_RAISE_BETWEEN_PROBINGS… 

 

Anyway @thinkyhead <https://github.com/thinkyhead> , thanks for all that work, I do not know how you get the time to do all the good work you do…

 

L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

thinkyhead, 2016-06-26:
Should be better since yesterday's commits. The raise-between value ahead of the XY move actually does make sense in all cases.

lrpirlet, 2016-06-27:
@thinkyhead I have not the code in front of me... so Sorry if I make a mistake, I will verify tonight. Because that ```Z_RAISE_BETWEEN_PROBINGS``` is a value fixed in the Configuration.h, to the best of my understanding, this is how it goes...

- First  probe, with deploy - will raise to ```Z_RAISE_BETWEEN_PROBINGS``` then raise to ```Z_RAISE_BEFORE_PROBING```, then deploy the probe, then probe, then raise to ```Z_RAISE_BEFORE_PROBING + Z_RAISE_BETWEEN_PROBINGS``` 

-  Next probes - Z will be at ```Z_RAISE_BEFORE_PROBING + Z_RAISE_BETWEEN_PROBINGS``` ... I guess we do NOT want to Crash the probe on the bed. So we want to take the MAX of ```Z_RAISE_BETWEEN_PROBINGS``` and whatever set value for Z when we enter the function....

thinkyhead, 2016-06-27:
The function to raise the Z axis is smart. It won't allow double raise.

lrpirlet, 2016-06-27:
@thinkyhead Thanks, I should first read the code before discussing.... 

Yes of course, this ```do_probe_raise()``` function that you used, does calculate the max...

closed 2016-06-27 by lrpirlet

lrpirlet, 2016-06-27:
@thinkyhead @Roxy-3D I forgot to report that the new code works perfectly... Thanks.

If interrested M48 starts at 22:27:58.421 in the ```log files/Repetier.txt``` in my MMBL_2_LRP branch https://github.com/lrpirlet/Marlin.git

Sorry, I had no time to test anything else but G28 and M48 (and my Motorized Mesh Bed Leveling, that want to use probe_pt)