MarlinFirmware/Marlin #4150: SanityCheck.h error with autoleveling on prusa i3

AbuShaqra, 2016-06-26:
Hello, I am configuring my HIC prusa i3 to do the autoleveling, every time I activate it and try to upload the sketch I get the following error:

> SanityCheck.h:252:

 error: #error "The given FRONT_PROBE_BED_POSITION can't be reached by the Z probe."

         #error "The given FRONT_PROBE_BED_POSITION can't be reached by the Z probe."

          ^

exit status 1
#error "The given FRONT_PROBE_BED_POSITION can't be reached by the Z probe.<

this is my configuration for the probing points 
`#define LEFT_PROBE_BED_POSITION 15
    #define RIGHT_PROBE_BED_POSITION 170
    #define BACK_PROBE_BED_POSITION 180
    #define FRONT_PROBE_BED_POSITION 20`

and my probe offset is
`#define X_PROBE_OFFSET_FROM_EXTRUDER 10   
  #define Y_PROBE_OFFSET_FROM_EXTRUDER 57  
  #define Z_PROBE_OFFSET_FROM_EXTRUDER 0  `

I also noticed that the sanitycheck.h has 
` #if ENABLED(AUTO_BED_LEVELING_GRID)
    #ifndef DELTA_PROBEABLE_RADIUS`

my printer is not delta and I wonder what it has to do with this piece of code.
I need your help urgently T_T. thank you

Blue-Marlin, 2016-06-26:
Can you find the correct values if i tell you that im missing iportand data?
```
// Travel limits after homing (units are in mm)
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 200
#define Y_MAX_POS 200
#define Z_MAX_POS 100
```

thinkyhead, 2016-06-26:
Since your `Y_PROBE_OFFSET_FROM_EXTRUDER` is `57` the smallest Y coordinate your probe can possibly reach is 57. You should set your `FRONT_PROBE_BED_POSITION` to 57 or higher.

AbuShaqra, 2016-06-27:
these are my travel limits 
`// Travel limits after homing (units are in mm)
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 220
#define Y_MAX_POS 270
#define Z_MAX_POS 185`

but I have a problem that when I enable auto leveling, the printer travels directly to the middle of the  bed after homing x and y axis and takes it as the origin although I did not set it manually. when auto t leveling is disabled there is no problem and the printer goes to the home position at the lower left corner of the bed!

AbuShaqra, 2016-06-27:
there is also one more problem, the z values of my printer are multiplied by 10, so if I want to move the extruder 5mm in the z it travels 50mm but showns that it traveled 5!

thinkyhead, 2016-06-27:
> travels directly to the middle of the bed after homing x and y axis
Did you enable `Z_SAFE_HOMING` perhaps?

Side-question: What kind of probe do you have?

thinkyhead, 2016-06-27:
> z it travels 50mm but showns that it traveled 5

What is your Z setting for `DEFAULT_AXIS_STEPS_PER_UNIT`?

If your Z uses threaded rods with 0.8mm pitch (for example), then you would get:
```cpp
(200 * 16) / 0.8 = 4000
```

AbuShaqra, 2016-06-28:
I enabled `Z_SAFE_HOMING.
I use a sn04-n inductive sensor.

the last question I figured out that my Z axis steps per unit has to be 400 instead of the default 2560 and the Z travel became accurate after that.

closed 2016-07-16 by jbrazio