MarlinFirmware/Marlin #4151: Auto Bed Leveling  - Limits are incorrect

psavva, 2016-06-26:
I have upgraded marlin today after installing a new 300 x 300 glass bed.

Auto Bed Leveling isn't working as it should.
Problems faced is:
1) Not probing in the correct positions, as would be expected as X/Y Points on the Bed
2) After probing, (once I set some extreme values to get the probe on the bed), The Z is wrong.
It starts printing about 15 mm above the actual bed...
3) Z-Homing is enabled, but still doesn't home in the middle of the bed.
This is due to the negative Min-Pos values.

Full configuration on Pastebin:
http://pastebin.com/r2GBxdJ3

```

0----------------------------------0
|                                  | 
|        +---BED BACK -----+       |
|   BED  |       (+)       | BED   |
|    L   |        P        |  R    |
|    E   |                 |  I    |
|    F   |   (-)  N  (+)   |  G    |  
|    T   |                 |  H    |
|        |       (-)       |  T    |
|◄ 66mm ►|                 |       |
|        O---BED FRONT ----+       |
|                ▲                 |
|             75mm                 |
|                ▼                 |
X----------------------------------0

The outer lines is the main frame.
X on bottom left of the main frame is the Endstops for X and Y
P = Probe, N=Nozzle
Probe is 90mm behind the Nozzle
Marlin Configuration looks like so:

	
//============================= Z Probe Options =============================
#define X_PROBE_OFFSET_FROM_EXTRUDER 90  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 0  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -2.60   // Z offset: -below +above  [the nozzle]

// @section machine

// Travel limits after homing (units are in mm)
#define X_MIN_POS -75
#define Y_MIN_POS -66
#define Z_MIN_POS 0
#define X_MAX_POS 350
#define Y_MAX_POS 350
#define Z_MAX_POS 350

//
// Probe Raise options provide clearance for the probe to deploy and stow.
//
// For G28 these apply when the probe deploys and stows.
// For G29 these apply before and after the full procedure.
#define Z_RAISE_BEFORE_PROBING 15   // Raise before probe deploy (e.g., the first probe).
#define Z_RAISE_AFTER_PROBING 15    // Raise before probe stow (e.g., the last probe).
#define Z_RAISE_BETWEEN_PROBINGS 10  // Raise between probing points.

//===========================================================================
//============================ Bed Auto Leveling ============================
//===========================================================================

// @section bedlevel

#define AUTO_BED_LEVELING_FEATURE // Delete the comment to enable (remove // at the start of the line)
//#define DEBUG_LEVELING_FEATURE

#if ENABLED(AUTO_BED_LEVELING_FEATURE)

  // There are 2 different ways to specify probing locations:
  //
  // - "grid" mode
  //   Probe several points in a rectangular grid.
  //   You specify the rectangle and the density of sample points.
  //   This mode is preferred because there are more measurements.
  //
  // - "3-point" mode
  //   Probe 3 arbitrary points on the bed (that aren't collinear)
  //   You specify the XY coordinates of all 3 points.

  // Enable this to sample the bed in a grid (least squares solution).
  // Note: this feature generates 10KB extra code size.
  #define AUTO_BED_LEVELING_GRID

  #if ENABLED(AUTO_BED_LEVELING_GRID)

    #define LEFT_PROBE_BED_POSITION 75
    #define RIGHT_PROBE_BED_POSITION 340
    #define FRONT_PROBE_BED_POSITION -66
    #define BACK_PROBE_BED_POSITION 180

    #define MIN_PROBE_EDGE 10 // The Z probe minimum square sides can be no smaller than this.

    // Set the number of grid points per dimension.
    // You probably don't need more than 3 (squared=9).
    #define AUTO_BED_LEVELING_GRID_POINTS 2

  #else  // !AUTO_BED_LEVELING_GRID

    // Arbitrary points to probe.
    // A simple cross-product is used to estimate the plane of the bed.
    #define ABL_PROBE_PT_1_X 15
    #define ABL_PROBE_PT_1_Y 180
    #define ABL_PROBE_PT_2_X 15
    #define ABL_PROBE_PT_2_Y 20
    #define ABL_PROBE_PT_3_X 170
    #define ABL_PROBE_PT_3_Y 20

  #endif // !AUTO_BED_LEVELING_GRID

  //#define Z_PROBE_END_SCRIPT "G1 Z10 F12000\nG1 X15 Y330\nG1 Z0.5\nG1 Z10" // These commands will be executed in the end of G29 routine.
                                                                             // Useful to retract a deployable Z probe.

  // If you've enabled AUTO_BED_LEVELING_FEATURE and are using the Z Probe for Z Homing,
  // it is highly recommended you also enable Z_SAFE_HOMING below!

#endif // AUTO_BED_LEVELING_FEATURE

// @section homing

// The position of the homing switches
//#define MANUAL_HOME_POSITIONS  // If defined, MANUAL_*_HOME_POS below will be used
//#define BED_CENTER_AT_0_0  // If defined, the center of the bed is at (X=0, Y=0)

// Manual homing switch locations:
// For deltabots this means top and center of the Cartesian print volume.
#if ENABLED(MANUAL_HOME_POSITIONS)
  #define MANUAL_X_HOME_POS 0
  #define MANUAL_Y_HOME_POS 0
  #define MANUAL_Z_HOME_POS 0
  //#define MANUAL_Z_HOME_POS 402 // For delta: Distance between nozzle and print surface after homing.
#endif

// Use "Z Safe Homing" to avoid homing with a Z probe outside the bed area.
//
// With this feature enabled:
//
// - Allow Z homing only after X and Y homing AND stepper drivers still enabled.
// - If stepper drivers time out, it will need X and Y homing again before Z homing.
// - Position the Z probe in a defined XY point before Z Homing when homing all axes (G28).
// - Prevent Z homing when the Z probe is outside bed area.
#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_MIN_POS + X_MAX_POS) / 2)    // X point for Z homing when homing all axis (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_MIN_POS + Y_MAX_POS) / 2)    // Y point for Z homing when homing all axis (G28).
#endif

```

Blue-Marlin, 2016-06-26:
x is to the right, y to the back!

thinkyhead, 2016-06-26:
Let's see. Your X min is -75 and your Y min is -66. So the first thing I notice that your diagram is backward, with the X and Y min values reversed. The Y should be shown in front, and the X on the left.

Your probe is 90mm to the right of the nozzle. So…
- When your nozzle is at the left edge of the bed, at X=-75, the probe will be over a point where X=15.
- When your nozzle is over its farthest point, X=350, the probe will be over a point where X=440.

psavva, 2016-06-27:
@thinkyhead I have updated the diagram, I had it wrong way around.
@Blue-Marlin your comment has confused me.
Is my X And Y direction correct?
Ie: going left to right being X
Going front to back being Y?

After Homing all 3 axis
X=47, Y=142, Z=17.6

When X=0, then my nozzle is on the Left Edge.
When Y=0, then my nozzle is on the Front Edge.
Z = 17.6 because 15mm after probing  + 2.6 offset.
This is as expected.

I think you have read my diagram wrong with regards to the probe position.
If I'm looking from the front on the printer, then the  probe is directly behind the nozzle.
Ie when X=0 for the nozzle, X is also 0 for the probe.
When Y =0 for the nozzle, Y = 90 for the probe.

The reason I had to set back probe bed position to 180 is because the probe will move off the bed.
I'm thinking the reason may be that they the firmware isn't taking the .... oops found an config problem.
I used X_PROBE_OFFSET_FROM_EXTRUDER =90 instead of Y_PROBE_OFFSET_FROM_EXTRUDER =90

After Homing: X = -75, Y = -66 

I'll fix X_PROBE_OFFSET_FROM_EXTRUDER and try again...
Post back in a short while.

psavva, 2016-06-27:
After fixing the config above, I have turned on Debug on Bed Leveling.
I issued, a G28, G29, and M48 reputability test.
http://pastebin.com/1q408wtX

Also, please find the updated configuration attached:
http://pastebin.com/PWeTrhsK

Probing seems to be much more logical now.  I'll fine tune the probe offset measuring a little more accurately.

Z Height is OK now...

It's obviously only configuration issues I had initially.

Thank You @thinkyhead

closed 2016-06-27 by psavva