MarlinFirmware/Marlin #4152: Feature Request - About/Info page in LCD menu.

lintz, 2016-06-26:
Hi I have been swapping between different firmwares and dev/rc/stable branch back and forth and It's not always that I can turn on and off the printer too see which firmware is curently installed. Would be really nice with a about/info page in the lcd menu where some of the firmware info could be posted. 

Such as 
Machine name
Baudrate
Motherboard
Power supply
Thermistor and other good to have info.

thinkyhead, 2016-06-26:
Could be useful. In the meantime, you can use `M115` to obtain this info.

birkett, 2016-06-30:
I'll snag this one. Give me a few days on this, I already have something similar in one of my forks, that can be ported over to 1.1 and updated with some additional information.

birkett, 2016-07-03:
Developed in #4175
Merged in #4188

Enable LCD_INFO_MENU in Configuration_adv.h to enable the menu.
Enable PRINTCOUNTER in Configuration.h to show print counts.

closed 2016-07-04 by thinkyhead