MarlinFirmware/Marlin #4157: User voting on new features

adamfilip, 2016-06-27:
Lots of people (including myself) have put in feature requests.
perhaps it would help is users voted on new features to help prioritize future development.

Roxy-3D, 2016-06-27:
We do try to be responsive to any feature request.  Especially if it is made by multiple people.   

But right now, we are not in a position to respond to what ever direction the vote da jour goes.   Our plate is absolutely full.    We are hoping to get RC-6 (in reality it will probably be RC-7) promoted to 'Golden' status and that will take another month.     

After that happens we have several development branches already planned.   One of them will be dedicated to cleaning up the code base.  It is a big job and it won't be for the purpose of adding features.  It very possibly will help improve stability and reliability for the users.   The real purpose of the clean up is to make it easier for developers to do thier thing.

Another planned Development Branch will be a Unified Bed Leveling system based off of the 'Golden' Release.   That should have a lot of features that people have been asking for with regard to the Bed Leveling systems.    That will tie up a number of people for several months.

All of this takes time.    

With that perspective, I can tell you how to make your 'Vote' really count and get the features approved that you want.   The best way to 'Vote' for a feature is to care enough about it to design and develope it.  And then when it is working correctly, get it merged into the Marlin Code Base.   If you bring well thought out algorythms and clean code, your 'Vote' is very likely going to carry the day.

adamfilip, 2016-06-27:
I wish I had the skills or ability to be able to code, Ive tried learning it and it quickly goes over my head. Im thankful that there are so many skilled people like yourself that dedicate so much time to making Marlin better. 

Im looking forward to seeing where Marlin is one year from now. 

I know you guys have a roadmap of what you want to develop and the main features you want to incorporate to make Marlin great. The idea of the voting is really for stuff outside the current roadmap to get items captured and prioritized.

Thanks again for all your effort and time.

WZ9V, 2016-06-28:
Is there somewhere to see the roadmap?

I've totally lost track of features vs bugfix since RC6. Please don't interpret as complaint, it has just been so fast and furious I can't keep up with it.

Roxy-3D, 2016-06-28:
The two most important things on the road map is to get RC6 (or RC7) to a state where we feel good about promoting it to a 'Golden' release.   The truth is, we could do it now.  It is pretty solid.  But we aren't driven by a corporate agenda and it is ok if it takes another month to be both Rock Solid and a little bit cleaner.

The other big thing on the road map is a development branch to start re-organizing and cleaning up the code base.  To the user this isn't a very important goal.  But it is important in order to make it easier to add features and to keep the code base rock solid.     Most likely, as a direct result of this, you will see features get added quicker, and hopefully support for 32-bit processors.

thinkyhead, 2016-06-28:
> Is there somewhere to see the roadmap?

No, but we do have a list of priorities for after this release, mostly in our heads.

> I've totally lost track of features vs bugfix since RC6. Please don't interpret as complaint, it has just been so fast and furious I can't keep up with it.

I hear you there! When we tag RC7 there will be a long Changelog itemizing the changes. In the meantime, you can search merged Pull Requests to see everything that has gone in. It's a long list, but many of the changes which were very extensive (multiple PRs and dozens of commits) will end up being a single line-item in the Changelog.

WZ9V, 2016-06-28:
Thanks.  In my opinion, it would help if the developers published a list of the goals the are working on.  At the office my team uses a Kanban board and it has been quite effective in communicating with our users and coordinating fast paced changes but we are all in the same building.  It would be great if somehow a similar concept could be deployed because then your user community could see what was planned how far you are any additions, etc.  Something like this for the current release and a roadmap for the larger picture would go a long way.  It might even help get more people involved because they could pick off some of the documentation or smaller development tasks.  

I'd dive in but my C++ is very rusty (been management for too long) on the other hand if I knew what still needed documenting and its design, I could at least try and help there.

AnHardt, 2016-06-28:
Voting for or against a feature request is as easy as adding :+1:  or :-1:  to the top message of the thread.
But as long there is not a clear vote for a feature (for me - at least 9 votes with 2/3 majority) it will not have much influence on a developer. Convincing a developer about the usefulness would have much more effect. If any kind of 'exotic' hardware is involved, offering this to a developer may help to motivate him/her.

Step two is convincing someone who can merge the PR for the feature. Lots of "it works for me" comments do help, but testers (willing or able to test or at least review a PR) are rare.

jbrazio, 2016-06-28:
I like the idea, but as we are now it will not work because we do not have enough active programmers.

AnHardt, 2016-06-29:
It will not work if you close the FRs after 14 days of inactivity - sometimes it needs some month to have the right idea for the realization, or to find the time.

thinkyhead, 2016-06-30:
It may help to assign issues to milestones as they arrive, or instead of reflexively closing due to inactivity, especially when they are obvious features or fixes that we need to keep in mind.

jbrazio, 2016-07-03:
I believe the milestone assignment is something that must be done after a voting.
What we could do is crate a tracking list of some sort.. don't have enough time to think about this now, rushing to have the documentation ready. I'm marking this one a as a FR.

jbrazio, 2016-07-19:
The first step has been taken, a tracking list for all the feature requests:
http://www.marlinfw.org/docs/development/feature_request.html

thinkyhead, 2016-07-20:
The reason I favor "abusing" milestones is because it groups PRs and issues together. Perhaps the only way to do so within this interface. It doesn't actually have to represent an actual milestone.

closed 2017-10-31 by thinkyhead