MarlinFirmware/Marlin #4164: VIKI2 / u8glib Compile Error

thinkyhead, 2016-06-28:
With `VIKI2` enabled and u8glib installed (using the Arduino IDE Library Manager) my client is seeing the following error:
```
Arduino: 1.6.9 (Windows 7), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
In file included from sketch\ultralcd.cpp:48:0:
dogm_lcd_implementation.h:152: error: 'DOGLCD_CS' was not declared in this scope
   U8GLIB_NHD_C12864 u8g(DOGLCD_CS, DOGLCD_A0);
                         ^
dogm_lcd_implementation.h:152: error: 'DOGLCD_A0' was not declared in this scope
   U8GLIB_NHD_C12864 u8g(DOGLCD_CS, DOGLCD_A0);
                                    ^
exit status 1
'DOGLCD_CS' was not declared in this scope
```
What is causing this?

Blue-Marlin, 2016-06-28:
Post your configuration please. :-)
I suppose you try to compile for a RAMPS. For RAMPS these pins (as far as i know) never have been defined for VIKI2 - at least not in the splitted pins-file.

In the branch panucat provides (https://github.com/royco/Marlin/blob/Development/Marlin/pins.h)
if i read it correct
```
        #define DOGLCD_A0  44
        #define DOGLCD_CS  45
```
.

thinkyhead, 2016-06-29:
Ha, that explains it. They want to use an MKS_13 board, which is a RAMPS, with a VIKI2. I'm adding the pins now… but only in the MKS_13 file because some other RAMPS derivatives (like Azteeg X3) define them differently.

closed 2016-06-29 by thinkyhead