MarlinFirmware/Marlin #4166: if checksum == NL or CR, false NOCHECKSUM generated?

i-make-robots, 2016-06-28:
get_serial_commands() reads each line off the serial buffer until it hits a \r (13) or a \n (10) character.
commands follow the format

  N[line number] [gocde];*[single byte checksum][CR and/or NL]

if the checksum of the line is equal to 10 or 13, the line would be truncated prematurely and a NOCHECKSUM error would be generated.  If the PC software sends the exact same line again, an infinite loop would result.

A possible fix is to treat * as an escape character for the \n or \r check.

Blue-Marlin, 2016-06-29:
The assumption the checksum is a single byte is wrong - it's ASCII.

i-make-robots, 2016-06-29:
Ah.  Ok then.  Sorry!

closed 2016-06-29 by i-make-robots

Blue-Marlin, 2016-06-29:
I admit this is a bit hidden
```
        if (apos) {
          byte checksum = 0, count = 0;
          while (command[count] != '*') checksum ^= command[count++];

          if (strtol(apos + 1, NULL, 10) != checksum) { //!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
            gcode_line_error(PSTR(MSG_ERR_CHECKSUM_MISMATCH));
            return;
          }
          // if no errors, continue parsing
        }
```
And somewhere is a misleading comment.
```
   /**
     * '#' stops reading from SD to the buffer prematurely, so procedural
     * macro calls are possible. If it occurs, stop_buffering is triggered
     * and the buffer is run dry; this character _can_ occur in serial com
     * due to checksums, however, no checksums are used in SD printing.
     */
```
That's wrong. '#' can't appear in the checksum.

spilz87, 2020-03-05:
hello

can someone explain to me what is wrong when i try to send it after a M28 command
`N1 ;TIME:10052*125`
I receive : 
` Error:No Checksum with line number, Last Line: 0
    Resend: 1
    ok
`
I don't understand, there is the line number N1 ther is the '*' and the checksum

why it seems to don't find the checksum ??

ellensp, 2020-03-05:
A) This is a closed bug report not a support forum. 
B) everything after the ; character is a comment. and is dropped. Should not be sent to controller.

spilz87, 2020-03-05:
Thank for your very quick reply

A) I know it’s not a support forum but there is no information in documentation about this function. As I worked 2 days to find why it doesn’t work, I could think there is an issue and open a new one. Before doing that I prefer asking somewhere people maybe know if it’s or not an issue.

B) it’s not obvious : the documentation explain M28 command is to send a file and the example is a .txt file. So the transfer shouldn’t care about what is in the file. You can want to send a config file or to keep comments in gcode file to use them later. I don’t really understand this choice.

Anyway thanks a lot, I will not find it alone

AnHardt, 2020-03-05:
I'd start with a search for the problem by reverting #16840.

thinkyhead, 2020-03-06:
@spilz87 — Here are the resources you should use instead of hijacking our issue queue:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y