MarlinFirmware/Marlin #4170: Mesh bed levelling not working anymore (RCBugFix)

Alex9779, 2016-06-29:
I am trying to get the RCBugFix of today but I can't get it working anymore since the new probe options in the config.
The problem is that after homing the bed is moved to Z=5. That is ok but when I start the mesh bed levelling the machine homes and rests at Z=5. Then I click to start and the machine does the Z hop move to the first position, undoes the Z hop but doesn't move to Z=0 instead goes back to Z=5.
Z=5 is not yet 5mm away from the nozzle but about 7mm and the correct is limited to 4mm so I can't get the bed to the nozzle.
Do I have to change something in my config?

```
//===========================================================================
//============================= Z Probe Options =============================
//===========================================================================

//
// Probe Type
// Probes are sensors/switches that are activated / deactivated before/after use.
//
// Allen Key Probes, Servo Probes, Z-Sled Probes, FIX_MOUNTED_PROBE, etc.
// You must activate one of these to use AUTO_BED_LEVELING_FEATURE below.
//
// Use M851 to set the Z probe vertical offset from the nozzle. Store with M500.
//

// A Fix-Mounted Probe either doesn't deploy or needs manual deployment.
// For example an inductive probe, or a setup that uses the nozzle to probe.
// An inductive probe must be deactivated to go below
// its trigger-point if hardware endstops are active.
#define FIX_MOUNTED_PROBE

// Z Servo Probe, such as an endstop switch on a rotating arm.
//#define Z_ENDSTOP_SERVO_NR 0
//#define Z_SERVO_ANGLES {70,0} // Z Servo Deploy and Stow angles

// Enable if you have a Z probe mounted on a sled like those designed by Charles Bell.
//#define Z_PROBE_SLED
//#define SLED_DOCKING_OFFSET 5 // The extra distance the X axis must travel to pickup the sled. 0 should be fine but you can push it further if you'd like.

// Z Probe to nozzle (X,Y) offset, relative to (0, 0).
// X and Y offsets must be integers.
//
// In the following example the X and Y offsets are both positive:
// #define X_PROBE_OFFSET_FROM_EXTRUDER 10
// #define Y_PROBE_OFFSET_FROM_EXTRUDER 10
//
//    +-- BACK ---+
//    |           |
//  L |    (+) P  | R <-- probe (20,20)
//  E |           | I
//  F | (-) N (+) | G <-- nozzle (10,10)
//  T |           | H
//    |    (-)    | T
//    |           |
//    O-- FRONT --+
//  (0,0)
#define X_PROBE_OFFSET_FROM_EXTRUDER 14.75  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 34  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -4   // Z offset: -below +above  [the nozzle]

// X and Y axis travel speed (mm/m) between probes
#define XY_PROBE_SPEED 8000

//
// Allen Key Probe is defined in the Delta example configurations.
//

// Enable Z_MIN_PROBE_ENDSTOP to use _both_ a Z Probe and a Z-min-endstop on the same machine.
// With this option the Z_MIN_PROBE_PIN will only be used for probing, never for homing.
//
// *** PLEASE READ ALL INSTRUCTIONS BELOW FOR SAFETY! ***
//
// To continue using the Z-min-endstop for homing, be sure to disable Z_SAFE_HOMING.
// Example: To park the head outside the bed area when homing with G28.
//
[... 210 lines cut ...]

// Manual homing switch locations:
// For deltabots this means top and center of the Cartesian print volume.
#if ENABLED(MANUAL_HOME_POSITIONS)
  #define MANUAL_X_HOME_POS 0
  #define MANUAL_Y_HOME_POS 0
  #define MANUAL_Z_HOME_POS 0
  //#define MANUAL_Z_HOME_POS 402 // For delta: Distance between nozzle and print surface after homing.
#endif

// Use "Z Safe Homing" to avoid homing with a Z probe outside the bed area.
//
// With this feature enabled:
//
// - Allow Z homing only after X and Y homing AND stepper drivers still enabled.
// - If stepper drivers time out, it will need X and Y homing again before Z homing.
// - Position the Z probe in a defined XY point before Z Homing when homing all axes (G28).
// - Prevent Z homing when the Z probe is outside bed area.
#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT (14.75)    // X point for Z homing when homing all axis (G28).
  #define Z_SAFE_HOMING_Y_POINT (34)    // Y point for Z homing when homing all axis (G28).
#endif
```

epatel, 2016-06-29:
@Alex9779 I hear you, but I have a hard time understanding what is going on. Sry. Though, 2 things. You have `Z_RAISE_BETWEEN_PROBINGS` set which changes MBL in a way I am not familiar with (meaning I have not investigated how or what that does, but I think I can imagine what it is supposed to do). Then `X_MIN_POS` looks strange, shouldn't really matter but it looks off, but maybe your endstop is placed outside the bed like that?

Alex9779, 2016-06-29:
`Z_RAISE_BETWEEN_PROBINGS` is what it was set to when the setting appeared, I just changed `Z_RAISE_PROBE_DEPLOY_STOW`because my probe does not need clearance when deployed, it is fixed...
`X_MIN_POS` is ok, my endstop is off the bed and that is the position where the nozzle is when the endstop is hit compared to 0,0 of the bed...

So you need more details?

Alex9779, 2016-06-29:
Oh BTW I think I tried to set `Z_RAISE_BETWEEN_PROBINGS`to 0 but without any effect on what happens, the bed rests at Z=5 after homing and the leveling sequence behaves the same...

epatel, 2016-06-29:
MBL doesn't use the probe, it is more of a manual process. Nothing is deployed or stowed.

With the endstops being outside of the bed one might need to manually modify the "inset" and how that is added. Right now it is uniform front/back, left/right. But it seems for your use case a longer inset for the "left" side (same side as the X endstop).

Z behaviour sounds very strange.

epatel, 2016-06-29:
Do you have a "probe" configured?

Alex9779, 2016-06-29:
I am using the exact position setting in adv for mesh bed leveling. The only problem I have is the Z...

Alex9779, 2016-06-29:
When the stow is set to the default value of 15 the bed moves much more down when homing... And also when doing the manual leveling sequence...

Alex9779, 2016-06-29:
I have `FIX_MOUNTED_PROBE` activated...

epatel, 2016-06-29:
Hmmm...could be that causes G28 to go into this part

https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Marlin_main.cpp#L2834

Which confuses MBL.

epatel, 2016-06-29:
Note that MBL "calls" G28 a couple of times.

epatel, 2016-06-29:
I recommend to disable the probe if that is possible. MBL is currently a manual process via the LCD. @Roxy-3D is working on adding auto-mesh probing feature, and more.

Alex9779, 2016-06-29:
Ok I am going to try that and report...

epatel, 2016-06-29:
See the *Probing the bed with the display* part here https://github.com/epatel/Marlin/blob/75f9fb50c4002f6d4ba0345f5594cbac1fab1fa6/Documentation/MeshBedLeveling.md

Alex9779, 2016-06-29:
Ok I disabled `FIXED_MOUNTED_PROBE` and `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`.
It is working again... So as I assumed it was a config problem...
Thanks for clearing it up, it seems I don't need the probe settings, though the probe offset are taken into account for the safe Z homing positions. I could set them all to 0 and just set safe Z homing to where the nozzle should be, couldn't I?

epatel, 2016-06-29:
Great it works again. For the other parts, I am not sure, nothing I know how it works, never used :-S

Alex9779, 2016-06-29:
Ok no problem... TYVM!

closed 2016-06-29 by Alex9779

stevewatts, 2017-06-01:
did you manage to get any of the auto bed-levelling working without the fixed probe?  I just see  "Auto Bed Leveling requires a probe! Define a Z Servo, Z_PROBE_ALLEN_KE..........'