MarlinFirmware/Marlin #4174: my 3D printer based on STM32 is working wrong

lin6328, 2016-06-30:
My 3D printer is working now,it can control by cura.But it sometimes will work wrong .I do not check the code line.Is the problem in here?

thinkyhead, 2016-06-30:
You'll have to provide more detail.

- What version of Marlin are you using?
- How exactly is it failing?

lin6328, 2016-07-01:
Marlin use 'N' to counter the gcode line.If the gcode is wrong,marlin will reply the wrong line to cura.But I do not use 'N' to counter.I think cura send the gcode one by one , it should not to be wrong.So whar is problem? @thinkyhead

thinkyhead, 2016-07-01:
If you're using 250000 baud rate, try 115200 and see if it's any better.

lin6328, 2016-07-02:
I am using 115200 baud rate,it sill work a shot time and then it stop.

thinkyhead, 2016-07-02:
@lin6328 We are aware of possible comm issues due to potentially higher demand from the stepper functions.

- Are you able to print from SD? Do you see the same issue when doing so?
- Is this a DELTA printer?
- Do you use a Graphical LCD?

Please be sure to use the `RCBugFix` branch for all your testing, so we are both working with the latest code.

AnHardt, 2016-07-02:
STM32 is a 32bit processor. For sure he can't use one of out branches.

lin6328, 2016-07-02:
I rebuild it on stm32.Do not worry about it

thinkyhead, 2016-07-02:
@lin6328 Please make a ZIP of all your Marlin source code and post it for us to examine. (Or, better, publish it as a new branch in your Marlin fork.)

closed 2016-07-11 by jbrazio