MarlinFirmware/Marlin #4176: Servo move on tool change

jaredabrandt001, 2016-06-30:
Hi,
I'm programming a servo to lift an extruder while not in use.  This is a dual extruder setup, with a servo connected to lift the inactive extruder.  I'm running the latest RC6 Marlin and I see M280 controls servo's but I'm running into an issue:
Slic3r is my slicer, which has a section for "tool change g-code" which runs every time a tool change occurs.  When T1 (the 2nd extruder) is active the servo needs to lower.  When T0 (first extruder) is active the servo needs to raise.  An IF statement would do the trick, but it's my understanding that Marlin doesn't support conditional statements, is this correct?  If Marlin does not interpret conditional statements, is there a way to program the firmware to initiate a tool change routine to move the servo whenever Marlin receives the T1 or T0 command?
Thanks in advance!
Jared

Blue-Marlin, 2016-06-30:
Somewhere in Marlin_Main.cpp, gcode_T(), #if HOTEMDS == 1
```
if (tmp_extruder == 0)
  MOVE_SERVO(servo_index, servo_position0);
else
  MOVE_SERVO(servo_index, servo_position1);
```

jaredabrandt001, 2016-06-30:
Yes, this is exactly what I'm looking for!

Below here is a snippet of Marlin_Main.cpp.  Admittedly I'm not a great programmer.  I'm not sure if this is the right spot to put the code in; the code may be a bit "over my head".  This code below begins at line 6284 .  Is this a good spot in the code do you think?

inline void gcode_M999() {
  Running = true;
  lcd_reset_alert_level();
  // gcode_LastN = Stopped_gcode_LastN;
  FlushSerialRequestResend();
}

/**
 * T0-T3: Switch tool, usually switching extruders
 *
 *   F[mm/min] Set the movement feedrate
 */

// Begin lifting cold extruder
if (tmp_extruder == 0)
  MOVE_SERVO(0, 32);
else
  MOVE_SERVO(0, 0);
// End lifting cold extruder

inline void gcode_T(uint8_t tmp_extruder) {
  if (tmp_extruder >= EXTRUDERS) {
    SERIAL_ECHO_START;
    SERIAL_CHAR('T');
    SERIAL_PROTOCOL_F(tmp_extruder, DEC);
    SERIAL_ECHOLN(MSG_INVALID_EXTRUDER);
  }
  else {
    target_extruder = tmp_extruder;

    #if EXTRUDERS > 1
      bool make_move = false;
    #endif

jaredabrandt001, 2016-06-30:
well, it compiles at least like this below.  I will test the code in a few days or so.

I will also unfortunately need a Z offset when T1 is active.  It looks like Marlin supports this, but unfortunately I only see that Slic3r or Cura has X and Y offsets.
I really need this Z offset for T1 to be set-able by gcode, at the beginning of the program (not at each tool change).  Looking at the gcode for a dual extruder print with X offsets, I don't see where this happens, but it must be here somewhere.  Would you be so kind as to point me in the right direction?  

/**
 * T0-T3: Switch tool, usually switching extruders
 *
 *   F[mm/min] Set the movement feedrate
 */

inline void gcode_T(uint8_t tmp_extruder) {
  if (tmp_extruder >= EXTRUDERS) {
    SERIAL_ECHO_START;
    SERIAL_CHAR('T');
    SERIAL_PROTOCOL_F(tmp_extruder, DEC);
    SERIAL_ECHOLN(MSG_INVALID_EXTRUDER);
  }
  else {

  // Begin lifting cold extruder
  if (tmp_extruder == 0)
    servo[0].move(0);
  else
    servo[0].move(45);
  // End lifting cold extruder
    
    target_extruder = tmp_extruder;

    #if EXTRUDERS > 1
      bool make_move = false;
    #endif

    if (code_seen('F')) {

      #if EXTRUDERS > 1
        make_move = true;
      #endif

jaredabrandt001, 2016-06-30:
Would this be M218 T1 Z-0.5   to properly set the T1 Z offset different from T0?
In the Marlin cheat sheet it just shows X and Y, but it sure looks supported in the firmware:)
But for some reason it's not included in my dual color print g code, which processes an X offset for T1 correctly.  It appears X is handled a little different in Marlin...I dunno?

thinkyhead, 2016-07-01:
You cannot set a different Z offset for your nozzles, currently.

Look at #3994 to see how I implemented both Z lift and servo articulation for an extruder that has a servo for switching the side of the filament feed. You should find everything you need by looking at those changes.

jbrazio, 2016-07-16:
Closing due to inactivity.
Please re-open this issue if still valid.

closed 2016-07-16 by jbrazio

MrJulz, 2017-03-02:
Hi Jared,

We found a solution to this, currently at: #5921 

Note: no tool-change gcode is required in your slicer,  simply "T1" or "T0" and thinkyhead's code will automatically take care of the rest. (Servo actuation, Z_Offset, etc)

BigBot-3D, 2017-03-02:
Right on, thanks!  I will give it a shot soon enough!