MarlinFirmware/Marlin #4177: Setting Z_RAISE_AFTER_PROBING to 0 ignores G92 after G29

radry, 2016-06-30:
When setting Z_RAISE_AFTER_PROBING to 0 any G92 command after completing G29 is ignored.

What I'm trying to do: After G29, my start script will set "G92 Z0.35" to set the correct nozzle height.
So to make this work I now (1.1.0 - RC6) have to set Z_RAISE_AFTER_PROBING to a positive value and then run G0 Z0 to get back down. Only then it accepts G92.

Probe is fixed.
Z_PROBE_OFFSET_FROM_EXTRUDER  is set to 0. (No, I don't want to adjust the gap via this variable.)

It worked fine in 1.1.0-RC3 because in that version Z_RAISE_AFTER_PROBING was ignored for some reason.

thinkyhead, 2016-06-30:
> Z_PROBE_OFFSET_FROM_EXTRUDER is set to 0. (No, I don't want to adjust the gap via this variable.)

But that's what this setting is for. Your anomalous behavior is probably because you're abusing `G92`. You can't move any lower than the physical limit (`Z_MIN_POS`) you have set, and `G92` actually updates the software endstops so they are still in the same relative position.

Why don't you want to set `Z_PROBE_OFFSET_FROM_EXTRUDER` to `-0.35`? If you do, Marlin should just "do the right thing."

radry, 2016-07-01:
G92 Z0.35 has no effect. The Z cordinate doesn't get updated to the new value and remains at 0. Thus I can't go any lower obviously because of the software endstop limit. If Z_RAISE_AFTER_PROBING is above 0, the Z cooridnate gets updated to the specified value.

G92 should set the current Z position to the specified value. It was always like that. So if I set G92 Z0.35 the FW should set the current Z position as 0.35 (but not move the axis) and thus I can move 0.35mm lower up to the "new" 0. (Which is -0.35 in the pre-G92 config). I hope that makes sense.

I think this should be marked as a bug. Since it works when Z_RAISE_AFTER_PROBING is above 0.

> Why don't you want to set Z_PROBE_OFFSET_FROM_EXTRUDER to -0.35?
If you did, Marlin would just "do the right thing."

If I change the extruder/nozzle the offset changes. Also different hotend temperatures result in a different offset (because of the heat induced extension of the nozzle). I would have to reflash the FW every time. It's much easier to adjust the start script in the slicer software to the right valvue with G92. I like it this way. It's fast and efficient for me.

Blue-Marlin, 2016-07-01:
@radry 
What is more efficient in `G92` than in `M851`?

radry, 2016-07-01:
The issue is not about how I do things. It's about a bug. A setting that causes a command not working for an unknown reason. 
Maybe issuing G0 Z0 after G29 will make G92 work afterwards, even if Z_RAISE_AFTER_PROBING is 0. I'll have to try it.

DjDemonD, 2016-07-01:
There was a method of determining z offset by sending g92 z10 after homing z. This was to circumvent the travel limits to allow offset measurement using the paper method. However I tried this using bugfix the other day and as thinkyhead points out it won't go below the travel limits even after a g92 z10 with the nozzle just above the bed. 

However you can temporarily disable software endstops or set them to allow a couple of mm below z=0. Or use babystepping while printing a test object which now reports the change in z height in mm. Before the next print use m851 to create a z offset, m500 to save. Perhaps we ought to ask Thomas Sanladerer on his youtube channel to update his video on implementing ABL as a lot of people use this guide where the g92 z10 method is advised.

radry, 2016-07-01:
So the problem is that the software endstop limit prevents G92 to go lower than 0. That means G92 is broken, since the purpose of G92 is to set absolute zero.

DjDemonD, 2016-07-01:
I'll not answer this directly as I am sure one of the Marlin dev's will have the correct answer but I presume G92 does not allow travel outside the travel limits for a good reason, which might not be immediately apparent. There is another method (above) if what you are doing is determining z offset.

thinkyhead, 2016-07-01:
Looking at the code for `G92` I can see no reason why it would not update the current position. In fact, it _always_ updates `current_position`.

AnHardt, 2016-07-02:
@DjDemonD 
Making the software endstops selectively switchable by g-code is still on my todo-list.
The probe cleanup project grow and grow but will be finalized soon (I hope). 

Well - let's look at the name - END STOP. 
The more interesting question is - where is the end. It should clearly __not__ be below the bed. And if the place called zero is below the bed you can't reach zero.

So - where is the bed? At the moment the probe triggers the bed is `Z_PROBE_OFFSET_FROM_EXTRUDER` mm away from the nozzle - that is the definition for `Z_PROBE_OFFSET_FROM_EXTRUDER`. 
'physical NULL' is now `Z_PROBE_OFFSET_FROM_EXTRUDER` mm below the nozzle - the software endstop is adjusted to this (homing).
Now we can add different offsets to adjust the number for that point. 
These offsets are also applied to the number we stored for the software endstop. So the endstop should always trigger at 'physical NULL'.     
Consequence is - if your  `Z_PROBE_OFFSET_FROM_EXTRUDER` is set to 0, you cant go below the probes trigger point. 
For adjusting the `Z_PROBE_OFFSET_FROM_EXTRUDER` this means - begin with a too small number (usually the number is negative). For example -10 if you expect -5 to be about correct. Now do the paper test. Be careful - you can go below the bed now. Then correct the value by the number in the display (if the other offsets are zero). If you read +6 the correct value is -10 + 6 = 4.

(At least - that's the plan)

radry, 2016-07-02:
Thank you for your explanation. My question is still: What's the purpose of G92 in this case? G92 is supposed to set "physical NULL", as you call it. So why can't it overwrite what the auto bed level determined?

thinkyhead, 2016-07-02:
Whatever the "purpose" of `G92` is, its effect is to redefine what the current position of the nozzle is.

So, for example, let's say you have typical software endstops, where you can't go below zero in X, Y, or Z. Now, let's say you move your carriage with `G1 X100` and then issue `G92 X0`. What do you expect to happen if you then issue `G1 X-100`? Do you expect the carriage to move left or to be limited?

If we didn't change the software endstop to `-100` in response to the `G92 X0` command you would be unable to move left. Earlier versions of Marlin behaved this way, leaving the software endstops unaltered. Feedback from the CNC community convinced us that this was a bug, so we have fixed it.

We could certainly extend `G92` to take an additional argument, such as `S1` to leave the software endstops unaffected. But that should not be the default behavior.

radry, 2016-07-02:
> Now, let's say you move your carriage with G1 X100 and then issue G92 X0. What do you expect to happen if you then issue G1 X-100? Do you expect the carriage to move left or to be limited?

Of course it shouldn't move left after that command. Why? Because the software endstop is X=0 and the carriage after `G92 X0` is at X0. If the endstop is not updated you could move to X=-100, which doesn't make sense, since the endstop is explicitly set to 0.
So if I move the nozzle `G1 Z1` then issue `G92 Z2`, it MUST be able to travel -2 on Z to the new 0.

thinkyhead, 2016-07-02:
@radry Sorry, but that is no longer the behavior of `G92` in Marlin. We keep the software endstops in the same physical positions, no matter what you do with `G92`.

Otherwise (in the above situation) you would also be allowed to move to X=200, past the physical limits of the machine, grinding the stepper motor, which is what the software endstops are there to prevent.

The way you're trying to use `G92` is simply not supported. If you want to be able to move below the bed, you should set your `Z_MIN_POS` to `-2` and then set your `Z_HOME_POS` separately to `0`. When you home, your Z will then be `0`, but you will still be allowed to move down by 2mm.

jstefanop, 2016-07-06:
Whats the difference between M206 and M851? Since we shouldn't use G92 for Z probe offset tweaking whats the better option?

thinkyhead, 2016-07-06:
**M851** sets the Z distance between the nozzle and probe. (i.e., the negative of the position of the nozzle when the probe is triggered). This is the one and only supported option to get your machine to properly assign the correct Z position when you probe the bed.

**M206** sets elements in the `home_offset` array and updates the current position based on the difference from the old `home_offset` values. After using `M206` and homing with `G28`, the current coordinates will be your configured home (min or max) modified by values given in `M206`. Thus if you normally home XY to { 0, 0 } then after `M206 X10 Y4` your next `G28` will set the XY coordinates to { 10, 4 }. If your nozzle was at { 100, 100 } it will now be at { 110, 104 ). `M206` has nothing to do with probing, per se. It merely sets the origin of the coordinate system.

jstefanop, 2016-07-06:
Yea I think the confusion with these commands is mostly around ABL and how many printers use an IR probe for both ABL and the Z endstop. Since these IR probes are finicky we can't rely on just Z_PROBE_OFFSET_FROM_EXTRUDER to be set in firmware and forgotten. This offset needs to be changed all the time between different bed treatments, layer heights, lighting, etc. 

Up until RC3 G92 Z1.1; was the go-to way to do this after homing and ABL. Since my IR probe currently triggers when my extruder nozzle is 1.1mm above the bed. 

Going forward it looks like both M852 Z-1.1; and M206 Z1.1; should do the same thing?

thinkyhead, 2016-07-06:
> Going forward it looks like both M852 Z-1.1; and M206 Z1.1; should do the same thing?

No, they don't do the same thing.

jstefanop, 2016-07-08:
Ok but its two different ways of doing the same thing no? The IR prob triggers 1.1mm before the hottend tip reaches the bed surfaces, so the "trigger" in traditional sense is -1.1mm offset from hotend tip. 

Setting M851 Z-1.1 tells the firmware this exact thing, so after homing G0 Z0 should bring the nozzle to touching the bed. 

Im assuming the same thing can be achieved by sending M206 Z1.1 AFTER homing, setting the Z coordinate system offset by 1.1mm so it knows when the Z prob triggered it actually trigger 1.1mm above the bed and not at Z0.

Am I understanding this incorrectly?

thinkyhead, 2016-07-09:
> Am I understanding this incorrectly?

I think so! If you use `M206 Z1.1` now the bed is not at Z=0 anymore. The bed is now at Z=1.1. You can no longer move the nozzle to 0 (which is now 1.1mm below the bed). The minimum Z coordinate where you can move it will now be 1.1.

closed 2016-07-16 by jbrazio