MarlinFirmware/Marlin #4178: avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x66

nkuck, 2016-06-30:
Get this message regardless of what I try.
Zonestar Prusa i3 Melzi board ATMega 1284P.
Anyone solved this? Thanks.

jbrazio, 2016-06-30:
That error happens when the Arduino IDE cannot communicate with the board.
Does your board have a reset button ? Can you try to hit the reset button right after the compilation finishes and the IDE fires the programmer ? (You should enable verbose compilation and upload from the IDE options)

nkuck, 2016-07-01:
I've tried that but probably not too accurately. Will try again. Thanks for the reply. 

Live simply!

nkuck, 2016-07-01:
What do you mean from the IDE options?

Live simply!

jbrazio, 2016-07-01:
See the `Show verbose during` checkboxes ? Check both of them.

[image]

nkuck, 2016-07-01:
Tried several times with no successes. More thoughts? Thanks for any ideas. I have been at this for weeks. 

Live simply!
You can only take with you what you have given.

KevMags, 2016-07-01:
That happens to me every time that I use a USB cable other than the one that came with my Arduino or Arduino based boards (blue/purple short cord, shielded).  Try another USB cable.

nkuck, 2016-07-01:
Thanks very much for the suggestion. I have tried a number of USB cables, including the short blue ones that come with the Arduino boards, with no luck. I am still able to load and print files to the printer, just not able to upload a new firmware. Baffled for several weeks. The board is a Merlin on a Zonestar 3D printer (Prusa i3).

Live simply!
You can only take with you what you have given.

jbrazio, 2016-07-01:
I will make question, are you **really** sure you have the right COM port selected ?

Did you use an ICSP programmer with board before ?

jbrazio, 2016-07-01:
Or do you have the correct board selected on the IDE ?

Blue-Marlin, 2016-07-01:
Or do you have an other connection open to the board? A host?

nkuck, 2016-07-01:
Yes, correct com port.The unit came preloaded with the bootloader and repetier 0.91 firmware.I am trying to upload a current Marlin firmware to the board.I have not tried this before, but I can run files to the printer successfully. 

Live simply!
You can only take with you what you have given.

nkuck, 2016-07-01:
I have selected the Sanduino board, as required.

Live simply!
You can only take with you what you have given.

thinkyhead, 2016-07-01:
Check out this thread, starting here: https://github.com/mtu-most/franklin/issues/5#issuecomment-189793398

jbrazio, 2016-07-11:
As this issue is not directly related with Marlin but with the Arduino toolkit and due to it's current inactivity I will close it.

closed 2016-07-11 by jbrazio

drones-squad, 2016-08-11:
hello all,

 please reopen this since I also have this issues . There is no answer and this wasn't solved for future references  so why close ... ? 

greets

nkuck, 2016-08-11:
I had to reflash the bootloader, and I think that that is what allowed me then to load the firmware.

drones-squad, 2016-08-11:
with what did you do it and how? Can it work with a arduino nano usb ?

greets

nkuck, 2016-08-11:
I used an Arduino Uno. Google how to use an Arduino to flash a bootloader and you will get a number of methods.

drones-squad, 2016-08-11:
yeah but will a nano do ?  The others isnt worth the expense I can just buy a new melzi board.
The nano is pretty cheap

greets

nkuck, 2016-08-11:
I haven't tried a nano. Try it and let me know. 

Live simply!
You can only take with you what you have given.

Bhalddin, 2018-03-03:
I know that this is an old post. Just i would like to put the solution i found for this issue. I solve this by changing the baud rate to 19200